Laxilef/OTGateway #2: Set maximum supply temperature

theelog, 2023-09-19:
Is there a way to control the maximum central heating supply temperature.

Laxilef, 2023-09-19:
The controller tries to get the min and max heating temperature from the boiler, if it supports it
https://github.com/Laxilef/OTGateway/blob/3cafcde58c21fbfbba2f645143a325f814e402b8/src/OpenThermTask.h#L319

In variables "vars.parameters.heatingMinTemp" & "vars.parameters.heatingMaxTemp" values by default

theelog, 2023-09-19:
Is the retrieved max reported somewhere?

Laxilef, 2023-09-19:
Via mqtt explorer
[image]

Laxilef, 2023-09-19:
Maybe later add entities to HA

theelog, 2023-09-19:
Nice, that's even better. I knew I've configured it in the furnace menu at one time.
[image]

Just sharing a quick way to get it into Home Assistant:
```
mqtt:
  sensor:
    - unique_id: opentherm_heating_max_temp
      name: Opentherm Heating Max Temp
      state_topic: "opentherm/state"
      value_template: "{{ value_json.parameters.heatingMaxTemp }}"
      device_class: temperature
      unit_of_measurement: "%C"
```

Laxilef, 2023-09-21:
Hi
I added new parameters for min and max temperature to the firmware. They now native show in Home Assistant. And now you can set them yourself, but within the limits receive from boiler.

Pls test it

Laxilef, 2023-09-21:
And if you want to receive the indoor temperature from a diyless board, you can set the "Indoor sensor pin" parameter to 14 in the settings (web). After that, in the home assistant, select "Indoor temperature source" = External

closed 2023-10-08 by Laxilef