Laxilef/OTGateway #3: Boilers compatibility

Laxilef, 2023-09-21:
Please write here the name of your boiler and its compatibility features (what works, what doesn’t work).
I will add to the table in READ ME.

This will be very useful for other people.

filipjurik, 2023-10-06:
My Baxi Nuvola DUO-TEC HT 16 boiler works well
I'm using Chinese WEMOS D1 MINI + DIYLESS ESP8266 THERMOSTAT SHIELD.
Everything works fine but I have one question.
I don't use equitherm.
Target temperature (heating) is the temperature of the water in the system.. right?

If I have heating turn on (climate.opentherm_heating), but boiler don't heating (i use ON/OFF + zone heating valves), the pump is still running. 
Is possible set target temperature in the house, and when it is reached, turning off circulation pump, and using "climate.opentherm_heating" only to control and setting the heating water temperature ?

Laxilef, 2023-10-07:
> My Baxi Nuvola DUO-TEC HT 16 boiler works well
Thanks for the info!

> Is possible set target temperature in the house, and when it is reached, turning off circulation pump, and using "climate.opentherm_heating" only to manually control and set the temperature of the heating water?
Hmm, I think you can turn off the heating using automation in the home assistant. Turning off the heating should turn off the pump.

In my system, just like yours, installed electric valves to accurately regulate the temperature in the zones. And I use equitherm mode. This prevents the heating system from swinging. Try this.

dante1613, 2023-10-16:
Котел: **Ferroli DOMIcompact C 24** (плата **MF08FA**)

Master Member ID: **211**

Конфиг: [WEMOS D1 MINI](https://www.aliexpress.com/item/32863218532.html?spm=a2g0o.order_list.order_list_main.4.e2c41802oCWMlQ) + [DIYLESS Master OpenTherm Shield](https://diyless.com/product/master-opentherm-shield) + датчик [Xiaomi LYWSD03MMC](https://www.aliexpress.com/item/1005004446445069.html?spm=a2g0o.order_list.order_list_main.22.63f81802zOuhX9) (через MQTT [отсылаю](https://github.com/Laxilef/OTGateway#:~:text=The%20temperature%20inside%20the%20house%20can%20be%20set%20using%20simple%20automation) температуру в топик) + [DS18B20](https://www.aliexpress.com/item/32467815969.html?spm=a2g0o.order_detail.order_detail_item.19.551bf19cngqpsI) на улице (так же в топик через MQTT). Ссылки кликабельные, для тех кто решит повторить мой опыт с таким же котлом

Логика работы: Эквитерм (PID не пробовал)

Не работающие сенсоры и кнопки: **Pressure** (показывает нули) и **Fault code** (ну или нужно дождаться появления ошибок?), **Use equitherm in emergency** 

Все остальное работает корректно:

- мин и макс температура для отопления и горячей воды (как в паспорте котла)
- включение и выключение отопления отопления и горячей воды
- отключение горячей воды через выключатель
- показывает процент модуляции
- статус включения обогрева и горячей воды
- статус пламени горелки
- установка температуры для воды и отопления
- установка гистерезиса
- турбо режим (я так понимаю для его работы нужно разница как минимум 1 градус между температурой в доме и установленной на термостате в режиме эквитерм? Так как иногда кнопка отключается и не меняется точка нагрева)

Баг или фича: показывает две **Outdoor temperature source**

<details><summary>Скриншот из Home assistant</summary>[image]</details>

Работает просто отлично, я прям не ожидал что будет все так просто (котел то довольно старый, 2005 год, думал что вообще не заведется). Сначала прошил собранную прошивку 1.3.1, но не сохранялись настройки MQTT после перезагрузки (еще до того как вы выложили по d1 mini), потом досмотрел что это для d1 mini pro и собрал через Platformio, стало все нормально. Раньше до этого пользовался реле с сухими контактами, а сейчас уже 5 день как пользуюсь вашей прошивкой, так что статистики пока мало. Да и холодов особо не было, до 5 градусов максимум. но температуру держит хорошо, теперь главное как с расходом.
<details><summary>Оффтоп по потреблению</summary>По замерам на минимуме модуляции (2%) потребляет где-то 0.75 куба газа в час, на максимуме (100%) выходит 1.5 куба (хотя по паспорту должно до 2.73 куба, то ли газ такой сейчас в Украине, то ли так настроена подача газа в котле). Теперь как бы сделать сенсор в Home assistant, который будет рассчитывать потребление газа брав данные из сенсора модуляции, по логике: с 2% до 50% модуляции это 0,75 куба, а с 50 до 100% это уже 1.5 куба, так как счетчика отдельного на котел нет и замеры проводились с секундомером и точно не могу посчитать, так как нет возможности самому регулировать модуляцию и точно высчитать сколько газа потребляет в на каждой ступеньке с 2 до 100%, а так хоть будет примерное понимание. Сейчас сделал по <a href="https://seanblanchfield.com/2022/02/virtual-gas-meters-home-assistant">гайду</a> (настраивал еще под реле с сухими контактами) и поставил 1 куб в час при работе отопления и горелки. Кстати по его же <a href="https://seanblanchfield.com/2022/02/virtual-gas-meters-home-assistant">методике</a> балансировал радиаторы отопления</details> 

<hr>

Большое спасибо за ваш труд!

Laxilef, 2023-10-16:
@dante1613 спасибо за отзыв!

Давление к сожалению не все котлы показывают. Мои Baxi тоже ничего не возвращают.
Чтобы проверить чтение ошибок можно перекрыть газ во время работы котла, котел должен упасть в ошибку, соответственно потом посмотреть в HA. Если ошибок нет, то сущность недоступна.
Use equitherm in emergency доступен только если выбран датчик уличной температуры External или Boiler. Для расчёта необходима уличная температура, а если мы её передаём из HA, то при отвале Mqtt её больше никто передавать не будет.

Селектор Outdoor temperature source должен быть один,  странно. Id у второго какой?

Что касается расхода газа, посмотрите в сторону съёма показаний с счётчика. Котел может по своим алгоритмом регулировать подачу газа несмотря на показания модуляции.

dante1613, 2023-10-17:
Проверил на чтение ошибок по вашему способу, работает. Сначала показал ошибку E00, а потом сменил статус на E01, который соответствует нужной неисправности (судя по данным с [интернета](https://www.freeboilermanuals.com/ferroli/fault-codes/), хотя там А01 указывают, возможно это для более новых моделей, не могу сказать, так как на моем котле нет экрана чтобы точно узнать). После теста отключил термостат в home assistant и включил заново, после этого котел стал работать как обычно, а Fault code сменился обратно на недоступно.
<details><summary>Скриншоты</summary>[image][image][image]</details>

Касаемо расхода газа, съем показаний с счетчика не подходит, так как он считает не только расход на отопление но и расход на подогрев воды, а так же газовую плиту. Мне то не так важны сами показания в home assistant, а больше как показатель работы системы отопления, чтобы смотреть какой алгоритм более комфортный и экономичный. Про opentherm вроде бы как говорят что с ним больше экономии, чем с термостатом с сухими контактами и в основном это говорилось про PID регулирование, а с ПЗА информации нет. Вот расход газа и показал бы наглядно, экономней это или на том же уровне но с большим комфортом.
А касаемо модуляции, то я в течении 5 дней смотрел и периодически замерял и точно получается что при 2 % потребление 0.75, а при 100% 1.5 куба, показания не меняются. Но эти цифры не линейные, взять 2 % 0,75 за точку отсчета не получится. Примерно с 2 до 50 % это 0,75, а с 50 до 100% уже 1,5. Опять все что не 2 и не 100, трудно выловить, так как чаще всего я вижу котел с этими процентами. По этому и появилась идея, чтобы сделать сенсор, который будет смотреть модуляцию и пересчитывать потом по формуле, но как это сделать я пока не знаю. Мне бы хватило и примерной точности, всяко лучше чем усредненный 1 куб, который сейчас считается на группе сенсоров отопления и горелки и времени их работы.

По поводу Outdoor temperature source, вот что показывает в home assistant:
<details><summary>Скриншоты</summary>[image][image]</details>

Еще хотел бы вас попросить, не могли бы вы скинуть скриншоты своих настроек эквитерм режима и графиков работы термостата, чтобы было более понятно правильно ли у меня ПЗА работает или нет. Или может мне стоит еще поиграться с выставлением N K и Т. Это все конечно индивидуально, от площади, утепления дома, температуры на улицы, но хотя бы примерно будет понятно в каком направлении оно должно работать

<details><summary>Скриншоты</summary>[image][image]</details>

На график с 11 до 18.30 не обращайте внимания, на улице было больше +15 градусов, по этому я выключил котел. Ну и там где много пробелов, это включался подогрев воды. Гистерезис стоит 0.2 (по 0.1 в каждую сторону верно?)

dante1613, 2023-10-17:
Вот что еще заметил, если к примеру у меня стоит на термостате 23 градуса и в комнате 23, котел на минимальной модуляции (2%) поддерживает температуру 43 градуса (Heating setpoint). Если повысить температуру термостата до 26 градусов, Heating setpoint поменяется на 65, котел включит горелку на 100 % и начнется набор температуры, но если поменять температуру на термостате обратно на 23 градуса, Heating setpoint вернется обратно к 43 градусам, но котел все так же продолжит набирать температуру со 100% модуляцией, помогает только выключение и включение обратно термостата, либо уменьшение температуры на термостате, к примеру с 23 до 22, тогда термостат его отключит сам. Вряд ли кто-то будет использовать в таком режиме, переключая туда-обратно, но заметил такой момент. Не понятно в чем именно причина, в прошивке или в мозгах котла или просто особенность работы такая.

dante1613, 2023-10-17:
Нашел причину, почему у меня два **Outdoor temperature source**. Перед тем как собрать прошивку через Platformio, я перепробовал все версии прошивок из релизов, от 1.2.1. до 1.3.1, думал дело в них или то что у меня глючит d1 mini, так как я ее снял с самодельного пульта для кондиционера (в запасе не было еще одной d1 mini, пульт для кондера до лета и не нужен). От в 1.2.1 и был этот селектор, а так как базовый топик один и тот же, то MQTT его сохранила, а я этого не заметил, так как когда пробовал прошивки, то только проверял их на сохранение данных MQTT и не запомнил какие именно должны быть настройки. Через MQTT Explorer нашел топик и там указана прошивка 1.2.1, после этого я уже вспомнил что менял их. Удалил этот топик, пропал селектор и больше не появляется
<details><summary>Скриншот</summary>[image]</details>

Balaganoff, 2023-10-17:
Котёл AEG GBA124 - все завелось с первого раза. Не показывает только давление в системе, хотя ошибку по давлению показывает. Скорее всего в котле просто датчик который выдает дискретный сигнал при понижении порогового давления, а верхний предел сбрасывает клапан. 
Спасибо за работу!!
 Ждем продолжения и расширенного функционала у автономного устройства. Мало-ли что с компьютером в зиму случится )) Хотелось бы чтобы HomeAssistant выполнял контролирующие функции и автоматику. А основное бы работало на ESP. Я имею ввиду чтобы можно было регулировать и просматривать через веб интерфейс.
 Типа такого 
[image]

Laxilef, 2023-10-17:
> Еще хотел бы вас попросить, не могли бы вы скинуть скриншоты своих настроек эквитерм режима и графиков работы термостата, чтобы было более понятно правильно ли у меня ПЗА работает или нет. Или может [...]

Хорошо, когда будут морозы - выложу. Сейчас у меня котел включается раз в несколько дней, там не оч интересные графики :)
Судя по вашим графикам, у вас с N коэф. перебор. Проще всего его подобрать установив гистерезис в 0, K = 0, T = 0, установить 20 градусов и подбирать коэф. N так, чтобы стабильно поддерживалось 20 градусов. Ночные показатели оптимальнее, т.к. солнцем не догревает помещения.

После этого крутить K коэф. для поддержания температур отличных от 20 градусов. T коэф. отвечает за резкость регулирования. А после подбора этих коэф. можно включить гистерезис, чтобы отключать нагрев, когда днем солнце нагревает помещение.

> но если поменять температуру на термостате обратно на 23 градуса, Heating setpoint вернется обратно к 43 градусам, но котел все так же продолжит набирать температуру со 100% модуляцией

Котел со 100% модуляцией греет теплоноситель до 43 или до 65?

> Мало-ли что с компьютером в зиму случится )) Хотелось бы чтобы HomeAssistant выполнял контролирующие функции и автоматику. А основное бы работало на ESP.

Сейчас так и есть. Если контроллер не сможет подключиться к mqtt серверу или не сможет подключиться к wifi, то он переходит в аварийный режим (emergency mode) и будет поддерживать заданную для emergency mode температуру в заданном режиме. Но вот менять параметры температуры не получится.

Добавить регулировку температуры из веба можно, но при отвале wifi это не решит проблему)

dante1613, 2023-10-17:
> Судя по вашим графикам, у вас с N коэф. перебор

N сначала использовал по дефолту 0.7, потом до 0.8 поднял, значит нужно опустить?

> Котел со 100% модуляцией греет теплоноситель до 43 или до _65?_

До 65 градусов

Balaganoff, 2023-10-17:
> Добавить регулировку температуры из веба можно, но при отвале wifi это не решит проблему)

Почему ? ESP ведь сама в качестве точки доступа и web сервера может

Laxilef, 2023-10-17:
> N сначала использовал по дефолту 0.7 (как в рекомендации по настройке), потом до 0.8 поднял, значит нужно опустить?

Попробуйте поставить N=0.7-0.8, K = 0, T = 0. Если будет поддерживать 20 +- 0.2 ночью, тогда K уменьшать. T выше 2.5 не рекомендую, будет раскачивать систему, только если нужно резко нагреть дом, но для этого есть "турбо режим", он задирает коэф. T.

> До 65 градусов

Такого не должно быть. Возможно, котел игнорирует частую установку температуры. Добавил в мастер ветку принудительную установку температуры раз в минуту. Попробуйте скомпилить прошивку из мастер ветки и проверить.

> Почему ? ESP ведь сама в качестве точки доступа и web сервера может

Если вы постоянно находитесь рядом с контроллером, то м.б. и выход. Но в этом случае проще починить сеть и/или mqtt сервер на HA. Если не рядом, то при отключении wifi вы к хотспоту не сможете же подключиться.

menuet88, 2023-10-17:
Hello,
Basically I've use this software with custom board to control the Thermet Ecocondens Silver 35kW boiler. First impression - it works. The **boiler doesn't report the water pressure** (there is no sensor for that), all other functions works ok. In the logs also requests 126 and 127 are invalid. I've just start testing it, for sure I came back to write my suggestions.

One question: is there any particular reason why you set DHW temperature to zero when DHW is off? It would be nice to have this reading all the time, because in my case it shows the water reservoir temperature.
And suggestion: Maybe, if the boiler doesn't support some features over opentherm (report INVALID) there is a chance to count this invalid response for request and if is for example 10 then remove this request? I know that this doesn't change anything, but I don't like when something throwing errors :)

Best regards,
Menuet88

Laxilef, 2023-10-17:
@menuet88, thanks for the info!

> One question: is there any particular reason why you set DHW temperature to zero when DHW is off?

I'll fix this in the next version.

> And suggestion: Maybe, if the boiler doesn't support some features over opentherm (report INVALID) there is a chance to count this invalid response for request and if is for example 10 then remove [...]

I thought about it. But I decided to leave it as is. Because different boilers have different software. We can break the logic of the controller on boilers that do not fully implement the opentherm specification.

Rolex48, 2023-11-04:
Baxi duo-tec compact 1.24 пока не выходит подключить, плату проверили по схеме несколько раз - всё в норме.(собрана на макетке) 
Откуда берёте "Идентификатор главного участника"? ЕСП прошилась адекватно и настроилась. 
[image]

Laxilef, 2023-11-05:
Даже с некорректным member master id котёл должен отвечать на базовые запросы. Подбирают методом перебора, но это необходимо только в том случае, если котлом не получается управлять (менять температуру и т.п.)

Если у вас вообще котел не отвечает, проверьте в настройках пины OT. Иногда помогает поменять их местами.

bigbrother72, 2023-11-06:
> Что касается расхода газа, посмотрите в сторону съёма показаний с счётчика.

Здравствуйте
Я два года уже пользуюсь для подсчета расхода газа проектом
https://github.com/jomjol/AI-on-the-edge-device

Данные берутся с фото счетчика, оцифровываются и выгружаются по mqtt в HA. Можно смотреть статистику по часам, дням, неделям, месяцам, годам..

[image]
[image]
[image]

По OTGateway.
Можно добавить в список совместтимости BAXI LUNA-3.

Данные о внешней температуре получаю с датчика котла.
Внутренняя температура через автоматизацию в HA с датчика esphome.

Также немного переделал код в части встроенного датчика температуры на контроллере Opentherm. У меня DS18B20 примотан к трубе обратки около самого котла и в HA передается температура обратки тоже. 
Было бы неплохо ка-то предусмотреть в проекте такой вариант.

[image]

Laxilef, 2023-11-06:
@bigbrother72, благодарю за информацию. А зачем снимать температуру обратки, если мы на нее нигде не ориентируемся?

aurelmarius, 2023-11-20:
Viessmann Vitodens 0-50w, working nicely.[image]

Laxilef, 2023-11-22:
@aurelmarius thanks for the info!

kulisau, 2023-11-23:
Boiler: **Radiant r2k 24**
Everything but the pressure sensor works. 
DHW is working without the heating on only if the following settings are disabled:
- Opentherm CH2 enabled
- Opentherm heating CH1 to CH2
- Opentherm DHW to CH2

And the following enabled:
- Opentherm DHW present
- Opentherm summer/winter mode

kerajkeraj, 2023-11-23:
Hello All,
i am using gas boiler BAXI duotec 24, current setup is:
dyiless gateway hw - master, slave and wemos d1 mini esp8266.
Boiler part and master part works as expected - i can manage temp of CH and DHW in homeassistant.

I am facing issues with existing thermostat BAXI connected to slave shield - it reports error 83 as can be seen on attached photos:

[image]
[image]

Anyone could help on slave-thermostat side?

Thank you for your help

Laxilef, 2023-11-26:
@kerajkeraj it won't work the way you want.  The baxi thermostat is the master, the firmware from this repository is also the master, and the boiler is the slave.
2 masters will conflict. Because in this firmware use algorithms to calculate the heating temperature. But your thermostat doesn't know about it. And the thermostat will try to overwrite the temperature that the controller set.

kulisau, 2023-11-28:
Boiler: Radiant R2KA
Central heating works out of the box, DHW requires the "Modulation sync with heating" setting toggled in controller's settings.
The pressure and DHW flow sensors are unavailable.

michielbeijer, 2023-12-06:
Intergas Kombi Kompakt working fine with Diyless Thermostat shield. Combi boiler, so hot water is on demand only. All parameter (relevant to me) reporting / being set correctly. Master ID = 0.

strk90, 2023-12-10:
Baltur Tesis 24 works.
Сontrol board - Nordgas 5002001904.
Thank a lot!

MikelillOi, 2023-12-12:
I have assembled a DIYLESS Thermostat shield with a D1 Mini and installed version 1.3.3

It works with my Domusa Bioclass HM 16 pellet boiler. I have left the master ID with value 0 (default)

Almost everything works, DHW, correct. Diagnostics, fault messages, etc. ok. But I don't see the % modulation, my old Tado showed me that value. Is there any way to see it? I'm interested to see if the modulation works well. I don't see the value of pressure either.

On the other hand, wouldn't it be more correct to show the HW controller in HA as water_heater entity and not as climate?

Laxilef, 2023-12-12:
@MikelillOi modulation and pressure entities are not activated by default. You need to [activate them](https://github.com/Laxilef/OTGateway/wiki/FAQ-&-Troubleshooting#how-can-i-activate-deactivated-sensors-and-settings-in-my-home-assistant) in Home Assistant.

MikelillOi, 2023-12-12:
> @MikelillOi thanks for the info! Modulation and pressure entities are not activated by default. You need to [activate them](https://github.com/Laxilef/OTGateway/wiki/FAQ-&-Troubleshooting#how-can-i-activate-deactivated-sensors-and-settings-in-my-home-assistant) in Home Assistant.

Thank you! Now I see both data. But modulation still at 0. An the boiler heating is always turning on, and off

[image]

I read about "Modulation sync with heating" parameter, but I can´t find it. Will this solve the problem I mentioned?

Laxilef, 2023-12-12:
When the burner is on, the entity "flame" is displayed on? 
Does the burner really turn off when the status is heating=off?

Suxsem, 2023-12-12:
Ferroli Kalis 24C, ALL SENSORS AND CONFIGS work perfectly. Default master member id. Boiler does not have a builtin external temperature sensor so I'm using weather forecasts with automation to copy the external temperature value.

Awesome project, I just replaced (and sold) my 150€ "ferroli connect" thermostat with a 15€ board that gives me lots of flexibility, thank you!!

MikelillOi, 2023-12-12:
> When the burner is on, the entity "flame" is displayed on? Does the burner really turn off when the status is heating=off?

i'm not at home now and I can't check if the boiler really turn off. Then, when I arrive at home I will make some tests.

[image]

Here you have the 3 sensors, heating, flame and modulation

Laxilef, 2023-12-12:
Your "flame" is always on, even when heating = off.
I don't know why this could be, but it's not right. The request for get modulation is standard.  If it is always 0, then the boiler does not support this request.

MikelillOi, 2023-12-12:
> Your "flame" is always on, even when heating = off. I don't know why this could be, but it's not right. The request for get modulation is standard. If it is always 0, then the boiler does not support this request.

I can confirm that the boiler continues to work even though it appears as off in the integration.

The comercial Tado thermostat showed the modulation percentage correctly. Can we read any log or something to investigate it?

DeadlySin2, 2024-01-19:
> Также немного переделал код в части встроенного датчика температуры на контроллере Opentherm. У меня DS18B20 примотан к трубе обратки около самого котла и в HA передается температура обратки тоже. Было бы неплохо ка-то предусмотреть в проекте такой вариант.
> 
А можно спросить где конкретно и какой код правили?

hoddl, 2024-01-29:
Thank you for the Projekt. Very Helpfull.
It works at an Remeha Avanta Plus, an older Gateway Module from DIYLESS with  D1 Mini. 
My DHW has no tank, it works as flow heatig  so wenn the DHW themp is set to 40° and on, the heating try to reach 40° so i have to switch off the DHW or is ther any chance to report the temperatur of the flow DHW?

Pressure sensor is not supported. (I found the entity and activated it)
I didn't found the DHW flow entity?

Laxilef, 2024-01-30:
@hoddl try unchecking "CH2 always enabled". DHW flow rate is supported from 1.4.0, you can build the firmware [from sources](https://github.com/Laxilef/OTGateway/wiki#build-firmware).

DeadlySin2, 2024-01-31:
Can confirm that this firmware works with Baxi Luna 3 Comfort. 
I have three-way valve instaled with with an indirect heating boiler so i swithced the folowing parametrs:
[image]

hoddl, 2024-01-31:
Thank you. I try to build version 1.4.0-rc-12 from Github. At the configuration screen at 192.168.4.1 comes "The Filesystem is not flashed". The precompiled version 1.3.3 works fine.

PavelTelnov, 2024-01-31:
Можно добавbть Navien NCB700
Собрал плату, прошил, подключил к котлу, подключил к HA. Наблюдаю работу в Grafana.
Котел регулирует температуру по обратке.
Панель Opentherm Gateway Heating - при изменении настроек температуры, на дисплее котла ничего не меняется.
Панель Opentherm Gateway DHW - при изменении настроек температуры, на дисплее котла ничего не меняется.
Так и должно быть?
Задача: управлять температурой горячей воды через HA. Стоит планшет в ванной, для себя хочу налить погорячее, для детей похолоднее. Не хочется бегать в котельную.
Вопрос 2: отображение температуры обратки можно реализовать?
[image]

Maximper, 2024-02-01:
Пробую с котлом Buderus u072 24. Собирал в platformioпоследнюю версию.  Плата ESP32-WROOM-32D. Подключился, можно устанавливать температуру отопления и бойлера. Вижу температуру контура отопления, с датчика бойлера через котел информация не приходит. Через некоторое время связь отвалилась, попробовал старый ot адаптер который стоял до этого с версией esphome и точно работал, тоже самое.
[image]

Laxilef, 2024-02-01:
@DeadlySin2 thanks for info!

@Maximper попробуйте сейчас собрать прошивку, поправил тайминги в последних коммитах. О проблемах лучше в отдельном issue.

Laxilef, 2024-02-04:
@andrey150960 после изменения флагов ESP перезагружали? Откройте отдельный issue.

andrey150960, 2024-02-04:
https://github.com/Laxilef/OTGateway/issues/3#issuecomment-1925822816
После д.бого изменения делал перезагрузку. В настройках котла поставил режим "1" -это термостат или ПДУ. 
При отключенном ПДУ ( без ESP32) термостат (сухой контакт)  работает.

Maximper, 2024-02-06:
Buderus U072 18 и Buderus U072 24 прекрасно работают! Спасибо! пробовал и на esp32 и на wemos d1 mini. Можно ставить в совместимость. Работает управление отоплением полностью, воды только выставляем температуру до которой греется, датчика бойлера не видно. Максимальные и минимальные значения температуры отопления/гвс только в режиме просмотра. Показывает % модуляции. Можно просмотреть и сбросить ошибку котла.

Druid336, 2024-02-12:
Котел Baxi ECO Life 1.24F работает прекрасно на прошивке 1.4.0 rc14
надо поиграться с настройками ПЗА и вывести на идеал, очень доволен. Спасибо за быстрый отклик в телеграм-канале и решение проблем.
[image]

DeadlySin2, 2024-02-13:
Changed parametrs for Baxi Luna3 Comfort на Dublicate DHW to CH2. With previous hot warter does'nt work when heating is disabled. (I have edited my previous message)

MrDogYar, 2024-02-29:
Работает.
Котел bosch gaz 6000 w wbn 6000-24
(Wemos D1 mini) - firmware_d1_mini_1.4.0-rc.13 
Внешние датчики температуры 

На 15 релизе не работает. Датчики уличной и комнатной температуры каждые 5 секунд обнуляються. 
Котел каждые 2 секунды включаеться и выключаеться.

Laxilef, 2024-02-29:
@MrDogYar спасибо за информацию! Проблема с датчиками исправлена в rc16.

MrDogYar, 2024-03-10:
BAXI eco4s не завелось :-(
Invalid response after setBoilerStatus: TIMEOUT
Беглый опрос гугл показал, что там protocol OpenTherm v2.2

Laxilef, 2024-03-11:
@MrDogYar откройте issue и приложите логи с момента старта esp, напишите какой адаптер, какая esp и какая версия прошивки. Протокол от версии к версии не менялся настолько, чтобы сломать обратную совместимость. TIMEOUT - нет ответа от котла.

MrDogYar, 2024-03-11:
Добрый день!
Завел Issue по проблеме отсутствия связи с котлом.
https://github.com/Laxilef/OTGateway/issues/57
 
 
PS: Так же хочу поделиться маленьким лайфхаком, который можно добавить в wiki
 
 
Что бы не писать автоматизацию по установке температуры с внешних датчика в Home Assistent.
Можно на прямую пробросить температуру с датчика. Надо в Тasmoto прописать Rule
 
Rule 1 ON DHT11#temperature DO publish opentherm/state/set {"temperatures": {"indoor": %value%}} ENDON
Rule 1 1

> 
> 
>@MrDogYar откройте issue и приложите логи с момента старта esp, напишите какой адаптер, какая esp и какая версия прошивки. Протокол от версии к версии не менялся настолько, чтобы сломать обратную совместимость. TIMEOUT - нет ответа от котла.
 
 
С уважением,
Денис Титов
***@***.***

arseniydidenko1, 2024-03-21:
Приветствую! А как можно управлять котлом(установка температуры ГВС и СО) с вебморды без использования НА и т.д.?
комнатный датчик температуры я подключил, температуру с улицы модуль тянет с самого котла.
Вводные данные:
котел: Ferroli DivaTech D C24
модуль: ESP32
Прошивка: 1.4.0-rc.17
инфа с вебморды:
[image]
[image]

MrDogYar, 2024-03-22:
> Приветствую! А как можно управлять котлом(установка температуры ГВС и СО) с вебморды без использования НА и т.д.? комнатный датчик температуры я подключил, температуру с улицы модуль тянет с самого [...]

Через связку mqtt brocker + mqtt client dashboard

arseniydidenko1, 2024-03-22:
> Через связку mqtt brocker + mqtt client dashboard.

Ну а без всего этого? Просто через веб или телнет/ссш? как в линукс - просто отредактировать файл...
Это же, млин, тоже своего рода линукс))

arseniydidenko1, 2024-03-22:
> Ну а без всего этого? Просто через веб или телнет/ссш? как в линукс - просто отредактировать файл... Это же, млин, тоже своего рода линукс))

Ну тут я более-менее разобрался: по апи курлом(знаю что это топорно) можно выставлять не обходимую температуру ГВС и СО. Не нашел возможности установить требуемую температуру в комнате согласно датчику температуры внутри помещения, что б модуль работал как обычный термостат с сухим контактом.
Безо всяких НА, МКТТ, КГБ, ГКЧП, ПМЖ и т.д.

madkeeper81, 2024-03-23:
Котел Mizudo, все работает по дефолту.

arseniydidenko1, 2024-03-23:
> Котел Mizudo, все работает по дефолту.

По дефолту чего? модуля или котла? где устанавливали целевую комнатную температуру? где температуру в СО или ГВС?

madkeeper81, 2024-03-24:
> > Котел Mizudo, все работает по дефолту.
> 
> По дефолту чего? модуль или котла? где установили целевую комнатную температуру? где температура СО или ГВС?

Mizudo 24mth, Master MemberID 0, уличная температура подтягивается с датчика котла, температуру дома взял с BLE через автоматизацию НА, температуру косвенника (ГВС) тоже с котла,  прошивка v1.4.0-rc.17, собрал на d1,

arseniydidenko1, 2024-03-24:
> через автоматизацию НА

Ваш ответ мне никак не помог, т.к. я спрашивал про настройку БЕЗ НА и подобной...

[33 more replies]