Laxilef/OTGateway #5: Работа в режиме PID

bigbrother72, 2023-10-10:
Если включен PID режим, то температура вычисляется, только когда котел набирает температуру
Функция getPidTemp() вызывается только в одну сторону.
```

    if (settings.pid.enable && vars.states.heating) {
      float pidResult = getPidTemp(....
 
```
По достижении температуры, котел просто отключается, и setpoint далее не меняется.

Потом когда температура в помещении упадет меньше заданной, то котел  опять включается и начинает работать с предыдущей температуры - т.е .просто дискретный термостат on/off

И это еще не всё. После нескольких циклов что-то зависает и далее просто не включается отопление.

Установлено 21 градусов целевая температура. 
Несколько раз набирало эту температур.
Сейчас до +20,8 уже упало. 
Через telnet вижу, что 
[INFO] Heating enabled: 1

Но отопление не включается

[image]
[image]
[image]

Laxilef, 2023-10-11:
Какой у вас котёл? Какое значение параметра hysteresis?
Предполагается, что vars.states.heating должен быть true даже если горелка котла выключена.

Laxilef, 2023-10-11:
Проверьте

https://github.com/Laxilef/OTGateway/releases/tag/1.3.1

bigbrother72, 2023-10-11:
> Какой у вас котёл? Какое значение параметра hysteresis? Предполагается, что vars.states.heating должен быть true даже если горелка котла выключена.

Котел Baxi Luna 3.
hysteresis =0.1

Еще вопрос:

В нижеуказанном коде наверное надо поменять лимиты 0-100 на vars.parameters.heatingMinTemp, vars.parameters.heatingMaxTemp, так как котел не может выдавать от 0 до 100 градусов температуру отопления.
```
    newTemp = round(newTemp);
    newTemp = constrain(newTemp, 0, 100);
    return newTemp;
```

bigbrother72, 2023-10-11:
> Проверьте. Должно быть исправлено. После обновления настройки сбросятся на дефолтные, сохраните их перед обновлением, чтобы потом заново выставить.
> 
> https://github.com/Laxilef/OTGateway/releases/tag/1.3.1

завтра попробую.

Laxilef, 2023-10-11:
> В нижеуказанном коде наверное надо поменять лимиты 0-100 на vars.parameters.heatingMinTemp, vars.parameters.heatingMaxTemp, так как котел не может выдавать от 0 до 100 градусов температуру отопления.

Там несколько лимитов дальше идет, последний:
https://github.com/Laxilef/OTGateway/blob/6d3172b73b2033a36a3b7c643ee0d3380d3fc5f4/src/RegulatorTask.h#L62C1-L64C6

closed 2023-10-21 by Laxilef

Symon84, 2025-01-22:
Hello, in my case I have both the PID and EQUITHERM active.. 
The hysteresis value is set to 0.5.I note that the PID calculates the value with respect to the setpoint temperature, i.e.: 20.5 degrees. Therefore, when the ambient temperature drops to 20.4 degrees, the PID increases the supply setpoint, but the boiler still has to start because it will start at 20 degrees. 
As a result, every time the boiler turns on at 20 degrees, the flow temperature setpoint is always at maximum (PID + EQUITHERM), and then gradually lowers as the ambient temperature rises. 
Is it possible to synchronize the PID with the boiler start-up? 
Or with the desired room temperature setpoint - the hysteresis value? 
It would be great if this thing could be enabled with a flag like "PID hysteresis sync"

Laxilef, 2025-01-23:
Hi @Symon84 
Can you show the graph of ``sensor.opentherm_heating_temp``, ``sensor.opentherm_heating_setpoint_temp`` and ``binary_sensor.opentherm_flame``?

Symon84, 2025-01-23:
> Hi [@Symon84](https://github.com/Symon84) Can you show the graph of `sensor.opentherm_heating_temp`, `sensor.opentherm_heating_setpoint_temp` and `binary_sensor.opentherm_flame`?

Yes:

[image]

and this is a detail with indoor temp:

[image]