Laxilef/OTGateway #6: Поддерживаемые котлы

Balaganoff, 2023-10-13:
С моим котлом AEG GBA124 прошивка не работает. Хотя другие (тасмота+опентерм и пр.) видят мой котел и регулируют. 
Тут использована какая-то урезанная библиотека ?

Я правильно понял - можно по-русски?

Laxilef, 2023-10-13:
Опишите что не работает: не регулирует температуру, не получает значения с датчиков и т.д.
У тасмоты такая же библиотека от Игоря Мельника.
Пины правильные в настройках вписаны?

Balaganoff, 2023-10-13:
Спасибо за ответ разобрался ! У меня пины не так распаяны (осталось от другого проекта) как на вашей схеме.
В настройках изменения не сохраняются. По крайней мере как я ни старался не получилось. Пришлось пересобрать из исходников по свои пины и котел ответил ! Я почти счастлив ! )) 
Еще один вопрос - как заставить работать 18b20 чтобы температуру внутри помещения учитывало именно с него ?
Датчик точно подключен правильно и работает. 
struct {
      // 1 - manual, 2 - ds18b20
      byte type = 2;
      byte pin = 12;
      float offset = 0.0f;
    } indoor;

Я даже попытался установить тип 2 , но это мне не помогло.
opentherm_indoor_temp можно регулировать только вручную, а хотелось-бы с подключенного датчика 
(так скоро освою программирование на старости лет :)))

Laxilef, 2023-10-14:
После сборки из исходников настройки сохраняются?

Balaganoff, 2023-10-14:
> После сборки из исходников настройки сохраняются?

Я не пробовал ( Я просто поправил в исходниках под свои пины и все заработало. 
Буду рядом еще раз попробую и доложу результат.  Ждем новых версий ! Спасибо за работу !

Laxilef, 2023-10-14:
Выложил билды для разных esp8622: https://github.com/Laxilef/OTGateway/releases/tag/1.3.1
Возможно у вас wemos с 4мб памяти, а вы залили билд для wemos 16мб, поэтому и не сохраняются настройки :)
Попробуйте залить firmware_d1_mini.bin или, если не поможет, firmware_d1_mini_lite.bin 

Напишите потом пожалуйста по результатам.

Balaganoff, 2023-10-16:
> Выложил билды для разных esp8266: https://github.com/Laxilef/OTGateway/releases/tag/1.3.1 Возможно у вас wemos с 4мб памяти, а вы залили билд для wemos 16мб, поэтому и не сохраняются настройки :) Попробуйте залить firmware_d1_mini.bin или, если не поможет, firmware_d1_mini_lite.bin
> 
> Напишите потом пожалуйста по результатам.

Да, у меня wemos d1 mini  с 4мб. Пока перезаписать прошивку нет возможности. 
А кроме несохранения настроек различия в объеме памяти еще на что-то влияет?

Laxilef, 2023-10-16:
Нет, не влияет. Но сохранение настроек - весьма важная функция :)

Laxilef, 2023-10-16:
По работе с котлом AEG GBA124 все нормально? Могу в список совместимых добавить?

ananyevgv, 2023-10-17:
Baxi Nuvola-3 необходимо включить ch_2,  иначе горячая вода не работает

Balaganoff, 2023-10-17:
> По работе с котлом AEG GBA124 все нормально? Могу в список совместимых добавить?

Да этот котел все отображает (кроме давления)  и реагирует корректно.

Laxilef, 2023-10-17:
> Baxi Nuvola-3 необходимо включить ch_2, иначе горячая вода не работает

Он всегда включен:
```c++
    localResponse = ot->setBoilerStatus(
      heatingEnabled,
      settings.opentherm.dhwPresent && settings.dhw.enable,
      false, false, true/* <<< ch2 */, false, false
    );
```

closed 2023-11-05 by Laxilef