Laxilef/OTGateway #7: FAQ

Balaganoff, 2023-10-14:
1. Какая логика в подключении датчиков,
  - Экспериментально я выяснил так:
  **External** - это датчик непосредственно подключенный к ESP
  **Boiler**  - ??
  **Manual**  - это внешний датчик, привязываемый через автоматизацию Home Assistant.  Не совсем понят смыл ручной регулировки. Зачем это сделано? Ведь при просмотре со смартфона можно нечаянно установить неверную температуру и нарушить логику работы.

Я правильно понял или где-то можно подробнее прочитать?

Laxilef, 2023-10-14:
External - внешний DS18B20 или аналог, который физически подключен
Boiler - съем с котла по opentherm. Некоторые котлы позволяют подключать датчик внешней температуры или имеют датчик на приточке
Manual - для передачи температуры контроллеру через автоматизацию, например с zigbee датчика,  который висит на улице
Сбить можно,  но автоматизация должна потом все равно установить реальную температуру

Balaganoff, 2023-10-16:
> Manual - для передачи температуры контроллеру через автоматизацию, например с zigbee датчика, который висит на улице
> Сбить можно, но автоматизация должна потом все равно установить реальную температуру.

А зачем это это реализовано ? Если все равно установится реальная температура. Мое - мнение ручная регулировка лишняя и вводит в заблуждение.

Balaganoff, 2023-10-16:
Шаг изменения температуры отопления в 1градус - это слишком грубо. Порог температурной чувствительности человека по разным источникам считается 0.2-0.4 градуса. Измерять мы можем с точностью 0.1. Почему-бы не использовать эту возможность?

В аналогичных теморегуляторах устанавливается с точностью от 0.1 до 0.5 градуса. (зависит от производителя)

Balaganoff, 2023-10-17:
Есть какой-то канал общения кроме гитхаба ? Форум или телеграмм ?

Laxilef, 2023-10-17:
@Balaganoff, пока только тут и на [форуме baxi](https://forum.baxi.ru/viewtopic.php?p=80598)

bigbrother72, 2023-10-19:
> Для climate.opentherm_heating установлен шаг 0.5 градуса.

пытался поменять в publishClimateHeating()

    doc[F("temp_step")] = 0.1;
    doc[F("target_temp_step")] = 0.1;

Не  помогает.

Balaganoff, 2023-10-19:
> Не помогает. Только на один градус шаг регулировки. А мне например 21 градус холодно, а 22 градуса уже жарко

Я для себя нашел выход - через автоматизаию пихаю нужное значение в number.opentherm_heating_target - тогда воспринимает половину градуса и работает по этому значению. Правда на карточке объекта округляет.
Вот пример установки 24.5 и ночной 22. ( такие высокие значения т.к. датчик пока стоит у котла)
[image]

RYZHOP, 2023-10-19:
А можешь скинуть файл .yaml со всеми настроенными датчиками.

Laxilef, 2023-10-19:
Попробуйте [mini-climate-card](https://github.com/artem-sedykh/mini-climate-card), через HACS ставится.
Конфиг:
```yaml
type: custom:mini-climate
entity: climate.opentherm_heating
target_temperature:
  step: 0.5
secondary_info:
  type: hvac-action
fan_mode:
  hide: true
hvac_mode:
  hide: true
  source:
    heat:
      name: Включено
    'off':
      name: Выключено
buttons:
  boost:
    icon: mdi:rocket-launch-outline
    location: bottom
    state:
      attribute: preset_mode
      mapper: 'state => (state == ''boost'' ? ''on'': ''off'')'
    toggle_action: >
      (state, climate_entity) => this.call_service('climate', 'set_preset_mode',
      { entity_id: climate_entity.entity_id, preset_mode:
      (this.toggle_state(state) == 'on' ? 'boost' : 'none') })
  heat:
    icon: mdi:power
    location: bottom
    state:
      mapper: 'state => (state == ''heat'' ? ''on'': ''off'')'
    toggle_action: >
      (state, entity, climate_entity) => this.call_service('climate',
      'set_hvac_mode', { entity_id: climate_entity.entity_id, hvac_mode:
      (this.toggle_state(state) == 'on' ? 'heat' : 'off') })

```

Balaganoff, 2023-10-20:
> Попробуйте [mini-climate-card](https://github.com/artem-sedykh/mini-climate-card), через HACS ставится. 
Да. регулирует с шагом 0.5
Правда на основной карточке округляет, что не очень красиво ;)
[image]

Balaganoff, 2023-10-20:
> ```yaml
> type: custom:mini-climate
> entity: climate.opentherm_heating
> ```

После этого наверное надо добавить:
name: Отопление

Laxilef, 2023-10-20:
В стандартной карточке шаг будет работать в следующей версии.

Rolex48, 2023-11-03:
Подскажите почему минимальный порог кривой "N" с 0.3 ? Есть возможность начинать с 0.1? (дело в том что отопление тёплыми полами котлом без подмеса в диапазоне 25-40°)

"N - heating curve coefficient. The coefficient is selected individually, depending on the insulation of the room, the heated area, etc.
Range: 0.3...10, default: 0.7, step 0.01"

Laxilef, 2023-11-03:
Через HA можете установить любое значение, начиная с 0.001.
В документации исправлю.

filipjurik, 2023-11-17:
Hi,
I have one question. 
Would it be possible to control the modulation level ?

Laxilef, 2023-11-17:
Hi,
@filipjurik no, the modulation is controlled by the boiler software. Openterm supports limiting the modulation level. But not all boilers support this function.

filipjurik, 2023-11-17:
Being able to limit the modulation would help me quite a bit. My boiler starts the flame and modulates to 50% and does not have time to adjust the modulation to maintain the temperature of the water in the heating, so the flame goes out. the boiler cycles. I would need to start it with 10-20% modulation and add it later as needed.How i do this?thakns for you replyFilip ***@***.***ňa 17. 11.

Laxilef, 2023-11-17:
I will try to add this setting in the next version. But, if I’m not mistaken, this parameter can be changed through the boiler’s engineering menu.

filipjurik, 2023-11-18:
Thank you for your response. yes, it is possible to change the maximum allowed heating power in the boiler, but I only need to change it in certain cases. I already have it set to 50%, but I need to temporarily set it to less.

Laxilef, 2023-12-16:
@filipjurik, this feature will appear in release 1.4.0, but now it is already in the master branch. Pls test it.

filipjurik, 2023-12-16:
Thanks, I'll test it and let you know if it works

closed 2024-01-06 by github-actions[bot]