Laxilef/OTGateway #8: Wemos S2 mini (ESP32) support?

mutasim, 2023-11-04:
Does this project support ESP32 boards? I have the DIYLESS adapter and an S2 mini ESP32 board

Laxilef, 2023-11-04:
Didn't check. I think it won't work because used ESP8266Scheduler library

Laxilef, 2023-11-06:
I think I'll add support for esp32 soon

kotjaraxx, 2023-11-08:
> Думаю, скоро я добавлю поддержку esp32. Сейчас у меня нет платы ESP32. Если вы готовы протестировать свою плату, я могу опубликовать прошивку здесь.

Я сегодня вечером получу свой адаптер, готов тестировать

kotjaraxx, 2023-11-08:
Это готовый контролер Smarttherm, контролер используется Esp-wroom-32, но бывает так же на 8266, тот работает на вашей прошивке

kotjaraxx, 2023-11-08:
https://github.com/Evgen2/SmartTherm
Вот проект на гит

Laxilef, 2023-11-08:
Ок. Попробуйте зашить. Пины OT не забудьте настроить и перезагрузить.

[firmware_nodemcu_32s_1.3.3.zip](https://github.com/Laxilef/OTGateway/files/13299148/firmware_nodemcu_32s_1.3.3.zip)

kotjaraxx, 2023-11-09:
> Ок. Попробуйте зашить. Пины OT не забудьте настроить и перезагрузить.
> 
> Из их исходников:
> 
> ```c++
>   int const int inPin = 16; 
>    int outPin = 4; 
> 
>    int DS1820_1 = 15; 
>    int DS1820_2 = 26; ,,/ / D26 esp32 7 снизу
> ```
> 
> [firmware_nodemcu_32s_1.3.3.zip](https://github.com/Laxilef/OTGateway/files/13299148/firmware_nodemcu_32s_1.3.3.zip)

Пока неуспешно, прошилась плата без ошибок, но на связь не выходит теперь, ну тут наверное я где то ошибся. Какие только прошивки не заливал, теперь она, не коннектится к wifi. Вечером сегодня продолжу эксперименты

Laxilef, 2023-11-09:
Странно. Вот билд с выводом всех сообщений в сериал порт. Скиньте логи после загрузки esp.

[firmware_nodemcu_32s_1.3.3_debug.zip](https://github.com/Laxilef/OTGateway/files/13305331/firmware_nodemcu_32s_1.3.3_debug.zip)

kotjaraxx, 2023-11-09:
> Странно. Вот вывод всех сообщений в последовательный порт, а не в телнет. Скиньте логи после загрузки, особенно.
> 
> [прошивка_nodemcu_32s_1.3.3_debug.zip](https://github.com/Laxilef/OTGateway/files/13306150/firmware_nodemcu_32s_1.3.3_debug.zip)

[image]

Laxilef, 2023-11-09:
Это лог процесса прошивки) Нужен лог из сериал порта. Если platformio или arduino ide нет, можно через putty, скорость 115200.

p.s. попробуйте прошиться через https://github.com/marcelstoer/nodemcu-pyflasher/releases
[image]

kotjaraxx, 2023-11-10:
> Это регистрация процесса прошивки) Нужен лог из порта порта. Если Platformio или arduino ide нет, можно через putty, скорость 115200.
> 
> ps форма прошиться через https://github.com/marcelstoer/nodemcu-pyflasher/releases [image]

Command: esptool.py --port COM5 --baud 115200 --after no_reset write_flash --flash_size detect --flash_mode dio 0x00000 C:\firmware_nodemcu_32s_1.3.3.bin --erase-all

esptool.py v3.0
Serial port COM5
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 48:e7:29:ac:ae:7c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 9.5s
Compressed 994752 bytes to 608303...
Wrote 994752 bytes (608303 compressed) at 0x00000000 in 53.7 seconds (effective 148.2 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.

kotjaraxx, 2023-11-10:
> > Это регистрация процесса прошивки) Нужен лог из порта порта. Если Platformio или arduino ide нет, можно через putty, скорость 115200.
> > ps форма прошиться через https://github.com/marcelstoer/nodemcu-pyflasher/releases [image]
> 
> Команда: esptool.py --port COM5 --baud 115200 --after no_reset write_flash --flash_size define --flash_mode dio 0x00000 C:\firmware_nodemcu_32s_1.3.3.bin --erase-all
> 
> esptool.py v3.0 Последовательный порт COM5 Подключение.... Определение типа чипа... Чип ESP32: ESP32-D0WD-V3 (версия 3). Характеристики: WiFi, BT, двухъядерный процессор, 240 МГц, калибровка VRef в efuse, схема кодирования. Нет Crystal с частотой 40 МГц MAC: 48:e7:29:ac:ae:7c Загрузка заглушки... Запуск заглушки... Работа заглушки... Настройка размера флэш-памяти... Автоматически определяется Размер флэш-памяти: 4 МБ Стирание флэш-памяти (это может потребуется некоторое время)... Стирание чипа успешно завершено за 9,5 с. Сжато 994752 байта до 608303... Записано 994752 байта (608303 сжато) по адресу 0x00000000 за 53,7 секунды (эффективная скорость 148,2 кбит/с)... Хэш данных проверен.
> 
> Ухожу... Остаюсь в загрузчике.
> 
> Прошивка успешно прошилась. Отключите/переподключите или перезагрузите устройство , чтобы вернуться в обычный режим загрузки.

[image]

kotjaraxx, 2023-11-10:
Вот что шлет прошитая плата в Arduino IDE

07:31:08.298 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
07:31:08.343 -> invalid header: 0x69646172
07:31:08.389 -> invalid header: 0x69646172
07:31:08.436 -> invalid header: 0x69646172
07:31:08.483 -> invalid header: 0x69646172
07:31:08.530 -> invalid header: 0x69646172
07:31:08.577 -> invalid header: 0x69646172
07:31:08.631 -> invalid header: 0x69646172
07:31:08.671 -> invalid header: 0x69646172
07:31:08.671 -> ets Jul 29 2019 12:21:46

Laxilef, 2023-11-10:
Я думаю у вас что-то с платой или с памятью на плате. Сегодня мне пришла плата nodemcu c модулем еsр-wrооm-32е, прошил через platformio, у меня работает. Но opentherm смогу проверить позже.

Laxilef, 2023-11-10:
Прошу прощения, бинарь не содержал загрузчик и разделы.
Объединил. Шить через [ESP Flash Download Tool](https://www.espressif.com/en/support/download/other-tools) в адрес 0x0. Telnet в этом билде не используется.
[firmware_nodemcu_32s_1.3.3_debug_full.zip](https://github.com/Laxilef/OTGateway/files/13322571/firmware_nodemcu_32s_1.3.3_debug_full.zip)

kotjaraxx, 2023-11-11:
> Прошу прощения, бинарь не содержал загрузчик и разделы. Объединил. Шить можно через тот же NodeMCU PyFlasher или через [ESP Flash Download Tool](https://www.espressif.com/en/support/download/other-tools) в адрес 0x0.
> 
> [build.zip](https://github.com/Laxilef/OTGateway/files/13323960/build.zip)

Спасибо, завтра протестирую, отпишусь о результате)

Laxilef, 2023-11-11:
Ок. Свежий билд.
[build.zip](https://github.com/Laxilef/OTGateway/files/13326242/build.zip)

kotjaraxx, 2023-11-12:
> Ок. Свежий билд. [build.zip](https://github.com/Laxilef/OTGateway/files/13327137/build.zip)

18:03:01.097 -> ets Jul 29 2019 12:21:46
18:03:01.097 -> 
18:03:01.097 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:03:01.097 -> configsip: 0, SPIWP:0xee
18:03:01.097 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:03:01.097 -> mode:DIO, clock div:2
18:03:01.097 -> load:0x3fff0030,len:1184
18:03:01.097 -> load:0x40078000,len:13232
18:03:01.097 -> load:0x40080400,len:3028
18:03:01.097 -> entry 0x400805e4

Результат загрузки последнего релиза

Laxilef, 2023-11-12:
Ошибок нет. Должна появиться точка доступа, подключитесь к ней и дальше по инструкции из readme.

kotjaraxx, 2023-11-14:
> Ошибок нет. Должна появиться точка доступа, подключитесь к ней и дальше по инструкции из readme.

Теперь все работает, ну плата у меня все равно какая то кривая)) Wifi ловит максимум на расстоянии метра от роутера. Разбираюсь теперь с этим

kotjaraxx, 2023-11-15:
> Opentherm корректно работает?
> 
> > Wifi ловит максимум на расстоянии метра от роутера.
> 
> Рядом с антенной (которая на плате) металлических предметов и/или проводов нет?

OpenTherm да, работает, ну о корректности не могу судить пока) Температуру воды и отопления, регулирует. Не видно датчиков температуры внешних, вроде пины корректно выставил в прошивке..

kotjaraxx, 2023-11-15:
Сразу вопрос задам, отключение опентерм как производить? Не отключая питания дёрнул провод, получил ошибку е31, ну по питанию перезагрузил, все заработало. Кнопки управления котлом не работают когда провод подключен, поэтому с кнопки выключить не смог)

Laxilef, 2023-11-15:
Ок, благодарю за информацию.

> Не видно датчиков температуры внешних, единственное что, хотя вроде пины корректно выставил в прошивке..

15 и 26 пины установили? В настройках External поставили?

> Сразу вопрос задам, отключение опентерм как производить? Не отключая питания дёрнул провод, получил ошибку е31, ну по питанию перезагрузил, все заработало. Кнопки управления котлом не работают когда провод подключен, поэтому с кнопки выключить не смог)

Все обесточить и вернуть как было. Программно - никак. Котел без перемычки будет слейвом и управляться через с родной панели вряд ли будет.

kotjaraxx, 2023-11-16:
> External

External выставил, заработал датчик)
Котел у меня Baxi Eco Home 24f, получается что все работает.

kotjaraxx, 2023-11-16:
> > External
> 
> External выставил, заработал датчик) Котел у меня Baxi Eco Home 24f, почему то не работает диагностика
[image]

Laxilef, 2023-11-16:
Скрин не грузится

kotjaraxx, 2023-11-16:
> Скрин не грузится

Поправил

Laxilef, 2023-11-16:
На baxi ampera у меня так же было изначально. С чем связано не разобрался, ни на что не влияет.

kotjaraxx, 2023-11-24:
Я решительно не понимаю что происходит)
У меня оставалась единственная нерешеная проблема, не был виден второй датчик температуры, при этом на родной прошивке производителя моего устройства-все работало.
Пока все это делал, все поломалось, wifi не работал, и прочее. Ну там были проблемы по питанию ESP32, теперь все хорошо с этим, НО. Ваша прошивка теперь не видит ни датчики температуры, ни OT. При этом прошивка производителя железа-работает.
Я не исключаю что я не то прошиваю, можете скинуть последнюю рабочую прошивку для ESP32?

Laxilef, 2023-11-24:
Здравствуйте
1.3.3 была рабочей на esp32. Датчики температуры, однако, не проверял - жду s2 mini.

По обоим датчикам нужно использовать External, если это DS18B20

kotjaraxx, 2023-11-26:
> Здравствуйте 1.3.3 была рабочей на esp32. Датчики температуры, однако, не проверял - жду s2 mini.
> 
> По обоим датчикам нужно использовать External, если это DS18B20

Да, датчики температуры я выставлял, и одме из них работал, до того как я перепрошил его.
Я могу предпринять что то, чтоб вы помогли мне?

kotjaraxx, 2023-11-26:
> Напишите свой логин в telegram.

@AVL87

SimplyKot, 2023-12-06:
Нет, у меня плата собственной разработки на D1 mini . Пробросил в Умный Дом Яндекса, все через колонку тоже отлично управляется.

DeadlySin2, 2024-01-28:
Плата Smartherm, такая же как на фото выше. Почему-то не работает с Baxi Luna-3 комфорт. Светодиод мигает 3 раза, а значит проблема в Opentherm. При этом подключено все внрно, т.к. с родной прошивкой протокол виден и работает. Может я шью как-то неправильно или не той прошивкой? Шил NodeMCU Pyflasher. Прошивка -

Laxilef, 2024-01-28:
@DeadlySin2 прошейте 1.4.0 из мастер ветки.

DeadlySin2, 2024-01-28:
А можно ссылочку или инструкцию. Не могу найти :(

Laxilef, 2024-01-28:
https://github.com/Laxilef/OTGateway/wiki#build-firmware

DeadlySin2, 2024-01-29:
Не смог сделать прошивку. Выбираю Nodemcu32 выбираю Build. Дальше папку built найти не могу, вернее могу, но в папке .pio/built. Видимо я должен прошить Firmware.factory.bin, но после прошивки вместо привычной веб морды вижу The filesystem is not flashed. Видимо я что-то делаю не так.

yanhanxing, 2024-02-18:
> Здравствуйте 1.3.3 была рабочей на esp32. Датчики температуры, однако, не проверял - жду s2 mini.
> 
> По обоим датчикам нужно использовать External, если это DS18B20

Your project is fantastic. I am currently using the ESP32-WROOM version to burn the code for ESP32-S. Currently, during the testing phase, I have not yet connected the wall mounted boiler. During the testing, I encountered the following issues. I am not sure if it is due to the wrong firmware version I have selected or a setup issue? I will provide the firmware for the most commonly used development version of ESP32.
I connected the DS18B20 temperature sensor to the GPIO-18 pin and was able to view the temperature value normally in the OpenTherm gateway. However, displaying the outdoor temperature sensor as unavailable in homeassist.
[image]
[image]
[image]

Laxilef, 2024-02-18:
Hi @yanhanxing 
Thanks! I looked, it's a bug in the code. The temperature entity is not available if there is no connection to opentherm, I will fix this later.

yanhanxing, 2024-02-19:
Okay, I look forward to your revised version and I will continue testing it.

Laxilef, 2024-02-20:
Hi @yanhanxing 
Try [rc15](https://github.com/Laxilef/OTGateway/releases/tag/1.4.0-rc.15)

Laxilef, 2024-02-29:
Hi @yanhanxing 
Try [rc16](https://github.com/Laxilef/OTGateway/releases/tag/1.4.0-rc.16)

closed 2024-04-06 by github-actions[bot]