Laxilef/OTGateway #10: Begginer's frustration...

kulisau, 2023-11-11:
Hi, 
I very recently started using the Home assistant and bought Igor's wemos hat, assembled it and was about to install it and then it clicked that I don't even know how to set the desired room temperature using OpenTherm.
I can't find any quides online for OpenTherm in particular. I was looking at DYILESS guide, but they assume I have an MQQT sensor.
My current setup is a Shelly relay inside of my boiler and Sonoff zigbee thermometer in my living room working in ZHA mode . 
From my understanding generic_thermostat isn't suitable for the OpenTherm as it's just a relay controller. 

Apologies if this is something obvious :)

Laxilef, 2023-11-11:
Hi,
Does your boiler support OpenTerm?
If it is supported, you can flash your ESP8266 with this firmware and the boiler will be integrated into your home assistant.

Just don't forget to install Mosquitto broker :)

kulisau, 2023-11-11:
Yes, it does :) 
My boiler is Radiant R2KA.
I already have flashed the wemos and soldered all the headers needed. 
I do have mosquitto container running. 
My question is how can I setup a card/thermometer to set the desired temperature at the room?

Laxilef, 2023-11-11:
If I understand correctly, then you need to create a "thermostat" card and select the entity "climate.opentherm_heating" there

kulisau, 2023-11-11:
Can you please provide any example how to use that with a zigbee thermometer in mind?

Laxilef, 2023-11-11:
Have you created an automation that transfers temperature from a zigbee sensor to a controller?

kulisau, 2023-11-11:
Not yet. I haven't even connected the OT solution to my boiler as I don't understand how to set the desired inside temp up.

Laxilef, 2023-11-11:
You need to flash your ESP, configure it: set pins and mqtt (server IP, login and password). Then create automation as written in the example and configure the PID or equitherm algorithm in the device settings in HA.

The controller itself will maintain the set temperature in the house by adjusting the coolant temperature.

bigbrother72, 2023-11-12:
Hi
I think you need to disconnect Shelly  relay and connect the Opentherm controller instead.
[image]

After that you must connect to Opentherm contorller and make some settings for MQTT etc
https://github.com/Laxilef/OTGateway#quick-start

[image]

After that in HA go to MQTT intergartions
192.168.xxx.xxx:8123/config/integrations/dashboard

[image]

Then MQTT

[image]

Then Devices

Choose Opentherm controller

[image]

Then add controls to dashboard:
[image]

kulisau, 2023-11-12:
Thank you for your answers!
The biggest question is how to set the desired temperature of the room. Is it possible to use a card like generic_thermostat to set the temperature?
Or maybe this isn't possible at all and the software always tries to keep 23 degrees?

Laxilef, 2023-11-12:
You can change the temperature using the thermostat "climate.opentherm_heating"

bigbrother72, 2023-11-12:
you will see the control elements after completing the last point of my previous message.

like this:
[image]

kulisau, 2023-11-18:
I just now found some time to get back to this...
So I do have another problem. I can flash D1 mini firmware no problem, the device boots ok, I connect to it and setup wifi, so far so good, however, as soon as I set MQQT and pins (D5 out and D6 in) for OT communication device keeps boot looping. I tried different firmwares but to no avail. My board is cheap Wemos D1 mini clone from the Aliexpress. I can confirm the board is  working as I also tried Tasmota firmware and it managed to communicate with the boiler ok.

Laxilef, 2023-11-18:
What firmware file did you flash?

Can you attach a log from the serial port at the time of the cyclic reboot?

kulisau, 2023-11-18:
I tried 1.3.3, 1.3.2 and one older for d1 mini lite and pro versions.
The output is a bit mangled (not really sure which port speed I should use. 115200 seemed the best fit):
`▒▒▒g▒{▒▒g|▒l
▒d`c▒
|{▒d▒o▒▒n▒
l`▒▒;▒d▒l▒
▒
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3424, room 16
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8
tail 0
chksum 0x2b
csum 0x2b
v000703c0
~ld
▒▒▒'▒r▒▒o|▒
l▒
l c▒▒|s▒l▒n▒
▒g▒$`▒▒{▒l▒d▒
`

Laxilef, 2023-11-19:
Hmm, this is strange. Try flashing this build and please send the logs from the serial port here.
[firmware_d1_mini_1.3.4_debug.zip](https://github.com/Laxilef/OTGateway/files/13402706/firmware_d1_mini_1.3.4_debug.zip)

kulisau, 2023-11-19:
Flashed it, configured PINs and MQQT. Serial output:
%▒▒w▒B▒Y.+▒h▒

             )▒\5▒▒PKT- )dУ▒    5Y
                                  )L▒A▒ ▒▒T▒/▒[▒P9▒▒Z%  )l▒@▒▒!NZ       HHi▒9▒D▒ҭ9▒F▒PED ▒▒
                                                                                           rf cal sector: 1020
freq trace enable 0
rf[112] : 0▒

Laxilef, 2023-11-19:
This is not the full log. Without a full log, I can’t understand where to look for the error.

kulisau, 2023-11-19:
I turned off flow control and getting this now:

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3424, room 16
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8
tail 0
chksum 0x2b
csum 0x2b
v00070090
~ld
▒▒▒▒n▒s▒▒o|▒d
▒d`c▒
|;▒d▒g▒▒o▒
d`▒▒s▒$▒l
▒

kulisau, 2023-11-19:
[image]

Laxilef, 2023-11-19:
Ooops. I made a mistake. This build (new) should display all logs.
[firmware_d1_mini_1.3.4_debug.zip](https://github.com/Laxilef/OTGateway/files/13402899/firmware_d1_mini_1.3.4_debug.zip)

kulisau, 2023-11-19:
a step further now...
[image]
I have extremely long MQTT pass, might that be a problem?

Laxilef, 2023-11-19:
Maybe. The password in the config is limited to 32 characters:
```
    char user[32];
    char password[32];
```

kulisau, 2023-11-19:
dang it :D 
About a month or so ago when I installed MQTT and tried to attach a Tasmota switch with short (5-6 chars) pass, Tasmota refused to work. I read somewhere that the pass is suppose to be super long...
I'll give a shorter pass a try and will let you know how that went

kulisau, 2023-11-19:
It connected to MQTT now, however, it still boot looping... 
[image]

Laxilef, 2023-11-19:
I think this is a problem with the library PubSubClient =\
If I'm right, then this build will work.
[firmware_d1_mini_1.3.4_debug.zip](https://github.com/Laxilef/OTGateway/files/13402954/firmware_d1_mini_1.3.4_debug.zip)

kulisau, 2023-11-19:
still the same...

Laxilef, 2023-11-19:
Ok, I need to think. I'll come back here a little later.

kulisau, 2023-11-19:
I re-flashed the board again. This time I haven't changed MQTT settings at all, just changed PIN accordingly and it started boot looping. 
Can it be that some GPIOs are in-use and overlaps with those PINs I describe? 
I use the Ihor's Melnyk master shield. Reboots takes place regardless shield is installed on the Wemos board or not

Laxilef, 2023-11-19:
What pins do you use?

kulisau, 2023-11-19:
Out - 5
In - 6
Internal temp - 7
External temp - I leave as is

Laxilef, 2023-11-19:
But on the wemos d1 mini there is no pin number 6 & 7: https://www.wemos.cc/en/latest/d1/d1_mini.html

Or did you mean D5 and D6?

Laxilef, 2023-11-19:
[image]
D5 = GPIO14
D6 = GPIO12
D7 = GPIO13

kulisau, 2023-11-19:
Yes, I meant D5, D6 and D7. What has to be provided in PIN fields, D or GPIO values?

Laxilef, 2023-11-19:
GPIO

kulisau, 2023-11-19:
I adjusted GPIO values and it no longer reboots! Thanks! 
Will try to connect the shield to the boiler later today.
P.S. I'd suggest using GPIO instead of PIN in the settings to avoid confusion 😀

kulisau, 2023-11-19:
Great success!
[image]

Laxilef, 2023-11-19:
Nice)

kulisau, 2023-11-19:
Something interesting is happening at home 😀 I can't get the boiler to calm down. Initially I turned on the Equitherm and PID and the HA decided to turn my flat into a terrarium (30+ C). I understand it  probably was required for PID calculations. I turned both of these options off. 
I read that a super precise thermometer is required for this PID tuning, right? 
Mine is a sonoff zigbee one with a display. What temperature sensor would you recommend? I heard the xiomi ble could be a good choice.
And one more question. Now that I turned off both of those options I can only select a heating setpoint. Is it still possible to use this opentherm solution as a dumb thermostat until I get a proper temp sensor in my possession?

Laxilef, 2023-11-20:
The maximum temperature can be configured through the entity "number.opentherm_heating_max_temp".  If it is not activated, then it must be activated in the entity settings.

kulisau, 2023-11-20:
It worked. Thanks!

Laxilef, 2023-11-20:
I saw your post regarding the overheating of the house.
Start by setting the equitherm mode. In most cases it regulates better than PID.
There is no need for any precise temperature sensors. You can read how to set up equitherm mode in the readme.

kulisau, 2023-11-20:
Will try that, thanks! 
Btw, I noticed that when DHW is being used, the flame modulation varies between 50-70%, however, when the heating kicks in, it's always 100%. Does it depend on Equitherm/PID settings or something else?
I was under the impression that the OpenTherm should regulate the modulation to keep it as low as possible.

Laxilef, 2023-11-20:
Modulation is controlled by the boiler software

kulisau, 2023-11-21:
First of all, the excel spreadsheet is amazing! I hope I understood it correctly, though 😄 
Not sure what's wrong, but second night in the row the software just gives up during the night. Last night I had to physically pull the plug of the boiler out of the socket because the wemos (it's connected internally inside of the boiler) wasn't responding, this morning I woke up by 17 degrees in the room. Had to turn the turbo heating to bring the temperature to somewhat liveable. I can't figure out why
[image]

Laxilef, 2023-11-21:
Does the controller continue to send data to the mqtt server?  Show graphs "Heating setpoint", "Heating temperature", "Heating" (on/off)

Laxilef, 2023-11-21:
> P.S. I use automations to update internal and external temperatures.

And graphs of external and internal temperatures.

kulisau, 2023-11-21:
Outdoor temp:
[image]
Indoor temp:
[image]
I enabled the heating setpoint only yesterday, hence less data:
[image]
Heating temp:
[image]

Laxilef, 2023-11-21:
This is interesting :)
Although I see that setpoint is correct.

Can you also show a graph of the "binary_sensor.opentherm_heating" entity?

kulisau, 2023-11-21:
binary_sensor.opentherm_heating has no graphs, only log. You're probably looking for this one. Let me know if that's not the case
[image]

kulisau, 2023-11-21:
Any idea why those intermittent dips to 0 takes place in previous graphs?

Balaganoff, 2023-11-21:
У меня тоже такие "прострелы" до нуля на графиках температур как у автора выше ((
Из-за этого увеличивается масштаб и плохо видно реальные изменения. Надо наверное какие-то фильтры-условия при сохранении прописать . 
[image]

Laxilef, 2023-11-21:
Please make such a data sample. This can be done on the "history" tab.
[image]

Balaganoff, 2023-11-21:
> @Balaganoff перезагрузок esp не было? Аптайм посмотрите на вкладке Info.

Да именно тогда и была перезагрузка. Непонятно почему. Все остальное не перезагружалось.

Laxilef, 2023-11-21:
Last reset reason на той же вкладке

kulisau, 2023-11-21:
Here you go:
[image]
I haven't checked my wemos uptime, but there's no point of doing that as I manually had to reboot the whole boiler, but I'll keep an eye on it. In my case I have installed inline power socket (single socket power extension thingy), then plugged iphone charger into it, then USB-A to USB-C cable and just then it connects to the wemos board. I have a feeling that socket is dodgy as I had to wiggle the charger one time for it to work.
It looks like this:
[image]

kulisau, 2023-11-21:
I edited the post with added heating and flame entities. You might have seen the non-updated version.
Which entities you're missing?

Laxilef, 2023-11-21:
I see it, thanks. For some reason, your boiler has stopped heating the coolant, although a command to heat was sent to the boiler.
I will check the code, thank you.

Do you think the boiler could display an error after the ESP was rebooted?

kulisau, 2023-11-21:
Hmm.. I don't think it displays/generates any error. The only think differs on the display when the controller is inaccessible by the boiler is missing "recycling" icon (indicates the OT connectivity) and that's it. The boiler falls back to the manual working mode and restores heating and DHW (temp) settings to the previously set ones. I will monitor the uptime. Thanks

Laxilef, 2023-11-21:
Try to simulate the loss of connection with the ESP.
When the burner is running on the boiler, reboot the ESP and check whether the boiler will heat the water after resetting the ESP.

kulisau, 2023-11-21:
I just tried to restart via the wemos via the browser while it was in action. It took a minute or so and it resumed heating without a hiccup. Another thing I observe is now for whatever reason it does not reach the desired temp at home.

[104 more replies]

closed 2023-12-30 by github-actions[bot]