Laxilef/OTGateway #11: Wishes

Balaganoff, 2023-11-22:
У платы Wemos / ESP осталось много свободных портов. В связи с этим есть такое пожелание на будущее.
1. Подключить по IC2 OLED дисплей на который вывести хотя-бы основные 4цифры - заданные температуры ГВС и отопления и текущие данные. Библиотеки как я понимаю под это дело уже готовы. Ну и из фантастики поуправлять автономно этими цифрами спомощью кнопок  ))
2. Управлять насосами. Хотелось-бы отключать насосы во время остановки обогрева.

Laxilef, 2023-11-22:
Если говорить про esp8266, то не все они безопасны для использования: https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/
И в рамках esp8266, там уже размер кучи (RAM) критично мал, т.к. используются task'и для параллельной работы. Когда остается менее 10кб esp8266 работает не очень стабильно.

Для управления насосами нужны реле или мощные мосфеты, это лучше выносить отдельным блоком с отдельным питанием на какой-нибудь attiny и общаться по i2c/spi с esp8266.
На esp32 можно сделать, но у меня таких задач пока нет, можете добавить PR с реализацией.

Balaganoff, 2023-11-22:
> Для управления насосами нужны реле или мощные мосфеты, это лучше выносить отдельным блоком ...... из HA с помощью автоматизации вкл/выкл их, отслеживая статус parameters.heatingEnabled.

Связка разных блоков через HA  делает систему более нестабильной.  ИМХО роль HA должна быть в дополнительном сервисе, анализе и регулирования. А все остальное блок должен делать автономно.  В конце концов я почти уверен "переезд" с 8266 на esp32 не станет "камнем преткновения" ни в финансовом, ни в техническом вопросе, - ни в первом, ни во втором сложностей не должно возникнуть.

Balaganoff, 2023-11-22:
А у светодиода L3 "Status" какая логика работы? 
Я лампочки не задействовал и может это то что нужно мне для управления дополнительным насосом ?

Laxilef, 2023-11-22:
https://github.com/Laxilef/OTGateway#leds

Balaganoff, 2023-11-22:
> https://github.com/Laxilef/OTGateway#leds
> 
> Я могу добавить пин, который будет подтянут к питанию, когда котёл должен греть теплоноситель. Но не оч рекомендую без опторазвязки туда что-то подключать.

Это было-бы здорово!
А по поводу развязки - все реализовано внутри недорогих твердотельных реле на семисторах
[image]

Laxilef, 2023-11-22:
Ок. Но вам нужно будет собрать прошивку самостоятельно через platformio.

Laxilef, 2023-11-22:
Добавил в мастер ветку.

Сборка в вашем случае будет с доп build_flags: -D HEATING_STATUS_PIN=НОМЕР_GPIO, т.е. например:
```yaml
[env:d1_mini]
platform = ${esp8266_defaults.platform}
board = d1_mini
lib_deps = ${esp8266_defaults.lib_deps}
lib_ignore = ${esp8266_defaults.lib_ignore}
extra_scripts = ${esp8266_defaults.extra_scripts}
build_flags = 
	${esp8266_defaults.build_flags}
	-D OT_IN_PIN_DEFAULT=4
	-D OT_OUT_PIN_DEFAULT=5
	-D SENSOR_OUTDOOR_PIN_DEFAULT=12
	-D SENSOR_INDOOR_PIN_DEFAULT=14
	;-D LED_STATUS_PIN=13
	;-D LED_OT_RX_PIN=15
	-D USE_TELNET=1
	-D HEATING_STATUS_PIN=13
```

Balaganoff, 2023-11-23:
> Ок. Но вам нужно будет собрать прошивку самостоятельно через platformio.

Я думаю, что смогу это сделать )  Спасибо! Буду пробовать.

Laxilef, 2023-12-06:
@Balaganoff добавил нормальное управление внешним насосом. На странице сетап новые параметры:
* Use external pump
* External pump GPIO
* External pump post circulation time (время в секундах)
* External pump anti stuck interval (время в секундах)
* External pump anti stuck time (время в секундах)

Скомпилируйте из мастер ветки, проверьте у себя. После обновления сбросятся настройки, сохраните их перед обновлением.

Balaganoff, 2023-12-07:
> @Balaganoff добавил нормальное управление внешним насосом. На странице сетап новые параметры:
...
> Скомпилируйте из мастер ветки, проверьте у себя. После обновления сбросятся настройки, сохраните их перед обновлением.

Спасибо ! Вечером попробую!

Laxilef, 2023-12-07:
Проверьте режим выбега насоса после отключения отопления и режим антизалипания. Для проверки тайминги можно временно установить меньше.

Balaganoff, 2023-12-07:
> Проверьте режим выбега насоса после отключения отопления и режим антизалипания. Для проверки тайминги можно временно установить меньше.

Выбег и интервальные запуски нормально работают! НО! Прошивка получилась нестабильна. Через время появляются перезапуски - это я увидел по графику температуры, а потом устройство вообще пропадает из сети и свою точку не поднимает. При этом котел видит связь и даже происходит регулирование. Помогает только ручной перезапуск.
Включу debug режим и попробую поймать этот момент в терминале

[image]

Laxilef, 2023-12-07:
У себя тоже наблюдаю перезагрузки. Похоже, что из-за телнета. Попробуйте собрать прошивку без него и понаблюдайте.

Balaganoff, 2023-12-07:
Все ! Без включенного "Debug" завис через несколько минут

Laxilef, 2023-12-07:
Поправил одну мелочь, может быть поможет. Проверьте у себя пожалуйста.

Balaganoff, 2023-12-07:
> Поправил одну мелочь, может быть поможет. Проверьте у себя пожалуйста.

ок ! Телнет включить при сборке ?

Laxilef, 2023-12-07:
Нужно через сериал порт смотреть что там происходит.
Займусь на выходных.

Laxilef, 2023-12-09:
Проверьте пожалуйста у себя с учетом последних коммитов.

closed 2023-12-30 by github-actions[bot]