Laxilef/OTGateway #14: Request: Send MSG ID24 Room Temperature and ID27 Outside Temperature

aurelmarius, 2023-11-28:
Hello,
Do you think is possible to send Room Temperature and Outside Temperature from HA to opentherm ? I'm running nodoshop's opentherm gateway in conjuction with your OTGateway and it supports ID24 and 27 from master device.
Thanks.

Laxilef, 2023-11-28:
Hi,
Yes, ID 23 and 24 can be sent. But why is this necessary?
[image]

ID57 must be sent:
https://github.com/Laxilef/OTGateway/blob/ead8c64e925812ef27b3dc7517a04b07b2e89b5b/src/OpenThermTask.h#L216

Laxilef, 2023-11-28:
I think this can be added.
P.S. What kind of display?

aurelmarius, 2023-11-28:
It is a waveshare epaper 7.5 screen hooked up to esp32.
If it is too hard to implement this, i will eventualy flash the screen with esphome and get sensor data from homeassistant.

closed 2023-12-24 by Laxilef