Laxilef/OTGateway #16: Report temperature from forecast doesn't work

SimplyKot, 2023-12-07:
Use this instruction:
https://github.com/Laxilef/OTGateway/wiki#reporting-outdoor-temperature-from-home-assistant-weather-integration

On saving integration get error:
"Message malformed: Entity {{ source_entity }} is neither a valid entity ID nor a valid UUID for dictionary value @ data['entity_id'"

Attributes is present:
[image]

Делаю по инструкции и после копирования и редактирования скрипта автоматизации получаю ошибку:
"Message malformed: Entity {{ source_entity }} is neither a valid entity ID nor a valid UUID for dictionary value @ data['entity_id'"

При этом атрибуты в объекте weather.home есть.

Версия HA core 2023.2.2

Laxilef, 2023-12-07:
Обновил

SimplyKot, 2023-12-07:
Сохранился нормально. 
При ручном запуске срабатывает норамльно, но при автоматическом запуске не выполнятеся.
Спотыкается о 
`condition:
  - condition: template
    value_template: "{{ has_value(source_entity) and has_value(target_entity) }}"`

С формулировкой 
Error: In 'template' condition: UndefinedError: 'has_value' is undefined

Трассировка:
[image]

Laxilef, 2023-12-07:
Какая версия HA? Функция has_value появилась с 2023.4

SimplyKot, 2023-12-07:
Версия HA core 2023.2.2
На праздниках обновлюсь. Спасибо.

closed 2023-12-07 by SimplyKot

Laxilef, 2023-12-07:
has_value можно заменить временно на 
```
source_entity not in ['0', 'unavailable']
```
Аналогично в других участках кода.

Laxilef, 2023-12-07:
обновил, вернул совместимость с предыдущими версиями HA