Laxilef/OTGateway #18: In some conditions OTGateway reports 0 for setpoint and outdoor temperature

SimplyKot, 2023-12-08:
In some conditions OTGateway reports 0 for setpoint and outdoor temperature

При некоторых условиях с контроллера проходят нули в параметрах "Установленная темпервтура" и  "Вненшяя температура".
[image]

SimplyKot, 2023-12-11:
Прошивка 1.3.3
Данные о внешней темературе переправлены с gismeteo
Ошибка наблюдается при первом вклбчении шлюза.

Кроме того, есть проблема в определении досупности шлюза.

При потери wifi переходит в аварийный режим с предустановленной температурой (что просто отлично), но  в HA остается активным.
[image]
Тут в 15-00 отлючен роутер.

При этом термостат отображается активным и показывает постоянный нагрев.
[image]

SimplyKot, 2023-12-11:
Провел тестирование на отсутствие связи и на отсутствие питания
[image]

Вывод - OpenTherm Status - всегда выдает ОК, хотя Status определяет состояние правильно.

Laxilef, 2023-12-11:
Спасибо! Поправлю.

Laxilef, 2023-12-15:
Исправлено в мастер ветке, проверьте.

closed 2024-01-12 by Laxilef