Laxilef/OTGateway #20: Opentherm Gateway Heating

strk90, 2023-12-09:
Подскажите как установить целевую температуру 23,0 °C?
На термостате можно установить минимальную температуру 35°C.
В регулировках термостата выведена температура теплоносителя,  а не регулировка температуру indoor?
  
(https://github.com/Laxilef/OTGateway/assets/119970190/9fda951b-f219-4945-a931-cf3f69fa9acc)

Laxilef, 2023-12-09:
Как и любой другой термостат, он регулирует температуру теплоносителя. Для расчета температуры теплоносителя для поддержки желаемой температуры в доме используется ПЗА (погодозависимое регулирование, equitherm) или PID.
Если ни один из этих режимов не включен, то температуру теплоносителя нужно устанавливать самостоятельно.
Почитайте quick start в wiki.

strk90, 2023-12-09:
Поскажите как задать желаемую температуру в доме? 
Нужно какой-то дополнительный термостат написать?

Laxilef, 2023-12-09:
Включить и настроить Equitherm, настроить датчики внешней и внутренней температуры и установить желаемую температуру в термостате в home assistant.
[image]

strk90, 2023-12-09:
Все я так сделал. 
Почему-то у меня целевой параметр, который отвечает за желаемую температуру, устанавливает желаемую температуру теплоносителя. 
Может проблема в моем котле? 
Модель котла baltur tesis 24.

[image]

strk90, 2023-12-09:
> Включить и настроить Equitherm, настроить датчики внешней и внутренней температуры и установить желаемую температуру в термостате в home assistant. <img alt="" width="40%" src="https://private-user-images.githubusercontent.com/34578544/289313022-1869d4a7-a70f-4353-8d69-1201590aafad.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIxNTc4OTcsIm5iZiI6MTcwMjE1NzU5NywicGF0aCI6Ii8zNDU3ODU0NC8yODkzMTMwMjItMTg2OWQ0YTctYTcwZi00MzUzLThkNjktMTIwMTU5MGFhZmFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA5VDIxMzMxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg3ZjAyYzA4ZWM2ODk3MzRjZDY1YWU2YWYwMDMyZTMwNDgzMWI0ZjdmMDE5NTkyOTE4ZTM5YWZjYzdjZTk2YmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IfTqsV36oxp0kGf-B2KlSo3gKqx5V_WLltm1GE5HE_M"> <img alt="" width="40%" src="https://private-user-images.githubusercontent.com/34578544/289312904-48d8e64e-26c9-4a30-adbe-0c3eb6d46bf3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDIxNTc4OTcsIm5iZiI6MTcwMjE1NzU5NywicGF0aCI6Ii8zNDU3ODU0NC8yODkzMTI5MDQtNDhkOGU2NGUtMjZjOS00YTMwLWFkYmUtMGMzZWI2ZDQ2YmYzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjA5VDIxMzMxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNjNWRkZTk4ZjBjNTc3MjgyMDM1NjI1NDFhMmY0OTc3Zjg3MWI2ZDFkMjkxNGZhMmU5N2Y4N2U1MDMzZTJiNmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.1A8wg61UIAH_H-fPD5nsp5Bl_bWizoUmp6prXtURr1I">

Буду разбираться.
Спасибо!

Laxilef, 2023-12-09:
Странно, после включения Equitherm через некоторое время конфигурация термостата ``climate.opentherm_heating`` должна была обновиться, в т.ч. нижний и верхний предел устанавливаемых температур. Попробуйте esp перезагрузить.
Версия 1.3.3 прошита?

strk90, 2023-12-09:
Да, версия 1.3.3.
Попробую прошить другую esp.

Laxilef, 2023-12-09:
Это последняя стабильная версия, всё нормально.
Попробуйте esp перезагрузить. После перезагрузки вся конфигурация сущностей должна обновиться.

strk90, 2023-12-10:
Заработало! 
Спасибо!

closed 2023-12-10 by strk90