Laxilef/OTGateway #21: Domusa Bioclass HM 16

Laxilef, 2023-12-13:
I have assembled a DIYLESS Thermostat shield with a D1 Mini and installed version 1.3.3

It works with my Domusa Bioclass HM 16 pellet boiler. I have left the master ID with value 0 (default)

Almost everything works, DHW, correct. Diagnostics, fault messages, etc. ok. But I don't see the % modulation, my old Tado showed me that value. Is there any way to see it? I'm interested to see if the modulation works well. I don't see the value of pressure either.

On the other hand, wouldn't it be more correct to show the HW controller in HA as water_heater entity and not as climate?

_Originally posted by @MikelillOi in https://github.com/Laxilef/OTGateway/issues/3#issuecomment-1851397379_

Laxilef, 2023-12-13:
@MikelillOi I think we should continue in another issue.
You can listen to the communication between the boiler and thermostat using the firmware monitor from Ihor Melnyk: https://github.com/ihormelnyk/opentherm_library/blob/master/examples/OpenThermGatewayMonitor_Demo/OpenThermGatewayMonitor_Demo.ino

Your boiler may be using non-specific IDs for communication.

MikelillOi, 2023-12-14:
Thank you for open a new issue.

[image]

Here you have 4 sensor history graphic. Heater, DHW, Flame and modulation.

In my old Tado, there was only one option to configure OpenTherm, which makes me think that the boiler will use the predefined message to indicate the modulation.

Another strange thing, as we saw, is that the "heating" sensor turns off and on, and the boiler doesn't really stop.

At the afternoon, I will make somo test with OTMonitorDemo, and compiling the latest build of yout project.

Laxilef, 2023-12-14:
> Another strange thing, as we saw, is that the "heating" sensor turns off and on, and the boiler doesn't really stop.

Try turning on "Modulation sync with heating"

Laxilef, 2023-12-19:
I think you should start by listening to the communication between the boiler and the Tado thermostat.

MikelillOi, 2023-12-20:
> I think you should start by listening to the communication between the boiler and the Tado thermostat.

Can I do it with Tasmota or other firmware?  So I don't have to buy more hardware

Laxilef, 2023-12-20:
No, the slave needs another board.

closed 2024-01-09 by github-actions[bot]