Laxilef/OTGateway #23: Предложения по работе с PID

bigbrother72, 2023-12-15:
Здравствуйте

У меня включена работа по ПЗА и ПИД.

По ПЗА температура выставляется достаточно грубо в зависимости только от температуры внешней. Не учитывается ветер, влажность итд.  Т.е. у меня в доме например при -25 градусах на улице и без ветра температура в СО может быть 60-65 градусов для температуры в комнате где стоит термосенсор.
А если с ветром - то и +75 можно загонять в батареи, чтобы было дома +21 градус.  Поэтому более точная регулировка осуществляется через PID.

В ваш код добавил немного модификаций.

1. Для того, чтобы PID не вычислялся каждые несколько секунд, нужно ввести задержку в несколько минут, так как процесс сильно инерционный и D коэффициент не отрабатывает, если вызывать слишком быстро расчет ПИД . Я у себя поставил раз в три минуты вызов вычисления пид .

В функции
```
  float getPidTemp(int minTemp, int maxTemp) {
    pidRegulator.Kp = settings.pid.p_factor;
    pidRegulator.Ki = settings.pid.i_factor;
    pidRegulator.Kd = settings.pid.d_factor;

// задержка для вычисления пид 3 минуты
    pidRegulator.setDt(180000);

......

//заменяем getResultNow на getResultTimer
    //return pidRegulator.getResultNow();
    return pidRegulator.getResultTimer();
```

И в src\MqttTask.h и в src\HomeAssistantHelper.h надо расширить диапазон возможных значений для P,I,D.
У меня например P=50, I=0.006, D=10000

[image]

[image]

Laxilef, 2023-12-15:
Изучу, спасибо!

Laxilef, 2023-12-15:
pidRegulator.setDt(180000) не сработает, он принимает тип signed short, куда влезает [-32767, +32767]. Надо в репо открыть issue, чтобы поменяли тип на unsigned int

Laxilef, 2023-12-15:
> По ПЗА температура выставляется достаточно грубо в зависимости только от температуры внешней. Не учитывается ветер, влажность итд. Т.е. у меня в доме например при -25 градусах на улице и без ветра [...]

А вы комнатный датчик не используете? ПЗА же корректирует температуру теплоносителя в т.ч. и по комнатному датчику.

bigbrother72, 2023-12-16:
> pidRegulator.setDt(180000) не сработает, он принимает тип signed short, куда влезает [-32767, +32767]. Надо в репо открыть issue, чтобы поменяли тип на unsigned int

да. забыл про это написать. я уже месяц примерно назад это сделал.
там тоже поставил другой тип.
Может быть даже поставить изменяемый через интерфейс HA параметр - Интервал подсчета ПИД. А то у каждого он может быть разный. Я начинал от 10 минут, потом 5 минут. Остановился на трех минутах - за это время есть какая-то осмысленная разница между предыдущим значением и текущим и проходит достаточно времени, чтобы ПИД понять мог.

bigbrother72, 2023-12-16:
> А вы комнатный датчик не используете? ПЗА же корректирует температуру теплоносителя в т.ч. и по комнатному датчику.

Комнатный датчик на esphome у меня стоит в самой холодной комнате. Через автоматизацию в HA потом передаю в opentherm контролер. Насколько понял эта температура не учитывается в эквитермических функциях. Только наружная учитывается.

Laxilef, 2023-12-16:
> Насколько понял эта температура не учитывается в эквитермических функциях. Только наружная учитывается.

Очень даже учитывается. Коэф. Т.

Laxilef, 2023-12-20:
@bigbrother72 добавил в мастер ветку возможность редактирования Dt и, надеюсь временно, заменил библиотеку на форк для корректной работы с Dt свыше 32 секунд. Протестируйте у себя.

bigbrother72, 2024-01-03:
вы добавили определение
GyverPID<int, unsigned long> pidRegulator(0, 0, 0);

а надо 
GyverPID<float,  unsigned long> pidRegulator(0, 0, 0);

иначе в целых числах там идет расчет

commit e38bda6b (2024-01-12): fix: calculating pid temperature in float, fixed #23
closed 2024-01-12 by Laxilef