Laxilef/OTGateway #24: Master branch doesn't compile for ESP8266

aurelmarius, 2023-12-17:
I'm trying to compile the master branch for esp8266 d1 mini but it is failing. Do you have any idea what i'm doing wrong ?

`Building in release mode
Compiling .pio/build/d1_mini/src/main.cpp.o
Generating LD script .pio/build/d1_mini/ld/local.eagle.app.v6.common.ld
Compiling .pio/build/d1_mini/libbfb/OpenTherm Library/OpenTherm.cpp.o
Compiling .pio/build/d1_mini/libe72/PubSubClient/PubSubClient.cpp.o
Compiling .pio/build/d1_mini/lib469/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pio/build/d1_mini/lib469/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pio/build/d1_mini/lib469/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pio/build/d1_mini/lib469/ESP8266WiFi/ESP8266WiFiAP.cpp.o
Compiling .pio/build/d1_mini/lib469/ESP8266WiFi/ESP8266WiFiGeneric.cpp.o
src/main.cpp:16:6: error: #elif with no expression
 #elif
      ^
In file included from src/main.cpp:27:0:
src/MainTask.h:68:10: error: #elif with no expression
     #elif
          ^
.pio/libdeps/d1_mini/PubSubClient/src/PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
.pio/libdeps/d1_mini/PubSubClient/src/PubSubClient.cpp:523:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     return (rc == expectedLength);
                   ^
In file included from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson.hpp:33:0,
                 from .pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson.h:9,
                 from src/main.cpp:3:
.pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp: In constructor 'ArduinoJson::V700PB2::JsonDocument::JsonDocument(ArduinoJson::V700PB2::JsonDocument&&)':
.pio/libdeps/d1_mini/ArduinoJson/src/ArduinoJson/Document/JsonDocument.hpp:33:51: error: no matching function for call to 'ArduinoJson::V700PB2::JsonDocument::JsonDocument()'
   JsonDocument(JsonDocument&& src) : JsonDocument() {
                     `

[image]

Laxilef, 2023-12-17:
Fixed. Thanks!

aurelmarius, 2023-12-18:
Thanks a lot.

closed 2023-12-19 by Laxilef