Laxilef/OTGateway #25: Gas Consumption over OpenTherme

martinarva, 2023-12-21:
I have Vitodens 100W gas heater and just have set up OTGateway. Great work and thank you for that. I have both gas heater and heat-water pump and now i'm able to automate between them so i can head my house most cost effective way.

Opening my gas heater to smart world made me think that is there a way to get gas consumption information from Vitodens over opentherm?

Laxilef, 2023-12-21:
Hi, thanks!
The boiler does not report gas consumption. You can try to calculate the gas consumption using the modulation level.

martinarva, 2023-12-21:
> Hi, thanks! The boiler does not report gas consumption. You can try to calculate the gas consumption using the modulation level.

Modulation is an option yes, but i don't have a clue how much it consumes in different levels.

Laxilef, 2023-12-21:
In the specification for the boiler you can see the gas consumption at 100% modulation :)

martinarva, 2023-12-22:
> In the specification for the boiler you can see the gas consumption at 100% modulation :) If the values are linear, then you can create a sensor in Home Assistant.

great idea, i'll look into it

closed 2023-12-22 by martinarva

martinarva, 2023-12-23:
@Laxilef , can i pick your brain a little bit?

It was not easy to find information about gas consumption of my gas heater but i found somehting:

[image]

So do i understand correctly that in 100% modulation consumption is 3.31 m2/h?

Laxilef, 2023-12-23:
Yes, it looks like it
Check this with your gas meter. You can look at the consumption per minute at 100% modulation and multiply this value by 60.
To check linearity, repeat this for modulation 75, 50, 25%

martinarva, 2024-02-21:
> Hi, I can't find out to create a sensor on HA for gas consumption through modulation level. The only solution I've found for now is this, but the values are not right...Are there other solutions? thanks
> 
> * platform: template
>   sensors:
>   consumo_gas:
>   unique_id: consumo_gas
>   friendly_name: "Consumo Gas"
>   unit_of_measurement: "m³/h"
>   value_template: >
>   {% set modulazione = states('sensor.opentherm_modulation_level') | float %}
>   {% set tabella = {
>   100: 3.60, 99: 3.58, 98: 3.55, 97: 3.50, 96: 3.46,
>   95: 3.47, 94: 3.42, 93: 3.37, 92: 3.33, 91: 3.30,
>   90: 3.27, 89: 3.28, 88: 3.25, 87: 3.18, 86: 3.14,
>   85: 3.10, 84: 3.05, 83: 3.07, 82: 3.04, 81: 3.00,
>   80: 2.97, 79: 2.93, 78: 2.90, 77: 2.86, 76: 2.83,
>   75: 2.79, 74: 2.76, 73: 2.72, 72: 2.68, 71: 2.65,
>   70: 2.62, 69: 2.58, 68: 2.55, 67: 2.51, 66: 2.48,
>   65: 2.43, 64: 2.41, 63: 2.37, 62: 2.34, 61: 2.30,
>   60: 2.28, 59: 2.25, 58: 2.20, 57: 2.16, 56: 2.13,
>   55: 2.09, 54: 2.06, 53: 2.02, 52: 1.99, 51: 1.95,
>   50: 1.92, 49: 1.88, 48: 1.85, 47: 1.81, 46: 1.78,
>   45: 1.74, 44: 1.71, 43: 1.67, 42: 1.64, 41: 1.64,
>   40: 1.57, 39: 1.54, 38: 1.50, 37: 1.47, 36: 1.43,
>   35: 1.43, 34: 1.36, 33: 1.33, 32: 1.30, 31: 1.27,
>   30: 1.27, 29: 1.23, 28: 1.15, 27: 1.12, 26: 1.09,
>   25: 1.06, 24: 1.06, 23: 1.03, 22: 0.94, 21: 0.91,
>   20: 0.88, 19: 0.85, 18: 0.85, 17: 0.81, 16: 0.73,
>   15: 0.70, 14: 0.67, 13: 0.63, 12: 0.63, 11: 0.59,
>   10: 0.52, 9: 0.48, 8: 0.44, 7: 0.40, 6: 0.37,
>   5: 0.33, 4: 0.35, 3: 0.28, 2: 0.31, 1: 0.23,
>   0: 0.00
>   } %}
>   {% set modulazione_arrotondata = modulazione | round(0) %}
>   {{ tabella[modulazione_arrotondata] }}

I did it like this:

`#Gas heater consumption and cost

  - platform: template
    sensors:
      current_gas_consumption:
        friendly_name: "Current Gas Consumption"
        unit_of_measurement: 'W'
        value_template: >-
          {% set modulation_level = states('sensor.opentherm_modulation_level') %}
          {% set max_power = 3310 %}
          {% if modulation_level not in ['unknown', 'unavailable', 'none'] and modulation_level | float >= 0 %}
            {{ (max_power * modulation_level | float / 100) | round(0) }}
          {% else %}
            0
          {% endif %}

  - platform: template
    sensors:
      current_gas_cost:
        friendly_name: "Current Gas Cost"
        unit_of_measurement: 'EUR'
        value_template: >-
          {% set price_per_kwh = states('sensor.total_gas_price_kwh_vat') %}
          {% set current_power = states('sensor.current_gas_consumption') %}
          {% if price_per_kwh not in ['unknown', 'unavailable', 'none'] and current_power not in ['unknown', 'unavailable', 'none'] %}
            {{ (price_per_kwh | float * current_power | float / 1000) | round(2) }}
          {% else %}
            0
          {% endif %}`
          
First sensor calculates the current gas consumption in watts (W) based on the modulation level of the gas heater. It uses the OpenTherm modulation level as a reference to determine the current power output, with a maximum power parameter set at 3310W, which is my heater max on full load (perumably 100% modulation). 

Complementing the consumption sensor, the Current Gas Cost sensor provides an estimation of the current cost of gas consumption in euros (EUR). It calculates the cost by taking the current gas consumption data and the price per kilowatt-hour (kWh), including VAT, to provide a real-time estimation of gas expenses. 

But seems that this method does not work and consumption is off. Here is comparison with my gas provider numbers:
[image]

Laxilef, 2024-02-21:
Template:
```yaml
{% set level = 3 %}
{% set matrix = {
100: 3.60,
95: 3.47, 90: 3.27,
85: 3.10, 80: 2.97,
75: 2.79, 70: 2.62,
65: 2.43, 60: 2.28,
55: 2.09, 50: 1.92, 
45: 1.74, 40: 1.57,
35: 1.43, 30: 1.27,
25: 1.06, 20: 0.88,
15: 0.70, 10: 0.52,
5: 0.33, 1: 0.23,
0: 0.00
}
%}
{% set ns = namespace(consumption = 0, prevValue = 0, prevModLevel = 0) %}
{% if level in matrix %}
  {% set ns.consumption = matrix[level] %}
{% else %}
  {% for modLevel in matrix|sort(reverse = False) %}
    {% set value = matrix[modLevel] %}
    
    {% if ns.consumption == 0 %}
      {% if level > modLevel %}
        {% set ns.prevValue = matrix[modLevel] %}
        {% set ns.prevModLevel = modLevel %}
      {% else %}
        {% set dividend = value - ns.prevValue %}
        {% set divisor = modLevel - ns.prevModLevel %}
        {% set delta = level - ns.prevModLevel %}
        {% set ns.consumption = (delta * dividend) / divisor + ns.prevValue %}
      {% endif %}
    {% endif %}
  {% endfor %}
{% endif %}

{{ ns.consumption|round(2) }}
```

vespadrift, 2024-02-21:
@Laxilef, it works, thank you!

martinarva, 2025-02-14:
Old topic, but i can see that there is a power entity in the newer versions
[image]

@Laxilef , can you plese tell what it is and can I use it to calculate the gas consumption?