Laxilef/OTGateway #27: Hot Water start and stop

martinarva, 2023-12-23:
I have set domestic hot water to 46 degrees. It stopped heating on 49.5 degrees
[image]

In my heat pump i can set heating start and stop. For example start heating when it's under 42 and stop at 46.

What is the logic here with this? When I select 46 on the thermostat in HA, what are the start and stop degrees in OTGateway and Gas Heater?

Laxilef, 2023-12-23:
When you set 46 degrees, the controller simply send this value to the boiler. The boiler itself decides when to turn on and off DHW heating.

blitzu, 2023-12-27:
I have noticed a strange behavior: if the set temperature for domestic hot water (DHW) is lower than the set temperature for heating supply, when the set DHW temperature is exceeded, the flame stops because the DHW temperature has reached its limit. Normally, the heating of the thermal agent should continue until the set room temperature is reached.

Laxilef, 2023-12-27:
The burner is controlled by the boiler, not the controller. However, on a boiler, DHW most often has a higher priority than heating if DHW is turned on (DHW tap is open).

blitzu, 2023-12-27:
> The burner is controlled by the boiler, not the controller. However, on a boiler, DHW most often has a higher priority than heating if DHW is turned on (DHW tap is open).

the behavior appears with tap closed

Laxilef, 2023-12-28:
@blitzu hmm, that's strange. Which version are you using?

closed 2024-01-17 by github-actions[bot]