Laxilef/OTGateway #28: Monitoring mode only ?

0BuRner, 2023-12-23:
Hello,
Just wanted to know if it's possible to avoid sending any data to the data bus, but only listening on it ?
I don't want to interfere with my manual wireless thermostat that comes with my boiler...

Laxilef, 2023-12-24:
Hi,
No, it's not implemented, but I was thinking about it

0BuRner, 2024-01-02:
Do you think it will require lot of work ? 
If you already had an idea on how to do it, maybe you can describe here what's your vision on how to implement that and I can give it a try in my free times

Laxilef, 2024-01-02:
It is not difficult. But this requires additional hardware, which I don’t have: a slave shield and a thermostat.

I think it should work like this:
* a thermostat must be connected to the slave shield
* the boiler must be connected to the master shield

The controller receives commands from the thermostat and sends them to the boiler, having received a response from the boiler, the controller sends it to the thermostat. This can be done in the ``OpenthermTask``.

However, I wouldn't want to just do a transparent proxy of commands. We need to leave the option to use PID and Equitherm modes. To do this, you need to filter the commands received from the thermostat.

Thus, if one of these modes is enabled on the controller, having received a command from the thermostat to set the temperature, we must either ignore this command or set this value as the desired room temperature.

7oxicshadow, 2024-01-11:
+1 on this. I have the same gateway board as @0BuRner and it sounds like we have the same goal. To be able to monitor the opentherm traffic between a thermostat and boiler without interference from the arduino interface.

I will also take a look at the source to see if I can understand how Tx and Rx are working.

closed 2024-01-31 by github-actions[bot]

0BuRner, 2024-04-13:
Hi,
Any news on this ? I looked a bit to the code the other day but unfortunately I'm not comfortable enough with it to be able to make such change...

Laxilef, 2024-04-13:
Hi,
No news. I do not have a thermostat and slave shield for testing.

7oxicshadow, 2024-04-13:
@0BuRner I have created a basic project which seems stable over the past couple of months usage. Its a bit bare bones and was not designed with flexibility in mind but might be good enough as a reference project for you.

I am afraid it is customised to my boiler (baxi)  / stat (ESi) combination

https://github.com/7oxicshadow/openThermGateway_HA