Laxilef/OTGateway #29: Start heating with HA automation instead of temp control

martinarva, 2024-01-07:
I hav now used OTGetaway for few weeks with equitherm logic and it works well.

However I have floor heating and using Uponor Smatrix collector control solution which is quite smart and can predict thermal inertia etc.

When collectors are in closed state it’s does not have a point for gas heater to be on as the hot water is not flowing in the under floor heating.

So I would like to turn on gas heater only when any of the floor heating contours are in open state. I have this information in Home Assitant. For example one:
[image]

Idle means off and heating means on. I can group all contours together so that if any of them is on I can make automation to turn on the gas heater.

But now I need some idea for a solution to turn gas heater on/off but at same time retain the heating temp vs outside temp logic. So basically use only heating curve and gas heater on/off will be controlled by automation.

martinarva, 2024-01-07:
Here are all my collectors and as you can se the need for heat is quite different
[image]

at the moment only living room temp is used in OTgateway.  

Would be nice to switch heating on from any of the collectors instead

martinarva, 2024-01-07:
workaround for time being.

I made a template sensor that will show state "heating" if any of the collectors are heating and "idle" if none. Also added "error" state if collectors are for some reason unavailable.
`  - platform: template
    sensors:
      overall_hvac_action:
        friendly_name: "Collectors Overall HVAC Action"
        value_template: >
          {% set error_states = ['unknown', 'unavailable'] %}
          {% if is_state('sensor.alumine_korrus_hvac_action', 'heating') 
             or is_state('sensor.saun_hvac_action', 'heating') 
             or is_state('sensor.garaaz_hvac_action', 'heating') 
             or is_state('sensor.2_korruse_trepihall_hvac_action', 'heating') 
             or is_state('sensor.vannituba_hvac_action', 'heating') 
             or is_state('sensor.magamituba_hvac_action', 'heating') 
             or is_state('sensor.marloni_tuba_hvac_action', 'heating') 
             or is_state('sensor.mathiase_tuba_hvac_action', 'heating') %}
            heating
          {% elif is_state('sensor.alumine_korrus_hvac_action', error_states) 
             or is_state('sensor.saun_hvac_action', error_states) 
             or is_state('sensor.garaaz_hvac_action', error_states) 
             or is_state('sensor.2_korruse_trepihall_hvac_action', error_states) 
             or is_state('sensor.vannituba_hvac_action', error_states) 
             or is_state('sensor.magamituba_hvac_action', error_states) 
             or is_state('sensor.marloni_tuba_hvac_action', error_states) 
             or is_state('sensor.mathiase_tuba_hvac_action', error_states) %}
            error
          {% else %}
            idle
          {% endif %}`

When collectors are heating i raise the climate.opentherm_heating to 25 degrees, so it's quite sure heating kicks in. When idle, I lower climate.opentherm_heating to 20 degrees, so its quite sure heating will stop. When error i put it to 22 degrees so that normal heating will continue. Also added a trigger for home assistant restart to check what to do.

`alias: Gas Heater Collectors Automation
description: Turn on/off gas heater from collectors states
trigger:
  - platform: state
    entity_id:
      - sensor.overall_hvac_action
    to: heating
    id: heating
  - platform: state
    entity_id:
      - sensor.overall_hvac_action
    to: idle
    id: idle
  - platform: state
    entity_id:
      - sensor.overall_hvac_action
    to: error
    id: error
  - platform: homeassistant
    event: start
    id: restart
condition: []
action:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - heating
        sequence:
          - service: climate.set_temperature
            metadata: {}
            data:
              temperature: 24
            target:
              entity_id: climate.opentherm_heating
      - conditions:
          - condition: trigger
            id:
              - idle
        sequence:
          - service: climate.set_temperature
            metadata: {}
            data:
              temperature: 21
            target:
              entity_id: climate.opentherm_heating
      - conditions:
          - condition: trigger
            id:
              - error
        sequence:
          - service: climate.set_temperature
            metadata: {}
            data:
              temperature: 22
            target:
              entity_id: climate.opentherm_heating
          - service: notify.mobile_app_iphone
            metadata: {}
            data:
              title: HVAC
              message: Collectors Error, switching Gas Heater to 22c
      - conditions:
          - condition: trigger
            id:
              - restart
        sequence:
          - choose:
              - conditions:
                  - condition: state
                    entity_id: sensor.overall_hvac_action
                    state: heating
                sequence:
                  - service: climate.set_temperature
                    metadata: {}
                    data:
                      temperature: 25
                    target:
                      entity_id: climate.opentherm_heating
              - conditions:
                  - condition: state
                    entity_id: sensor.overall_hvac_action
                    state: idle
                sequence:
                  - service: climate.set_temperature
                    metadata: {}
                    data:
                      temperature: 19.8
                    target:
                      entity_id: climate.opentherm_heating
              - conditions:
                  - condition: state
                    entity_id: sensor.overall_hvac_action
                    state: error
                sequence:
                  - service: climate.set_temperature
                    metadata: {}
                    data:
                      temperature: 22
                    target:
                      entity_id: climate.opentherm_heating
                  - service: notify.mobile_app_iphone
                    metadata: {}
                    data:
                      title: HVAC
                      message: Collectors Error, switching Gas Heater to 22c
mode: single
`

This should cover my needs for now. I also changed Equitherm factors to quite minimum (k=0.1, t=0.1), as my collectors will handle the on/off and heatcurve N to 0.2 which shoul be enough looking at the heat curve. If needed i will raise it.

Laxilef, 2024-01-23:
Try this:
1. Send the temperature inside the house to the controller from this sensor:
```yaml
- platform: min_max
  name: min_temperature_rooms
  unique_id: min_temperature_rooms
  type: min
  entity_ids:
    - sensor.room1
    - sensor.room2
    - sensor.room3
    - sensor.room4
    - sensor.room5
    - sensor.room6
```
2. Send to the controller the value of this sensor as the desired temperature:
```yaml
- platform: min_max
  name: max_desired_temperature_rooms
  unique_id: max_desired_temperature_rooms
  type: max
  entity_ids:
    - sensor.room1_setpoint
    - sensor.room2_setpoint
    - sensor.room3_setpoint
    - sensor.room4_setpoint
    - sensor.room5_setpoint
    - sensor.room6_setpoint
```

Of course, the code needs to be modified to suit your config. But I think the algorithm is clear. In this algorithm, you should not use very cold rooms such as a garage.

closed 2024-02-13 by github-actions[bot]