Laxilef/OTGateway #30: Wifi Watchdog

blitzu, 2024-01-12:
Hello, is possible to implement an watchdog for wifi or maybe for mqtt connection? If it cant't connect to mqtt for some time the board should reboot.

Laxilef, 2024-01-12:
Hi. What for? The board will connect automatically when possible.
In addition, this will disrupt the emergency mode.

blitzu, 2024-01-13:
Because is second time when my device get offline. It doesn't connect back even if restarted my wifi router. The signal is high. The device reconnected after power cycle

Laxilef, 2024-01-13:
Version 1.3.3?

blitzu, 2024-01-13:
V 1.4.0-rc 5

Laxilef, 2024-01-13:
Can you test this on version 1.4.0-rc.9?

aurelmarius, 2024-01-13:
> Can you test this on version 1.4.0-rc.9?

Compiled 1.4.0-rc.9, can't connect to wifi hotspot anymore. I had to manually reflash esp with older version.
[image]

Laxilef, 2024-01-13:
Oh, I forgot to say that starting from version rc9 it is necessary to flash the file system via platformio.

aurelmarius, 2024-01-13:
> Oh, I forgot to say that starting from version rc9 it is necessary to flash the file system via platformio.

There is separate file for fs or flash just bin via platformio ?

Laxilef, 2024-01-13:
[image]

aurelmarius, 2024-01-14:
> [image]
> 
> 

Future versions after 1.4.0-rc.9 will support ota ?

Laxilef, 2024-01-14:
Yes. There is already an "Upgrade" page there.

aurelmarius, 2024-01-15:
[image]
Up and running. Thanks for the info.

aurelmarius, 2024-01-15:
After 4 hours it became unavailable and boiler reported F95 error (no ot connection). Running on d1_mini. Moved back to rc5 for now. I also noticed i get random reboots with rc5 but not hang ups.
[image]

aurelmarius, 2024-01-16:
> Can you save the logs on the latest version and upload them here? I'm interested in what happens at the moment of freezing and after.

Sure, but i don’t know how to get the logs.

Laxilef, 2024-01-16:
You need to connect the ESP to your computer via USB, connect to the port using a serial client and configure recording to a file.
[image]

Laxilef, 2024-01-16:
I also noticed that some users have similar problems when using DHCP. Can you try using a static ip for your esp?

closed 2024-02-06 by github-actions[bot]