Laxilef/OTGateway #31: Intergrating TRV's

nekocentral, 2024-01-16:
At the moment i'm looking at replacing my Tado thermostat with the OTGateway mainly as i want to be able to intergrate heating a bit more with zigbee TRV's, do you have some idea's on how to do this or i can just make a heating call using home assistant but ignore the temp sensor?

Laxilef, 2024-01-17:
Zigbee trv are not good regulators as they rely on a temperature sensor that is installed in them. The temperature near radiators will always be higher than in the rest of the room.
To delicately regulate the temperature in the rooms, I use thermal heads installed on the collector group. These thermal heads are connected to relays, and the relays are controlled by the ``generic_thermostat`` component in the home assistant.
In this config, thermal heads and OTGateway are not connected to each other. OTGateway still controls the temperature in the house, but the thermal heads allow you to slightly reduce or increase the temperature in individual rooms.

nekocentral, 2024-01-17:
Interesting
I was thinking of using 2 sensors per room the trv and a secondary sensor on the other side. And maybe make it so I can use the average of them to trigger heating on or off. Or even set a temperature offset calculated of the second one.

Laxilef, 2024-01-17:
You have to try. But I was unable to get any of the zigbee trvs available on Aliexpress to work properly.

nekocentral, 2024-01-17:
> You have to try. But I was unable to get any of the zigbee trv available on Aliexpress to work properly.

I'm going to try the sonoff trv's just need to see if I can tell my radiator to just keep pumping heat untill I say stop

Laxilef, 2024-01-17:
Despite the convenience of using zigbee trv, I would recommend using electric thermal heads with relays.
I used zigbee trv in my apartment for about 2 years, but they did not maintain the normally set temperature and I had to change the batteries 2 times during the winter. If there are only 2-3 radiators in the house, then this is probably acceptable, but when there are 10 or more, it’s tiring :)

nekocentral, 2024-01-17:
> Despite the convenience of using zigbee trv, I would recommend using electric thermal heads with zigbee relays.
> I used zigbee trv in my apartment for about 2 years, but they did not maintain the normally set temperature and I had to change the batteries 2 times during the winter. If there are only 2-3 radiators in the house, then this is probably acceptable, but when there are 10 or more, it’s tiring :)
> 
> > I'm going to try the sonoff trv's just need to see if I can tell my radiator to just keep pumping heat untill I say stop
> 
> Almost no zigbee trv support the “dumb” control mode. They all try to maintain the set temperature. You can see more details on the [zigbee2mqtt page](https://www.zigbee2mqtt.io/supported-devices/#s=trv).

I need to check if I can even properly do that due to the design of the heating in my house,
All rooms atm have a radiator with a complete dumb valve and I can't sadly run power to all of them across multiple floors.

Laxilef, 2024-01-17:
Hmm. Do you have a collector group installed?
[image]

nekocentral, 2024-01-18:
> Hmm. Do you have a collector group installed?
> [image]
> 

Not exactly[image]

This is how the lines from my boiler leave
One is to all radiators, other one return
And the only lvl of controll per room I have is this
[image]

Laxilef, 2024-01-18:
In this case, it makes sense to try zigbee trv. But you shouldn’t expect precise regulation from them :)
p.s. It seems you have taps installed, not valves. They may not fit the thermal heads.

closed 2024-02-07 by github-actions[bot]