Laxilef/OTGateway #33: Можно ли использовать как замену термостату?

DeadlySin2, 2024-01-18:
В данный момент на котле Baxi Luna-3 Comfort стоит панель (штатная на котле) и HomeAssistant через умную розетку и реле с сухим контактом управляет термостатом в котле. 
Логика простая - если никакая батарея не греет в текущий момент - терморегулятор выключается, если какая-либо включена - терморегулятор включается.
Есть несколько вопросов: 
1. Возможно ли управлять термостатом оборудованием с данной прошивкой? 
2. Если поставить OTGGateway будет ли работать штатная панель или ее необходимо будет отключить?
3. Будет ли работать OTGGateway совместно с терморегулятором (как сейчас, только информация по работе котла будет в НА)

DeadlySin2, 2024-01-19:
У меня по другому не получится. В разьем на котле вставлен штекер в который подключена панель. Тут или параллельно подключать блок или отключать панель и подключать блок. 
Я думаю отключить панель, чтобы что-нибудь не случилось...

Laxilef, 2024-01-19:
На сайте зонта есть [схема подключения](https://www.tvp-electro.ru/wp-content/uploads/2023/05/baxi-luna-3-tsifrovoe-podklyuchenie-smart-i-smart-2.0-dineco-1.jpg) вашего котла. Тут она тоже применима.

DeadlySin2, 2024-01-19:
Спасибо. Читал, у меня немного по другому но разберусь. Прибор заказал буду экспериментировать. Спасибо еще раз

alexkoh1, 2024-01-22:
А будет ли возможность управлять котлом прямо из веб интерфейса прошивки, без ha? :) Задать целевые значения, посмотреть статус.

alexkoh1, 2024-01-22:
> Если у вас есть возможность, можете создать отдельную страницу dashboard с необходимыми полями для настройки всех значений и открыть PR. Все параметры, которые можно менять через mqtt, можно менять и через http api.

А можете примерно подсказать в какую сторону ковырять? :) Возможно справлюсь!

Laxilef, 2024-01-22:
В data лежат html файлы страниц, в src_data/app.js есть готовые функции загрузки данных и функции сохранения форм, в PortalTask.h handler'ы страниц.

closed 2024-02-11 by github-actions[bot]