Laxilef/OTGateway #35: Invalid values for number.opentherm_pid_p and number.opentherm_pid_d

Pajuris, 2024-01-30:
Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:194
Integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: 07:38:02 (382 occurrences)
Last logged: 10:07:49

    Invalid value for number.opentherm_pid_p: 50.0 (range 0.001 - 10.0)
    Invalid value for number.opentherm_pid_d: 10000.0 (range 0.0 - 10.0)

Device: https://diyless.com/product/opentherm-thermostat
Firmware: firmware_nodemcu_32s_1.4.0-rc.5
Boiler Wiessmann Vitodens 100-W B1KC 26 kW

I use 1.4.0 firmware because 1.3.3 firmware does not have Summer/winter mode, Heating CH1 to CH2 and Modulation sync with heating options. Without these options my boiler does not respond to setting the heating temperature.

Laxilef, 2024-01-30:
This has been fixed in the latest RC. Build the firmware [from sources](https://github.com/Laxilef/OTGateway/wiki#build-firmware).

Pajuris, 2024-01-30:
Buid, install... but something is wrong. Can not connect to OpenTherm Gateway with password otgateway123456.

Laxilef, 2024-01-30:
Why? Did you flash the filesystem?

Pajuris, 2024-01-30:
I dont know why :) Yes, I flashed system with the new firmware and after reboot device lost connection to my wifi and appears wifi name: OpenTherm Gateway.  And I can not connect to OpenTherm Gateway with password otgateway123456. (WIndows or Mac show, the password is wrong).

Laxilef, 2024-01-30:
In addition to the firmware, you need to flash the file system. Has it been done?

Pajuris, 2024-01-30:
Ups... no... sorry... :)

Pajuris, 2024-01-30:
file system image is littlefs.bin?
Error: Detected overlap at address: 0x37 for file .... /littlefs.bin

Pajuris, 2024-01-30:
Flashed file system image via PlatformIO.
Blogadariu za pomasc, Laxilef. :)

DeadlySin2, 2024-01-31:
Can confirm this issue. The only way to flash the filesystem file is to use PlatformIO, anothe flashers send an error message

Laxilef, 2024-02-01:
documentation fixed.