Laxilef/OTGateway #38: delfis ctfs 24" nova florida no heating control

andrey150960, 2024-02-04:
Приветствую. Мой котел "delfis ctfs 24" nova florida,  фирмы "findital". Плата esp32 прошивка 1.4.0-rc.12 PlatformIO. 
При подключении почти все параметры получаю. Управление ГВС работает, а управление отоплением нет. Т.е. не работает термостат. Пробовал играться с настройками, как рекомендовано https://github.com/Laxilef/OTGateway/wiki/FAQ-&-Troubleshooting#how-can-i-activate-deactivated-sensors-and-settings-in-my-home-assistant. Ни каких результатов
[image]
[image]
[image]
При подключении к котлу ESP32, на самом котле регулировка ГВС не регулируется , а температура в помещении поддается регулированию в ручную.
Что можете посоветовать  для исправления

Laxilef, 2024-02-04:
1. Если отключить PID и Equitherm и попробовать установить температуру 50 градусов, например, котёл как-то реагирует?
2. Запишите логи через telnet или serial порт.

andrey150960, 2024-02-04:
Нет ни какой реакции, живет своей жизнью.
Вот чего получаю, с отключенными PID и Equitherm и вкл. Equitherm
[00:10:08][OT][HEATING][WARN] Failed set max modulation 100%
[00:10:11][OT][HEATING][INFO] Set temp = 35
[00:10:24][OT][DHW][INFO] Set temp = 41
[00:10:25][OT][DHW][WARN] Failed set ch2 temp
[00:11:10][REGULATOR][INFO] New target: 50.00
[00:11:10][OT][HEATING][INFO] Set temp = 50
[00:11:12][OT][HEATING][WARN] Failed set max modulation 100%
[00:11:26][OT][DHW][INFO] Set temp = 41
[00:11:27][OT][DHW][WARN] Failed set ch2 temp
[00:12:06][SETTINGS][INFO] Updated
[00:12:12][OT][HEATING][INFO] Set temp = 50
[00:12:14][OT][HEATING][WARN] Failed set max modulation 100%
[00:12:27][OT][DHW][INFO] Set temp = 41
[00:12:28][OT][DHW][WARN] Failed set ch2 temp
[00:13:13][OT][HEATING][INFO] Set temp = 50
[00:13:17][OT][HEATING][WARN] Failed set max modulation 100%
[00:13:28][OT][DHW][INFO] Set temp = 41
[00:13:29][OT][DHW][WARN] Failed set ch2 temp
[00:14:14][OT][HEATING][INFO] Set temp = 50
[00:14:20][OT][HEATING][WARN] Failed set max modulation 100%
[00:14:29][OT][DHW][INFO] Set temp = 41
[00:14:30][OT][DHW][WARN] Failed set ch2 temp
[00:15:00][REGULATOR][INFO] New target: 22.00
[00:15:00][OT][HEATING][INFO] Set temp = 39
[00:15:01][OT][HEATING][INFO] Disabled
[00:15:01][OT][HEATING][WARN] Failed set max modulation 100%
[00:15:32][OT][DHW][INFO] Set temp = 41
[00:15:32][OT][DHW][WARN] Failed set ch2 temp

Laxilef, 2024-02-05:
Судя по логам, CH2 у вас там либо не используется, либо котел запрещает установку. А вот температура отопления устанавливается нормально. Странно, что котёл не реагирует.
Стоит попробовать перебрать Master Member ID от 1 до 255, возможно в прошивке залочено управление отоплением только под определенные термостаты. Из флагов можно оставить только DHW present.

andrey150960, 2024-02-05:
Спасибо попробую перебирать ID. Прикладываю telnet c различными режимами.
Странно при включении PID показывает  [HEATING][INFO] Set temp = 78, а при вкл.  Equitherm эти показания не вижу.
Пробовал прошить ESP8266 [1.4.0-rc.13] не шьется, а [v1.3.3] на ура.
[новый 1.txt](https://github.com/Laxilef/OTGateway/files/14160748/1.txt)

andrey150960, 2024-02-05:
Вроде разобрался , заработало. По наблюдаю несколько дней , отпишусь.

Laxilef, 2024-02-05:
> Вроде разобрался , заработало.

Вы хоть напишите что сделали и что помогло.

> Подскажите как вытащить параметр "Modulation:" на панель Home Assistant , т.е. каким запросом в MQTT

Объект с уровнем модуляции и так существует, посмотрите карточку устройства в HA.

andrey150960, 2024-02-05:
Как говорят " Ларчик открывался просто. Включаем котел , заходим в режим программирования. Нажатием одновременно 
"Reset" и "минус отопления", устанавливаем P17-TSP17 в режим "1" (термостат/или ПДУ). Выходим из программирования. Выключаем котел. Отключаем на плате котла перемычку между  "1" и "2". Подключаем  "open therm" к контактам "3" и "4".
Подаем питание на "open therm" и после этого включаем котел (это важно). На экране котла появляется надпись " Con".
Все, котел управляется "open therm".

> Объект с уровнем модуляции и так существует, посмотрите карточку устройства в HA.

не увидел 
[image]

closed 2024-02-05 by andrey150960

andrey150960, 2024-02-06:
> > Еще бы насос отключался и была бы "песня"
> 
> При использовании Equitherm и/или PID и гистерезиса контроллер отключает отопление при достижении заданной комнатной температуры по opentherm (но в HA это не видно). При отключении отопления котел должен отключить насос, но это работает не со всеми котлами.

Да спасибо, Котел работает с отключением насоса, просто нужно было  подождать пока выйдет на режим. Пока настраивал t упала до 19 градусов, а нужно 23.