Laxilef/OTGateway #39: sensor.opentherm_dhw_flow_rate unavailable

Pajuris, 2024-02-05:
Newest firmware: 1.4.0-rc.13

Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:324
Integration: MQTT (documentation, issues)
First occurred: 16:06:20 (4 occurrences)
Last logged: 16:10:24

Error 'expected SensorDeviceClass or one of 'date', 'enum', 'timestamp', 'apparent_power', 'aqi', 'atmospheric_pressure', 'battery', 'carbon_monoxide', 'carbon_dioxide', 'current', 'data_rate', 'data_size', 'distance', 'duration', 'energy', 'energy_storage', 'frequency', 'gas', 'humidity', 'illuminance', 'irradiance', 'moisture', 'monetary', 'nitrogen_dioxide', 'nitrogen_monoxide', 'nitrous_oxide', 'ozone', 'ph', 'pm1', 'pm10', 'pm25', 'power_factor', 'power', 'precipitation', 'precipitation_intensity', 'pressure', 'reactive_power', 'signal_strength', 'sound_pressure', 'speed', 'sulphur_dioxide', 'temperature', 'volatile_organic_compounds', 'volatile_organic_compounds_parts', 'voltage', 'volume', 'volume_storage', 'water', 'weight', 'wind_speed' for dictionary value @ data['device_class']' when processing MQTT discovery message topic: 'homeassistant/sensor/opentherm/dhw_flow_rate/config', message: '{'availability': {'topic': 'opentherm/status'}, 'enabled_by_default': False, 'unique_id': 'opentherm_dhw_flow_rate', 'object_id': 'opentherm_dhw_flow_rate', 'entity_category': 'diagnostic', 'device_class': 'volume_flow_rate', 'state_class': 'measurement', 'unit_of_measurement': 'L/min', 'name': 'DHW flow rate', 'icon': 'mdi:water-pump', 'state_topic': 'opentherm/state', 'value_template': '{{ value_json.sensors.dhwFlowRate|float(0)|round(2) }}', 'expire_after': 120, 'device': {'identifiers': ['opentherm'], 'sw_version': '1.4.0-rc.13', 'model': 'OpenTherm Gateway', 'name': 'OpenTherm Gateway', 'configuration_url': 'http://192.168.1.242/'}, 'platform': 'mqtt'}'

Laxilef, 2024-02-05:
On which version of Home Assistant does this occur?

Pajuris, 2024-02-05:
Latest...  core 2024.1.6

Laxilef, 2024-02-05:
This will [start working](https://github.com/home-assistant/core/commit/cece117c93c35ecdf0f7c1624cb6db6d445b57a1) with release 2024.2

Pajuris, 2024-02-07:
Thank's. Now working...

closed 2024-02-07 by Pajuris