Laxilef/OTGateway #40: Permanent connection to the BLE sensor, fast discharge

Laxilef, 2024-02-07:
Hi @mennodegraaf!
I received information from users that in the current implementation, the connection to BLE sensors is permanent. Because of this, the device's battery quickly drains.

Users have provided an example [implementation in OpenMQTTGateway](https://github.com/1technophile/OpenMQTTGateway/blob/development/main/ZgatewayBLEConnect.ino), which seems to use a slightly different method for polling sensors. Do you think it is possible to port this implementation?

mennodegraaf, 2024-02-08:
Not sure yet, how different the implementation is, but I can have a look.

Laxilef, 2024-02-10:
If I understand correctly, OpenMQTTGateway uses a subscription to events (messages) from BLE sensors, rather than connecting and reading the value. Are you using a BLE sensor currently? How's it going with the battery drain?

Laxilef, 2024-02-20:
Implemented in rc15. Closed.

closed 2024-02-20 by Laxilef