Laxilef/OTGateway #41: Can't get the correct values ​​for the Heating temp and setpoint

vespadrift, 2024-02-08:
Good evening everyone, 
I can't get the correct values ​​for the Heating temperature and setpoint. I try to change the Master MemberID code without result...
I have a Radiant r2k and i use the esp8266 with the OpenTherm Adapter by https://github.com/ihormelnyk
[image]

And on home assistant I can't adjust the temperature under 30°...but the value can be set between 30 and 80°
[image]
Any solution? thanks

Laxilef, 2024-02-08:
If you are not using PID or equitherm mode, then everything is correct. In your case, you can set the temperature from 30 to 80 degrees.

vespadrift, 2024-02-08:
Thank you so much! It works!

closed 2024-02-08 by vespadrift