Laxilef/OTGateway #42: Request: exhaust temperature

Pajuris, 2024-02-08:
My  gas boiler: Viessmann Vitodens W-100 B1KC 26 kW has exhaust temperature sensor. Would be great to see this information in opentherm gateway...

Laxilef, 2024-02-10:
Where can this be useful? I didn't add this sensor because I don't see where it can be used.

Pajuris, 2024-02-10:
It's important for gas consumption. If exhaust temperature is high (50-55 C or more), condensing gas boiler not working in condensing mode.

Laxilef, 2024-03-29:
Hi @Pajuris 
Pls test it

Pajuris, 2024-03-30:
Hello
Installed 1.4.0-rc.18m but no new sensors appears... I must compile manually?

Laxilef, 2024-03-30:
Yes, you need to compile from source.

> And... my previos version was 1.4.0-rc.14. So it worked much better. RC.18 version runs much slower:
> for example modulation value appears only ~30-40 seconds after real flame turn on.
> I could see DHW flow rate 5-7 l/min and flame on, but modulation show 0% another 20-30 seconds.

Can you upload the logs?

Pajuris, 2024-03-31:
OK, compiled and installed... 
OpenTherm Gateway Exhaust temperature and OpenTherm Gateway Heating return temperature appears.
OpenTherm Gateway Heating return temperature always show 0.
OpenTherm Gateway Exhaust temperature show wrong values 0.1-0.2 C
What logs do you need? From where I must take it?

Laxilef, 2024-03-31:
> OpenTherm Gateway Exhaust temperature show wrong values 0.1-0.2 C

Fixed it in the last commit. Please recompile the firmware and test it now.

> What logs do you need? From where I must take it?

Connect to ESP using telnet or serial port and save the logs.

Pajuris, 2024-03-31:
OK, Exhaust temperature seems to work corectly.
Logs attached.
[ZOC2403_Telnet Connection Example0109.log](https://github.com/Laxilef/OTGateway/files/14817249/ZOC2403_Telnet.Connection.Example0109.log)
[ZOC2403_Telnet Connection Example0107.log](https://github.com/Laxilef/OTGateway/files/14817250/ZOC2403_Telnet.Connection.Example0107.log)

Laxilef, 2024-03-31:
Thanks for the tests. Looking at the logs, the data is updated without delay. I don’t see random INVALID or TIMEOUT response statuses. Except ID 28, it is not supported by your boiler.

> for example modulation value appears only ~30-40 seconds after real flame turn on.
> I could see DHW flow rate 5-7 l/min and flame on, but modulation show 0% another 20-30 seconds.

In fact, the modulation can be 0% and this does not mean that the flame does not burn. 0% is the minimum level. It would have been more correct to make the minimum level 1%, but the manufacturers decided otherwise.

Pajuris, 2024-03-31:
When DHW turns on (mixer is full open), boiler modulation when I look on boiler display, is more than 50 % and changes. But modulation in HA show 0 % and does not changes about 30 seconds.
Anyway, I will do more tests tomorrow.

Laxilef, 2024-03-31:
Try looking at these parameters on the portal page, the data is updated there every 10 seconds

Pajuris, 2024-03-31:
On portal page modulation is 0 % too. When portal start showing modulation value, after few secons value appears and on HA.

Laxilef, 2024-03-31:
Well, I don’t know why the boiler can send data with such a delay. In the logs you can see the "raw" data of communication between the ESP and the boiler. In the response, you should be interested in the first 16 bits.

Pajuris, 2024-04-02:
OK... 1.4.0-rc.14 version modulation value appears with delay too. It's Viessmann boiler problem.
Thanks Laxilef for new, exhaust value sensor. Good job.

closed 2024-04-02 by Laxilef