Laxilef/OTGateway #43: delay в библиотеках DallasTemperature и OpenTherm

BCometto, 2024-02-09:
В библиотеках **DallasTemperature** и **OpenTherm** присутствуют вызовы функции **delay()**, которые подвешивают исполнение всего кода. В **DallasTemperature** функция **delay()** вызывается в **blockTillConversionComplete()** для ожидания выполнения расчетов датчиком и, в зависимости от **bitResolution**, составляет от 94 до 750 миллисекунд. В **OpenTherm** функция **delay()** вызывается в **activateBoiler()** на 1 секунду. Не знаю насколько это критично в Вашем проекте, но я при использовании библиотеки **DallasTemperature** избавлялся от этого **прописав в функции millisToWaitForConversion() нули во всех return и используя библиотеку SimpleTimer** для задания периодичности чтения данных с датчика. Хотя, можно тоже самое сделать установив **waitForConversion = false** и **checkForConversion = false** в функции **setOneWire()** и использовать таймер для считывания показаний с датчика.

Laxilef, 2024-02-10:
Здравствуйте.
В коде используется ``setWaitForConversion(false)``:
https://github.com/Laxilef/OTGateway/blob/1a031172578cdbce7b52d7b954fd1dadd11e17d8/src/SensorsTask.h#L22C1-L24C53

и неблокирующее ожидание:
https://github.com/Laxilef/OTGateway/blob/1a031172578cdbce7b52d7b954fd1dadd11e17d8/src/SensorsTask.h#L174C1-L190C61

В opentherm некоторые тайминги должны строго соблюдаться, поэтому целесообразно использовать блокирующее ожидание.

BCometto, 2024-02-10:
Спасибо! Замечательный проект!

closed 2024-03-01 by github-actions[bot]