Laxilef/OTGateway #53: Losing connection with boiler

shaggee, 2024-02-29:
Im using 1.4.0-rc.15 on Wemos D1 mini with Termet Ecocondens 25 Integra II boiler. The connection works well but after couple hours, the connection with boiler is lost.

Manually restart OTGateway - no results
Manually power cycle OTGateway - no results
Manually power cycle boiler - connection established again

How can I check the connection logs? Any advice to resolve this problem?

[image]

Laxilef, 2024-02-29:
Connect via telnet and save logs when the problem occurs.
This may be due to the boiler settings or the wiring. What is the length of the wire? What wire is used? Are there power wires near the opentherm wire?

shaggee, 2024-03-01:
> Connect via telnet and save logs when the problem occurs. This may be due to the boiler settings or the wiring. What is the length of the wire? What wire is used? Are there power wires near the opentherm wire?

The wire is arround 30cm length. Yes the are power cables near opentherm wire. I will move the board and wires and test it again.

Laxilef, 2024-03-08:
Hi @shaggee 
Did you find the problem?

shaggee, 2024-03-08:
> Hi @shaggee Did you find the problem?

What I did:
1. Shortened the cable to 15cm. The cable is a simple 2x1,5mm2 speaker cable.
2. Moved the Wemos D1 and OT adapter away from electric cables

After this upgrades, im still losing the connection

Im recording the logs since yesterday 14:00 . So far the connection is stable. Strange.

[image]

shaggee, 2024-03-13:
Strange. I saved the telnet logs for 4 days without any connection lose.
Today I turned off logging and tadam. After 10 hours connection losed:

[00:01:22][MQTT][TRACE] Successfully publish 0 of 0 bytes to topic: opentherm/state/set
[00:01:22][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:01:22][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  4   Status: TIMEOUT
[00:01:23][MQTT][TRACE] Successfully publish 474 of 474 bytes to topic: opentherm/state
[00:01:24][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  5   Status: TIMEOUT
[00:01:24][OT][WARN] Invalid response after setBoilerStatus: TIMEOUT
[00:01:25][MAIN][VERB] Free heap size: 25056 of 81920 bytes (min: 20776, diff: 0), max free block: 22296 (min: 15016, diff: 0, frag: 11%)
[00:01:26][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  1   Status: TIMEOUT
[00:01:27][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  2   Status: TIMEOUT
[00:01:28][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  3   Status: TIMEOUT
[00:01:29][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  4   Status: TIMEOUT
[00:01:30][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  5   Status: TIMEOUT
[00:01:30][OT][WARN] Invalid response after setBoilerStatus: TIMEOUT
[00:01:32][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  1   Status: TIMEOUT
[00:01:33][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  2   Status: TIMEOUT
[00:01:34][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  3   Status: TIMEOUT
[00:01:36][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  4   Status: TIMEOUT
[00:01:37][OT][TRACE] ID:    0   Request:     1200   Response:        0   Attempt:  5   Status: TIMEOUT
[00:01:37][OT][WARN] Invalid response after setBoilerStatus: TIMEOUT
[00:01:38][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:01:38][MQTT][TRACE] Successfully publish 474 of 474 bytes to topic: opentherm/state

Seems I need to set up the server for 24/7 telnet logging to get it 100% work.

blitzu, 2024-03-13:
Try to reflash firmware, I had the same problem

shaggee, 2024-03-13:
> Try to reflash firmware, I had the same problem

I tried. It doesnt help

Laxilef, 2024-03-14:
I think it is necessary to save the log at the moment of connection loss. Perhaps sending some request breaks the openterm on the boiler.

shaggee, 2024-03-14:
Okay give me one week to record the logs 24/7.

closed 2024-04-11 by github-actions[bot]