Laxilef/OTGateway #54: Request: Multiple PID based Climates for multi-room purposes?

GravitySandwichh, 2024-02-29:
I currently heat all my rooms seperately with often different set temps. Having a single climate entity would cause OpenTherm to sometimes throw my heater at 100% because the living room temperature is low, but the room i'm actually heating is only 0.5c off from where it needs to be. Having multiple climates that check each room individually together with their assigned temperature sensors would solve this issue. Is this something that could be implemented? So far i have not seen a single solution that makes it possible.

blitzu, 2024-02-29:
I have trv in every room. I made an node red script that checks every climate if it needs heat. If multiple climate need heat I find the lowest current temperature and the highest set point and send that values to opentherm as setpoint and interior temperature. If none of my climate need heat I send 15 degree as setpoint and 30 degree as current temperature.

Laxilef, 2024-02-29:
Yes, this is the solution. But instead of setting temperatures, you can simply turn off the climate entity.

GravitySandwichh, 2024-03-02:
> I have trv in every room. I made an node red script that checks every climate if it needs heat. If multiple climate need heat I find the lowest current temperature and the highest set point and send [...]

Any possibility to share this node-red flow?

blitzu, 2024-03-02:
> Any possibility to share this node-red flow?

[
    {
        "id": "ea96edc559a48086",
        "type": "tab",
        "label": "Incalzire",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "924bf21a0fc0a186",
        "type": "function",
        "z": "ea96edc559a48086",
        "name": "Gaseste temperatura interioara",
        "func": "// Obține datele de stare ale entităților \"climate\"\nvar climateEntities = msg.payload.filter(entity => entity.entity_id.startsWith('climate.'));\n\n// Inițializare variabile pentru a ține evidența celei mai mari diferențe\nvar maxDifference = -Infinity;\nvar entityWithMaxDifference = null;\n\n// Iterează prin entitățile \"climate\"\nclimateEntities.forEach(entity => {\n    // Calculează diferența dintre temperature și current_temperature\n    var temperature = parseFloat(entity.attributes.temperature);\n    var currentTemperature = parseFloat(entity.attributes.current_temperature);\n    var difference = Math.abs(temperature - currentTemperature);\n\n    // Actualizează valoarea maximă și entitatea corespunzătoare\n    if (difference > maxDifference) {\n        maxDifference = difference;\n        entityWithMaxDifference = entity;\n    }\n});\n\n// Verifică dacă a fost găsită o entitate cu diferența maximă\n\n    // Verifică dacă a fost găsită o entitate cu diferența maximă\n    if (entityWithMaxDifference) {\n         // Trimite current_temperature la entitatea number.opentherm_indoor_temp\n        msg.payload = parseFloat(entityWithMaxDifference.attributes.current_temperature);\n       \n        // Setează serviciul pentru Home Assistant\n        msg.service = \"number.set_value\";\n    } else {\n        // Dacă nu s-a găsit nicio entitate, setează un mesaj de eroare\n        msg.payload = 30;\n\n        // Restul codului care nu necesită delay\n        // ...\n    }\n\n    return msg;\n // 5000 milisecunde = 5 secunde",
        "outputs": 1,
        "timeout": 0,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 210,
        "y": 240,
        "wires": [
            [
                "10d7d4b3cd2a2c1a"
            ]
        ]
    },
    {
        "id": "7f67b377f2a74528",
        "type": "cronplus",
        "z": "ea96edc559a48086",
        "name": "Cronometru",
        "outputField": "payload",
        "timeZone": "",
        "storeName": "",
        "commandResponseMsgOutput": "output1",
        "defaultLocation": "",
        "defaultLocationType": "fixed",
        "outputs": 1,
        "options": [
            {
                "name": "schedule1",
                "topic": "topic1",
                "payloadType": "default",
                "payload": "",
                "expressionType": "cron",
                "expression": "*/10 * * * * * *",
                "location": "",
                "offset": "0",
                "solarType": "all",
                "solarEvents": "sunrise,sunset"
            }
        ],
        "x": 150,
        "y": 40,
        "wires": [
            [
                "0a1f393bc0ecc888"
            ]
        ]
    },
    {
        "id": "1e80635bfed65ad0",
        "type": "function",
        "z": "ea96edc559a48086",
        "name": "Gaseste temperatura maxim ceruta",
[... 136 lines cut ...]
    },
    {
        "id": "d01d6bb5.176d08",
        "type": "server",
        "name": "Home Assistant",
        "version": 5,
        "addon": true,
        "rejectUnauthorizedCerts": true,
        "ha_boolean": "y|yes|true|on|home|open",
        "connectionDelay": true,
        "cacheJson": true,
        "heartbeat": false,
        "heartbeatInterval": 30,
        "areaSelector": "friendlyName",
        "deviceSelector": "friendlyName",
        "entitySelector": "friendlyName",
        "statusSeparator": "at: ",
        "statusYear": "hidden",
        "statusMonth": "short",
        "statusDay": "numeric",
        "statusHourCycle": "h23",
        "statusTimeFormat": "h:m",
        "enableGlobalContextStore": true
    }
]

Laxilef, 2024-03-03:
Automation in HA:

```yaml
alias: Sync TRV with boiler
description: ""
trigger:
  - platform: time_pattern
    seconds: /30
condition: []
action:
  - variables:
      current_actions: "{{ thermostats | map('state_attr', 'hvac_action') | list }}"
      current_action: "{{ iif('heating' in current_actions, 'heat', 'off') }}"
      max_target_temp: >-
        {{ thermostats | map('state_attr', 'temperature') | list | max |
        float(0) | round(2) }}
      min_current_temp: >-
        {{ thermostats | map('state_attr', 'current_temperature') | list | min |
        float(0) | round(2) }}
  - if:
      - condition: template
        value_template: "{{ states(target_thermostat) != current_action }}"
    then:
      - service: |-
          {% if current_action == 'heat' %}
            climate.turn_on
          {% else %}
            climate.turn_off
          {% endif %}
        target:
          entity_id: "{{ target_thermostat }}"
  - if:
      - condition: template
        value_template: >-
          {{ (state_attr(target_thermostat, 'temperature') | float(0) -
          max_target_temp) | abs | round(2) > 0.1 }}
    then:
      - service: climate.set_temperature
        target:
          entity_id: "{{ target_thermostat }}"
        data:
          temperature: "{{ max_target_temp }}"
  - if:
      - condition: template
        value_template: >-
          {{ (states(indoor_temp_entity) | float(0) - min_current_temp) | abs |
          round(2) > 0.1 }}
    then:
      - service: number.set_value
        target:
          entity_id: "{{ indoor_temp_entity }}"
        data:
          value: "{{ min_current_temp }}"
variables:
  thermostats:
    - climate.bedroom_1
    - climate.bedroom_2
    - climate.bathroom
    - climate.livingroom
  target_thermostat: climate.opentherm_heating
  indoor_temp_entity: number.opentherm_indoor_temp
mode: single
```

Edit before saving:
```yaml
variables:
  thermostats:
    - climate.bedroom_1
    - climate.bedroom_2
    - climate.bathroom
    - climate.livingroom
  target_thermostat: climate.opentherm_heating
  indoor_temp_entity: number.opentherm_indoor_temp
```

SanFable, 2024-03-05:
I did similar but much simpler,

I have 6 smart TRVs with 6 xiaomi temperature sensors as their temp input
TRVs are doing the job to heat up to desired temperature.
TRVs are reporting "PI heating demand" - the percentage of needle insertion (0 is closed, 100 is fully open)

I did simple automation:
If sum of all these is over 41 that means there is a heating demand -> heating on
If sum of all these is under 20 that means we heated up enough -> heating off

values 41 and 20 are my experimentable but it seems to work fine.

If i Will have to heat only one room, it will for sure open thermostat for more than 40%, so "all the heat" will go into this one.
If 3 rooms will request 16% for just little heat up, boiler will turn on.

closed 2024-03-25 by github-actions[bot]