Laxilef/OTGateway #55: Request: Ethernet connection support

pksoft585, 2024-03-02:
Thank you for your project. Now I'm using Wemos D1 Mini32 board (& Tindie Opentherm master shield).

It will be great if your gateway could support Ethernet connection instead of wi-fi. It is more reliable and especially with PoE it can be used for power supply. I think (when you don't use **_Wifi-Manager_** for configuration) it could be maybe simpler now.

I like more complex Ethernet board with **LAN8720** as **WT32-ETH01**, **Olimex ESP32-POE** or simple module with SPI as **Wiznet W5500**.

What do you thinkin about it? Unfortunately my skills aren't to high to fork your project and modify it for Ethernet. Maybe someone else...

Laxilef, 2024-03-03:
Hi,

I thought about it, but now I don’t have free time and devices with an Ethernet port for testing. The project code for working with the network is not complicated. If you have experience in programming, try to add this feature and create a PR.

closed 2025-02-19 by pksoft585

Eelviny, 2026-11-02:
I'm currently developing an open source hardware Opentherm master board (that I will also sell) and it uses the W5500 alongside an ESP32-C6. The above PR should be ready soon, and hopefully others can expand upon its base to support different ethernet implementations in the future!

The board itself should be released in a month or so, if all goes to plan.