Laxilef/OTGateway #56: Baxi Ampera не включается

c0dr1ver, 2024-03-07:
Прошивка: v1.4.0-rc.16

Установил согласно документации ID 4 
сущности в HA прокинулись, вижу температуру теплоносителя, могу через HA изменять уставку, но котел всегда на паузе. 

OpenTherm settings: 
V - Sync modulation with heating
все остальное выключено, поигрался с другими флагами, но результат такой же. 

Сенсор в HA OpenTherm Gateway Diagnostic показывает проблему. 
[image]

Не подскажите какие еще настройки нужно произвести и как на котле правильно выбрать термостат?

c0dr1ver, 2024-03-07:
вывод telnet : 
```
[00:06:13][OT][TRACE] ID:   57   Request: 90395500   Response: 60390000   Attempt:  1   Status: INVALID
[00:06:13][OT][TRACE] ID:   18   Request:   120000   Response: f0120000   Attempt:  1   Status: INVALID
```

Всё остальные ID - Status: SUCCESS

Laxilef, 2024-03-07:
В настройках котла включили opentherm? Там это называется zont connect.

c0dr1ver, 2024-03-08:
> В настройках котла включили opentherm? Там это называется zont connect. После изменения master member id перезагрузили esp?

Да, а не подскажете какой у вас сенсор выбран, там в настройках Zont/OT на котле есть 3 варианта

Laxilef, 2024-03-08:
Никакой. Не стал их использовать.
Попробуйте эти master member id: 4, 260, 516, 772, 1028

c0dr1ver, 2024-03-09:
> Никакой. Не стал их использовать. Попробуйте эти master member id: 4, 260, 516, 772, 1028

можете прислать фото настроек и что котел на главном экране в итоге отображает, я настраиваю его удаленно и мне нужнен какой-то референс

у меня при установке в HA меняются значения уставки в графе котел, но что бы я не пробовал из ID и других настроек, по данным я вижу что котел не стартует. Вот как мой экран выглядит 

Если указать какой-то из датчиков ZONT
[image]
Если не указывать номер используемых датчиков ZONT 
[image]

Если я ставлю температуру снаружи в нутри через mqtt она не появляется на экране, но уставка на многих ID переключается без проблем.

Laxilef, 2024-03-09:
Температуру с датчиков в амперу не пробрасывал, т.к. не видел в этом смысла.
Рядом с котлом буду на неделе, сделаю фото.
По памяти, все так же, но у меня нет индикации ГВС, потому что не использую. Загрузите логи, посмотрю. Интересует ID 0, 1, 14.

Настройки:
[image]

c0dr1ver, 2024-03-10:
> По памяти, все так же, но у меня нет индикации ГВС, потому что не использую. Загрузите логи, посмотрю. Интересует ID 0, 1, 14.

```
[12:26:17][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:17][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:17][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:19][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:19][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:19][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:20][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:20][OT][TRACE] ID:   14   Request: 900e6400   Response: 500e6400   Attempt:  1   Status: SUCCESS
[12:26:21][OT][HEATING][SILENT] Set max modulation 100%
[12:26:21][OT][TRACE] ID:   49   Request: 80310000   Response: 4031551e   Attempt:  1   Status: SUCCESS
[12:26:21][OT][TRACE] ID:   57   Request: 90395500   Response: 60390000   Attempt:  1   Status: INVALID
[12:26:22][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: ot-electro/status
[12:26:22][OT][TRACE] ID:   18   Request:   120000   Response: f0120000   Attempt:  1   Status: INVALID
[12:26:22][MQTT][TRACE] Successfully publish 470 of 470 bytes to topic: ot-electro/state
[12:26:22][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:23][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:24][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:24][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:24][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:25][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:26][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:26][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:27][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:27][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:28][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:28][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: ot-electro/status
[12:26:28][MQTT][TRACE] Successfully publish 470 of 470 bytes to topic: ot-electro/state
[12:26:29][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:29][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:29][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:30][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:31][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:31][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:32][MQTT][MSG][TRACE] Topic: ot-electro/settings/set
>  {"heating": {"target" : 56.0}}

[12:26:33][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:34][MQTT][TRACE] Successfully publish 0 of 0 bytes to topic: ot-electro/settings/set
[12:26:34][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:34][MAIN][VERB] Free heap size: 89148 of 237956 bytes (min: 37452, diff: 0), max free block: 69620 (min: 25588, diff: 0, frag: 21%)
[12:26:35][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:35][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: ot-electro/status
[12:26:36][MQTT][TRACE] Successfully publish 470 of 470 bytes to topic: ot-electro/state
[12:26:36][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:36][MQTT][TRACE] Successfully publish 555 of 555 bytes to topic: ot-electro/settings
[12:26:36][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:37][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:38][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:38][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:38][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:39][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:39][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:40][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:40][REGULATOR][INFO] New target: 56.00
[12:26:41][OT][TRACE] ID:    0   Request: 80000100   Response: 40000142   Attempt:  1   Status: SUCCESS
[12:26:41][OT][TRACE] ID:   17   Request:   110000   Response: c0110000   Attempt:  1   Status: SUCCESS
[12:26:41][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: ot-electro/status
[12:26:41][OT][TRACE] ID:   25   Request: 80190000   Response: c0192c00   Attempt:  1   Status: SUCCESS
[12:26:42][OT][HEATING][INFO] Set temp = 56
[12:26:42][MQTT][TRACE] Successfully publish 470 of 470 bytes to topic: ot-electro/state
[12:26:42][OT][TRACE] ID:    1   Request: 90013800   Response: 50013800   Attempt:  1   Status: SUCCESS
```

c0dr1ver, 2024-03-10:
Я могу лишь предположить что шлюз меняет значение уставки внутреннего термостата, но при подключении OT котел переходит именно на термостат ОТ, а судя по индикации - по ОТ он температуру не видит(прочерки) и по этому собственно не стартует

Laxilef, 2024-03-10:
В настройках котла погодозависимое управление случайно не включено?
У меня практически все настройки те, что были по заводу, за исключением включенного opentherm. Т.е. котел не использует никакие датчики для управления нагревом.

c0dr1ver, 2024-03-10:
Да, вы совершенно правы. Оказывается включён был внешний термостат вместе с zont’ом. 

Все работает. Спасибо

closed 2024-03-10 by c0dr1ver

dullabha, 2026-10-20:
> 4\. Никакой. Не стал их использовать.
>     Попробуйте эти master member id: 4, 260, 516, 772, 1028

При попытке установить, например, 260 - получаю ошибку "Значение должно быть меньше или равно 255"

[image]

Laxilef, 2026-10-20:
> 12\. При попытке установить, например, 260 - получаю ошибку "Значение должно быть меньше или равно 255"

Для амперы нужно установить 2 или 4