Laxilef/OTGateway #57: Baxi eco 4s + WeMos D1 sheld. [OT][WARN] Invalid response after setBoilerStatus: TIMEOUT

MrDogYar, 2024-03-11:
Добрый день!
Связка Baxi eco 4s + WeMos D1 sheld.
Нет связи с котлом. [OT][WARN] Invalid response after setBoilerStatus: TIMEOUT
Меняли местами OT PIN без результатно

Часть Log
Основной в аттаче
[Log_BaxiEco4s+WenosD1 Shield .txt](https://github.com/Laxilef/OTGateway/files/14562595/Log_BaxiEco4s%2BWenosD1.Shield.txt)

00:00:00][NETWORK][TRACE] Set hostname 'zelenay19': success
[00:00:05][SENSORS][OUTDOOR][INFO] Starting on gpio 12...
[00:00:05][SENSORS][OUTDOOR][TRACE] Devices on bus: 1, DS18* devices: 1
[00:00:05][SENSORS][OUTDOOR][INFO] Started
[00:00:06][PORTAL][WEBSERVER][TRACE] Started: AP up or STA connected
[00:00:06][MQTT][INFO] Enabled
[00:00:06][MQTT][INFO] Started
[00:00:06][OT][INFO] Started. GPIO IN: 4, GPIO OUT: 5
[00:00:07][MAIN][VERB] Free heap size: 27440 of 81920 bytes (min: 27440, diff: 4294939856), max free block: 27144 (min: 27144, diff: 4294940152, frag: 1%)
[00:00:07][NETWORK][INFO] Connected, downtime: 7 s., IP: 192.168.0.107, RSSI: -73
[00:00:08][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  1   Status: TIMEOUT
[00:00:09][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  2   Status: TIMEOUT
[00:00:10][REGULATOR][INFO] New target: 25.00
[00:00:10][REGULATOR][EQUITHERM][INFO] New result: 59 (59.38)
[00:00:10][MAIN][VERB] Free heap size: 27216 of 81920 bytes (min: 27216, diff: 224), max free block: 27144 (min: 27144, diff: 0, frag: 0%)
[00:00:10][SENSORS][OUTDOOR][TRACE] Raw temp: 27.437500
[00:00:10][SENSORS][OUTDOOR][INFO] New temp: 27.430000
[00:00:10][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  3   Status: TIMEOUT
[00:00:12][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  4   Status: TIMEOUT
[00:00:13][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  5   Status: TIMEOUT
[00:00:13][OT][WARN] Invalid response after setBoilerStatus: TIMEOUT
[00:00:15][SENSORS][OUTDOOR][TRACE] Raw temp: 27.687500
[00:00:15][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  1   Status: TIMEOUT
[00:00:15][MQTT][INFO] Connecting to 91.203.125.214:1883...
[00:00:15][MAIN][VERB] Free heap size: 26944 of 81920 bytes (min: 26944, diff: 272), max free block: 26752 (min: 26752, diff: 392, frag: 0%)
[00:00:16][MQTT][INFO] Connected (downtime: 16 s.)
[00:00:16][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  2   Status: TIMEOUT
[00:00:17][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  3   Status: TIMEOUT
[00:00:18][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  4   Status: TIMEOUT
[00:00:19][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  5   Status: TIMEOUT
[00:00:19][OT][WARN] Invalid response after setBoilerStatus: TIMEOUT
[00:00:20][REGULATOR][EQUITHERM][INFO] New result: 40 (40.00)
[00:00:20][SENSORS][OUTDOOR][TRACE] Raw temp: 27.437500
[00:00:21][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  1   Status: TIMEOUT
[00:00:22][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  2   Status: TIMEOUT
[00:00:24][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  3   Status: TIMEOUT
[00:00:25][SENSORS][OUTDOOR][TRACE] Raw temp: 27.500000
[00:00:25][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  4   Status: TIMEOUT
[00:00:26][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  5   Status: TIMEOUT
[00:00:26][OT][WARN] Invalid response after setBoilerStatus: TIMEOUT
[00:00:28][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  1   Status: TIMEOUT
[00:00:29][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  2   Status: TIMEOUT
[00:00:30][SENSORS][OUTDOOR][TRACE] Raw temp: 27.562500
[00:00:30][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  3   Status: TIMEOUT
[00:00:31][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: zelenay19/status
[00:00:31][MAIN][VERB] Free heap size: 26784 of 81920 bytes (min: 26784, diff: 160), max free block: 25368 (min: 25368, diff: 1384, frag: 5%)
[00:00:31][MQTT][TRACE] Successfully publish 471 of 471 bytes to topic: zelenay19/state
[00:00:31][MAIN][VERB] Free heap size: 26296 of 81920 bytes (min: 26296, diff: 488), max free block: 24864 (min: 24864, diff: 504, frag: 5%)
[00:00:31][MQTT][TRACE] Successfully publish 553 of 553 bytes to topic: zelenay19/settings
[00:00:31][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  4   Status: TIMEOUT
[00:00:32][MAIN][VERB] Free heap size: 26088 of 81920 bytes (min: 26088, diff: 208), max free block: 24704 (min: 24704, diff: 160, frag: 5%)
[00:00:32][MQTT][TRACE] Successfully publish 633 of 633 bytes to topic: homeassistant/switch/zelenay19/emergency/config
[00:00:32][MAIN][VERB] Free heap size: 24640 of 81920 bytes (min: 24640, diff: 1448), max free block: 24104 (min: 24104, diff: 600, frag: 2%)
[00:00:32][MQTT][TRACE] Successfully publish 692 of 692 bytes to topic: homeassistant/number/zelenay19/emergency_target/config
[00:00:32][MAIN][VERB] Free heap size: 24448 of 81920 bytes (min: 24448, diff: 192), max free block: 23848 (min: 23848, diff: 256, frag: 2%)
[00:00:32][MQTT][TRACE] Successfully publish 820 of 820 bytes to topic: homeassistant/switch/zelenay19/emergency_use_equitherm/config
[00:00:32][MAIN][VERB] Free heap size: 24376 of 81920 bytes (min: 24376, diff: 72), max free block: 24056 (min: 23848, diff: 0, frag: 1%)
[00:00:33][MQTT][TRACE] Successfully publish 783 of 783 bytes to topic: homeassistant/switch/zelenay19/emergency_use_pid/config
[00:00:33][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  5   Status: TIMEOUT
[00:00:33][OT][WARN] Invalid response after setBoilerStatus: TIMEOUT
[00:00:33][MQTT][TRACE] Successfully publish 649 of 649 bytes to topic: homeassistant/switch/zelenay19/heating/config
[... 3 lines cut ...]
[00:00:34][MQTT][TRACE] Successfully publish 774 of 774 bytes to topic: homeassistant/sensor/zelenay19/boiler_heating_min_temp/config
[00:00:35][MQTT][TRACE] Successfully publish 772 of 772 bytes to topic: homeassistant/sensor/zelenay19/boiler_heating_max_temp/config
[00:00:35][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  1   Status: TIMEOUT
[00:00:35][SENSORS][OUTDOOR][TRACE] Raw temp: 27.437500
[00:00:35][MQTT][TRACE] Successfully publish 691 of 691 bytes to topic: homeassistant/number/zelenay19/heating_min_temp/config
[00:00:35][MQTT][TRACE] Successfully publish 690 of 690 bytes to topic: homeassistant/number/zelenay19/heating_max_temp/config
[00:00:36][MQTT][TRACE] Successfully publish 689 of 689 bytes to topic: homeassistant/number/zelenay19/heating_max_modulation/config
[00:00:36][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  2   Status: TIMEOUT
[00:00:36][MQTT][TRACE] Successfully publish 589 of 589 bytes to topic: homeassistant/switch/zelenay19/pid/config
[00:00:37][MQTT][TRACE] Successfully publish 579 of 579 bytes to topic: homeassistant/number/zelenay19/pid_p_factor/config
[00:00:37][MQTT][TRACE] Successfully publish 578 of 578 bytes to topic: homeassistant/number/zelenay19/pid_i_factor/config
[00:00:37][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  3   Status: TIMEOUT
[00:00:37][MQTT][TRACE] Successfully publish 577 of 577 bytes to topic: homeassistant/number/zelenay19/pid_d_factor/config
[00:00:38][MQTT][TRACE] Successfully publish 595 of 595 bytes to topic: homeassistant/number/zelenay19/pid_dt/config
[00:00:38][MQTT][TRACE] Successfully publish 671 of 671 bytes to topic: homeassistant/number/zelenay19/pid_min_temp/config
[00:00:38][MQTT][TRACE] Successfully publish 670 of 670 bytes to topic: homeassistant/number/zelenay19/pid_max_temp/config
[00:00:38][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  4   Status: TIMEOUT
[00:00:39][MQTT][TRACE] Successfully publish 629 of 629 bytes to topic: homeassistant/switch/zelenay19/equitherm/config
[00:00:39][MQTT][TRACE] Successfully publish 611 of 611 bytes to topic: homeassistant/number/zelenay19/equitherm_n_factor/config
[00:00:39][MAIN][VERB] Free heap size: 24280 of 81920 bytes (min: 24280, diff: 96), max free block: 24104 (min: 23848, diff: 0, frag: 0%)
[00:00:39][MQTT][TRACE] Successfully publish 606 of 606 bytes to topic: homeassistant/number/zelenay19/equitherm_k_factor/config
[... cut ...] ":5,"memberIdCode":0,"dhwPresent":true,"modulationSyncWithHeating":true,"summerWinterMode":false,"heatingCh2Enabled":false,"heatingCh1ToCh2":false,"dhwToCh2":false,"dhwBlocking":false}}
[00:00:39][MAIN][VERB] Free heap size: 23296 of 81920 bytes (min: 23296, diff: 984), max free block: 22728 (min: 22728, diff: 1120, frag: 2%)
[00:00:39][OT][TRACE] ID:    0   Request:     2200   Response:        0   Attempt:  5   Status: TIMEOUT
[00:00:39][OT][WARN] Invalid response after setBoilerStatus: TIMEOUT

Laxilef, 2024-03-12:
Какой адаптер используется?

MrDogYar, 2024-03-12:
Собран самостоятельно по вашей схеме только без светодиодной индикации. 
https://github.com/Laxilef/OTGateway/blob/master/assets/Schematic.pdf
 
Вчера вечером подключил к своему котлу Bocsh Gaz6000. Плата работает. :-(

> 
> 
>Какой адаптер используется?
 
 
С уважением,
Денис Титов
***@***.***

Laxilef, 2024-03-12:
Значит аппаратную часть адаптера исключаем.
На [сайте зонта](https://www.tvp-electro.ru/wp-content/uploads/2023/05/baxi-eco-4s-tsifrovoe-podklyuchenie-smart-2.0-dineco-1.jpg) eco 4s заявлен как совместимый, значит opentherm там работает.

Подключали вместо перемычки? Настройки котла пробовали сбрасывать?

MrDogYar, 2024-03-12:
Да. Перемычку удаляем и на ее место подключаем ОТ
 
Настройки не меняли и не сбрасывали.
 
 
 
Но что еще меня смутило. После удаления перемычки на контактах ОТ-котла 110 вольт.
С подключенным  к ОТ контроллера падает до 1-2 вольт.
 
У меня на Bosch без подключения ОТ 30 вольт, с подключеным ОТ 6-8 вольт.

> 
> 
>Значит аппаратную часть адаптера исключаем.
>На  сайте зонта eco 4s заявлен как совместимый, значит opentherm там работает.
>Подключали вместо перемычки? Настройки котла пробовали сбрасывать?
 
 
С уважением,
Денис Титов
***@***.***

MrDogYar, 2024-03-12:
А вот это интересно. Получаеться, что надо удалить перемычку, но подключаться не в нее как на всех котлах, а в разъем X6 

> 
>Да. Перемычку удаляем и на ее место подключаем ОТ
> 
>Настройки не меняли и не сбрасывали.
> 
> 
> 
>Но что еще меня смутило. После удаления перемычки на контактах ОТ-котла 110 вольт.
>С подключенным  к ОТ контроллера падает до 1-2 вольт.
> 
>У меня на Bosch без подключения ОТ 30 вольт, с подключеным ОТ 6-8 вольт.
>
> 
>> 
>> 
>>Значит аппаратную часть адаптера исключаем.
>>На  сайте зонта eco 4s заявлен как совместимый, значит opentherm там работает.
>>Подключали вместо перемычки? Настройки котла пробовали сбрасывать?
> 
> 
>С уважением,
>Денис Титов
>  
 
 
С уважением,
Денис Титов
***@***.***

MrDogYar, 2024-03-12:
> 
>А вот это интересно. Получаеться, что надо удалить перемычку, но подключаться не в нее как на всех котлах, а в разъем X6 
>
> 
>> 
>>Да. Перемычку удаляем и на ее место подключаем ОТ
>> 
>>Настройки не меняли и не сбрасывали.
>> 
>> 
>> 
>>Но что еще меня смутило. После удаления перемычки на контактах ОТ-котла 110 вольт.
>>С подключенным  к ОТ контроллера падает до 1-2 вольт.
>> 
>>У меня на Bosch без подключения ОТ 30 вольт, с подключеным ОТ 6-8 вольт.
>>
>> 
>>> 
>>> 
>>>Значит аппаратную часть адаптера исключаем.
>>>На  сайте зонта eco 4s заявлен как совместимый, значит opentherm там работает.
>>>Подключали вместо перемычки? Настройки котла пробовали сбрасывать?
>> 
>> 
>>С уважением,
>>Денис Титов
>>  
> 
> 
>С уважением,
>Денис Титов
>  
 
 
С уважением,
Денис Титов
***@***.***

Laxilef, 2024-03-12:
Это один и тот же разъем.
Проверьте корректность подключения питания котла. Судя по документации, один из контактов этой колодки соединен с нулем питания. И если там не ноль, а фаза, то ничего хорошего не будет.
110в там не должно быть, по спецификации OT до 45в, если не ошибаюсь.
[image]

MrDogYar, 2024-03-12:
Пообщался т техподдержкой БАКСИ.
 
*  Говорят, что ОТ разъем это как раз X6.
*  Перемычку надо удалить.
*  На ОТ должно быть 24 в.
 
Вечером пойдем с другом опять эксперементировать.
По итогам отпишусь.

> 
> 
>Это один и тот же разъем.
>Проверьте корректность подключения питания котла. Судя по документации, один из контактов этой колодки соединен с нулем питания. И если там не ноль, а фаза, то ничего хорошего не будет.
>110в там не должно быть, по спецификации OT до 45в, если не ошибаюсь.
>Screenshot_20240312_123730_Drive.jpg (view on web)
 
 
С уважением,
Денис Титов
***@***.***

MrDogYar, 2024-03-12:
Добрый вечер!
Все заработало!
Можно добавлять в копилку BAXI ECO 4S + WemosD1 Shield.
Для подключения надо удалить перемычку, а ОТ контакты подсоединить к разъему X6 на плате.
 
Единственное. Почему-то постоянно показывает, что в ГВС есть проток
sensor.opentherm_dhw_flow_rate.
 
Но на функционал ни как не влияет.
 
Спасибо за техподдержку.!!!

> 
>Пообщался т техподдержкой БАКСИ.
> 
>*  Говорят, что ОТ разъем это как раз X6.
>*  Перемычку надо удалить.
>*  На ОТ должно быть 24 в.
> 
>Вечером пойдем с другом опять эксперементировать.
>По итогам отпишусь.
> 
>> 
>> 
>>Это один и тот же разъем.
>>Проверьте корректность подключения питания котла. Судя по документации, один из контактов этой колодки соединен с нулем питания. И если там не ноль, а фаза, то ничего хорошего не будет.
>>110в там не должно быть, по спецификации OT до 45в, если не ошибаюсь.
>>Screenshot_20240312_123730_Drive.jpg (view on web)
> 
> 
>С уважением,
>Денис Титов
> 
 
 
С уважением,
Денис Титов
***@***.***

MrDogYar, 2024-03-12:
> В чем проблема была?

Перепутана L и N на котле. Плюс подключались к колодке термостата где была перемычка, а надо подключаться к разъему X6 на плате. (Он в низу)

closed 2024-03-13 by Laxilef

MrDogYar, 2024-03-20:
Добрый день!
А возможно реализвать в прошивке локальный MQTT на примере
https://github.com/martin-ger/uMQTTBroker
что б контроллер заворачивал свои сообщения сам на себя?