Laxilef/OTGateway #58: Viessmann Vitodens 100-W DHW temperature is displayed as Heating in opentherm.

nedievas, 2024-04-12:
[image]
[image]
[image]

Laxilef, 2024-04-12:
Please describe the problem in more detail.

nedievas, 2024-04-13:
I made a custom scheme of OpenTherm Gateway. Here's the schematics:
[image]
Constructed something like this:
[image]
Flashed with 1.4.0-rc.20.
Results are that LED does not work. Maybe faulty schematics (please inspect if have time). Managed to connect to my Viessmann Vitodens 100-W boiler. Observing that hot water temperature on the boiler is displayed as heating temperature in mqtt. Here the photos:
[image]
[image]
[image]
My boiler settings are: 55 degrees for hot water and 21 (not real temperature but the curve value for heating according to outside temperature) for heating. Here's the graph of this curve:
[image]
I suppose that is something like Equitherm. I did not activate this function. Though I tried to activate 'DHW flow rate' and 'Pressure', but I see 0.0 all the time. 
Do not understand if it's working as supposed to work. Please advise:)

Laxilef, 2024-04-14:
Your heating temperature is set to 30 degrees. According to the table you provided, this is about 60 degrees at an outside temperature of 10 degrees. I don’t know what delta your boiler has, but perhaps this is normal behavior. Can you set 21 degrees on the heating thermostat in Home Assistant?

nedievas, 2024-04-14:
> Your heating temperature is set to 30 degrees. According to the table you provided, this is about 60 degrees at an outside temperature of 10 degrees. I don’t know what delta your boiler has, but perhaps this is normal behavior. Can you set 21 degrees on the heating thermostat in Home Assistant?

Updated to [v1.4.0-rc.22](https://github.com/Laxilef/OTGateway/releases/tag/1.4.0-rc.22). Still...
I am able to change DHW and heating temperature in HA, but these changes does not change on the boiler. I am able to turn on/off heating in HA which affects the boiler - it just calms or resumes it's work, but on the boiler menu it still show 21 for heating, and  55 for DHW /should show OFF as per manual/. Turning on/off DHW on HA does not show any signs on the boiler.

Laxilef, 2024-04-15:
Hmm. It looks like we have to use other IDs to set the temperature if the curve mode on the boiler is enabled.
For now, I can suggest disabling this mode on the boiler and enabling equitherm in the firmware to calculate the temperature.

nedievas, 2024-04-17:
> Hmm. It looks like we have to use other IDs to set the temperature if the curve mode on the boiler is enabled. For now, I can suggest disabling this mode on the boiler and enabling equitherm in the firmware to calculate the temperature.
To disable this mode on the boiler I have to detach the outside temperature sensor from boiler. The boiler then automatically sets higher temperature for heating, as per manual.
Tried this (+ enabling equitherm), and I can not put more than 30 degrees in HA, and heating temp from opentherm is 0. Here is the photo:
[image]
If Equitherm is deactivated, then the temperature is back  and I may set it higher than 30. Also, after I attached the Opentherm gateway to the boiler and tried to sync temperature settings on HA with boiler settings, I've got the house cold, but DHW as usual.
> 
> Does the boiler react to changes in the DHW temperature from the home assistant?
As I mentioned in previous post, I may change it in HA, but it does not change on the boiler. Though, I observe that when I lower the DHW temperature on HA then the temperature is displayed on the right, the DHW on the boiler turns off, if I set it higher, then turns on! It just does not change the set temperatures on the boiler display itself. So the opentherm affects the boiler job, but I am not able to sync if it works as it should:)

Laxilef, 2024-04-17:
So, let's go in order:

1. When PID or/and Equitherm is turned on, the temperature on the thermostat in the home assistant is limited to 5..30 because this is the target indoor temperature.
2. When PID or/and Equitherm is turned on, the current temperature on the thermostat in the home assistant is the indoor temperature. If 0 is displayed, it means there is no sensor or automation is not configured to transmit indoor temperature.
3. If the "curve mode" is enabled on the boiler, then we cannot transfer the desired heat carrier temperature to the boiler. I think that in this mode the boiler expects the setpoint **indoor** temperature in ID 16, 23, but this is not currently supported in the firmware.

I'll think about what can be done, but not right now. Now I see this solution:
1. Disable "curve mode" (by turning off the sensor or in the service menu).
2. Installation of the outdoor sensor DS18B20 and connecting it to the ESP (for Equtherm). You can transmit the indoor temperature [using automation](https://github.com/Laxilef/OTGateway/wiki#reporting-indooroutdoor-temperature-from-any-home-assistant-sensor).
3. Using PID or Equitherm.

nedievas, 2024-04-17:
> So, let's go in order:
>
>     1. When PID or/and Equitherm is turned on, the temperature on the thermostat in the home assistant is limited to 5..30 because this is the target indoor temperature. [...]

Did add the internal temperature via automation just after my post.

>     3. If the "curve mode" is enabled on the boiler, then we cannot set the desired heat carrier temperature to the boiler. I think that in this mode the boiler expects the setpoint **indoor** temperature in ID 16, 23, but this is not currently supported in the firmware.
> 
> 
> I'll think about what can be done, but not right now. Now I see this solution:
> 
>     1. Disable "curve mode" (by turning off the sensor or in the service menu).
> 
>     2. Installation of the outdoor sensor DS18B20 and connecting it to the ESP (for Equtherm). You can report the indoor temperature [using automation](https://github.com/Laxilef/OTGateway/wiki#reporting-indooroutdoor-temperature-from-any-home-assistant-sensor).
> 
>     3. Using PID or Equitherm.
What about Danfoss thermostats that are used for setting separate room temperatures. I was expecting Opentherm for fault troubleshooting, and not for room temperature control. Or is OpenTherm still better option than original boiler settings? If no, is it possible for Opentherm gateway not to control the boiler heating, just DHW and reset boiler on F2 fault via HA automation?

Laxilef, 2024-04-17:
Let's start by testing my hypothesis. Try disabling the "curve mode" and check if the boiler will respond correctly to changes in setpoint temp.
Equitherm is the same curve mode, but more flexibly customizable.

Laxilef, 2024-04-19:
Any news?
I added the ability to use the temperature controller built into the boiler, but it needs to be tested. If my hypothesis is correct, then I will commit the changes.

nedievas, 2024-04-19:
> Any news? I added the ability to use the temperature controller built into the boiler, but it needs to be tested. If my hypothesis is correct, then I will commit the changes.

Now I am testing Equitherm and am at 2. point, where N=0,7, and K, and T are 0-es. Now going for K to set on 2 and test.
Would like to test your hypothesis, but let's calibrate my equitherm as per your excel_calc this weekend.

nedievas, 2024-04-19:
> Ok thx With the curve mode disabled, does the boiler correctly accept the heating temp and heat to the set temperature? Pls upload a screenshot of the graph `sensor.opentherm_heating_setpoint` and `sensor.opentherm_heating_temp`

I did not change anything on HA except setting  N to 0.7, and later today adding K with 2.0 , just set all rooms including garage temperature to 23 on Danfoss Link. 

[image]

[image]

My current outputs in HA:
[image]
[image]

Laxilef, 2024-04-19:
``sensor.opentherm_heating_temp`` looks strange. This value should be smoother, example:
[image]

I have no experience with Danfoss Link, is it also connected to the opentherm bus?

nedievas, 2024-04-19:
> `sensor.opentherm_heating_temp` looks strange. This value should be smoother, example: ![Screenshot_7](https://private-user-images.githubusercontent.com/34578544/324063556-a2e63561-5a2b-4a45-b486-5be3e0261a10.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM1NTA2NDcsIm5iZiI6MTcxMzU1MDM0NywicGF0aCI6Ii8zNDU3ODU0NC8zMjQwNjM1NTYtYTJlNjM1NjEtNWEyYi00YTQ1LWI0ODYtNWJlM2UwMjYxYTEwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE5VDE4MTIyN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFmYjQzZWRjMzVmNDkxMDI5ZmZlMjcwNjhjYzJkZGEyZTk3ZjQzMGI3M2ZkYzE2OWQyMDgwYWFmNGI1YzA0MDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8Kb84h0GyQxchWz3UWFhP1JTxc7lShwHDPtd42Kji94)
> 
> I have no experience with Danfoss Link, is it also connected to the opentherm bus?

No. It controls thermostats for floor heating and two radiators in the boiler room and garage.

Laxilef, 2024-04-19:
Ok. After the weekend, please upload screenshots of these graphs again.

Pajuris, 2024-04-21:
@nedievas Kas tau neveikia su tuo boileriu? Aš turiu tokį pat ir pas mane viskas gerai veikia. Sorry neturiu laiko į šį topiką įsigilinti ir viską išanalizuoti, tad jei gali trumpai mestelk info. Temperatūra kaip rašai pirmam poste taip pat veikė ir pas mane blogai, bet atrodo įjungus PID (būtent PID) viskas buvo tvarkoj. Taip pat su nustatymai reiktų pažaisti, uždėti varneles. Pas mane (žr. screeną)
[image]

nedievas, 2024-04-21:
> Ok. After the weekend, please upload screenshots of these graphs again.

[image]

Feel a lot chiller than before :)

> @nedievas Kas tau neveikia su tuo boileriu? Aš turiu tokį pat ir pas mane viskas gerai veikia. Sorry neturiu laiko į šį topiką įsigilinti ir viską išanalizuoti, tad jei gali trumpai mestelk info. Temperatūra kaip rašai pirmam poste taip pat veikė ir pas mane blogai, bet atrodo įjungus PID (būtent PID) viskas buvo tvarkoj. Taip pat su nustatymais reiktų pažaisti, uždėti varneles. Pas mane: ![Screenshot 2024-04-21 at 09 56 13](https://private-user-images.githubusercontent.com/156599470/324228885-bf93f4bc-5876-4c9e-8a52-7fbac6a6d4d8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM3MjM0MTUsIm5iZiI6MTcxMzcyMzExNSwicGF0aCI6Ii8xNTY1OTk0NzAvMzI0MjI4ODg1LWJmOTNmNGJjLTU4NzYtNGM5ZS04YTUyLTdmYmFjNmE2ZDRkOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyMVQxODExNTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYWM0NTljMmM0YWM3NGFlOWQ5MjhmNjM4OTdhZjE1MzdlZmZjYTM5MTdmNzY3ZjdmZWZiOGVlMjBlNjJiYTE1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Ds5iZ2Pj31NHnTXnhE-EOqjVVA8e4wsqICMt3Hhq4mY)

O kaip atrodo visi settings? PID ir panašiai... pas mane grindinis šildymas per Danfoss Link kuris su HA nekalba ir boileris buvo pajungtas per lauko temp daviklį. Tai kaip ir viskas veikė, bet F2 kodą mesdavo kai staiga lauko temp nukrisdavo, tai užmačiau Opentherm gateway - pasigaminau ir bandau pajungti tam kad per HA turėčiau galimybę reset jei tas F2 išlenda, bet čia jis perima valdymą, todėl bandau prisiderinti.

Pajuris, 2024-04-21:
Nustatymai iš HA:

[image]
[image]
Kai pajungti termostatą, katilas nebereaguoja į lauko temperatūrą. Taigi, tai nebeaktualu. Be to ir nebereikalinga, nes per HA gali geriau susidėlioti automatikas, nei Viessmann sugalvojo kreives.
Beje, jei kartais norėtum, galiu pigiai perleisti ir Viessmann originalų termostatą su OT jungtimi. Veikia ir integracija į HA. Tik aišku valdymo tiek nebus, kiek tarkim su termostatu ir šituo firmware.

nedievas, 2024-04-21:
> Nustatymai iš HA:
> 
> ![Screenshot 2024-04-21 at 21 57 51](https://private-user-images.githubusercontent.com/156599470/324273467-746eae17-113b-496a-9e42-71ce82e74640.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM3MjY2NTEsIm5iZiI6MTcxMzcyNjM1MSwicGF0aCI6Ii8xNTY1OTk0NzAvMzI0MjczNDY3LTc0NmVhZTE3LTExM2ItNDk2YS05ZTQyLTcxY2U4MmU3NDY0MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyMVQxOTA1NTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMzZjODdmNzYzM2NjNTFhZWI2ZDM3ZmQ1NjEzMTI5NDY3ZDVlYmU5MDU0YWQxNTUxMGEzMDM1ZmE0ZDczOThjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.xAPAqHB_Hn-5WHqFSzsAmWYqlORwVe11QuWE2_-R4xE) ![Screenshot 2024-04-21 at 21 58 11](https://private-user-images.githubusercontent.com/156599470/324273470-f7ccd4eb-02a3-4389-af3f-0d41c325907e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM3MjY2NTEsIm5iZiI6MTcxMzcyNjM1MSwicGF0aCI6Ii8xNTY1OTk0NzAvMzI0MjczNDcwLWY3Y2NkNGViLTAyYTMtNDM4OS1hZjNmLTBkNDFjMzI1OTA3ZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyMVQxOTA1NTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kNjc5YWNmYzliYzZlOGQyZDY4MWIxZWE4MTQwNzhhY2RhNDUyNzY2ZjM3YjhlMTcyYjJmNTUyNzVhNDI0ODQ4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5JkWliNAhlqC_dQ61aLOX1E-0pMrC9Y9GPd8-bbYEA0) Kai pajungti termostatą, katilas nebereaguoja į lauko temperatūrą. Taigi, tai nebeaktualu. Be to ir nebereikalinga, nes per HA gali geriau susidėlioti automatikas, nei Viessmann sugalvojo kreives. Beje, jei kartais norėtum, galiu pigiai perleisti ir Viessmann originalų termostatą su OT jungtimi. Veikia ir integracija į HA. Tik aišku valdymo tiek nebus, kiek tarkim su termostatu ir šituo firmware.

Tai mano termostatai yra danfoss, kiekvienam kambariui atskiras. Juos gal pakeisiu į ESP32 HA, bet dabar šį OT bandau. Įjungiau PID - nesiskiria nustatymai nuo jūsų. Tik ar jus reguliuojate per HA boilerio temperatūrą ar per kitus termostatus?

Pajuris, 2024-04-21:
Pas mane gridinis 1 aukšte ir radiatoriai - 2. Visur Danfoss pavaros, zigbee jungikliai jas junginėja ir temperatūros jutikliai kiekviena kambary. Viskas suintegruota į HA.
Šalčiausio kambario temperatūros jutiklis, termostate nurodytas kaip vidaus temperatūra, lauko temperatūros jutiklis, kaip lauko temperatūra.
Viską valdau per HA ir per HA automatikas. Ten pat nurodau maks temperatūras, kitus parametrus, kuriuos leidžia termostato firmware.

Pajuris, 2024-04-21:
> Tik ar jus reguliuojate per HA boilerio temperatūrą ar per kitus termostatus?

Tai katilo temperatūra tik per HA termostato valdymą ir tesireguliuoja. Daugiau nėra būdų.
Ar klausi apie  kambarių temperatūrą? Kiekvienam kambariui norima temperatūra, reguliuojasi per HA virtualius termostatus (ima duomenis iš temperatūros daviklių ir reguliuoja zigbee jungiklius per kuriuos įjungia\išjungia Danfoss pavaras).

nedievas, 2024-04-21:
> Pas mane gridinis 1 aukšte ir radiatoriai - 2. Visur Danfoss pavaros, zigbee jungikliai jas junginėja ir temperatūros jutikliai kiekviena kambary. Viskas suintegruota į HA. Šalčiausio kambario [...]

Mano ir toks planas pasidaryti per jau esamas Danfoss pavaras. Kokius tam zigbee naudojate?

nedievas, 2024-04-21:
> Tai katilo temperatūra tik per HA termostato valdymą ir tesireguliuoja. Daugiau nėra būdų. Ar klausi apie kambarių temperatūrą? Kiekvienam kambariui norima temperatūra, reguliuojasi per HA [...]

Tai aš supratau, kad katilas mano kaip buvo nustatytas - aš jo in nereguliavau, bet tas error F2 mane užkniso ir sugalvojau ieškoti sprendimų.

Pajuris, 2024-04-21:
Paprasčiausius.... tokius (yra LT jau):
[image]
Netobuli, bet veikia.
Manau kad temperatūros jutiklis grybą pjauna. Bet šiaip tos reguliuoti pagal lauko temperatūrą nėra gerai, gaila dujų. Ypač kai šalta lauke ir sukelia temperatūrą boileryje. Tada nebedirba kond. rėžimu ir netaupo dujų.
Perėjus prie vidaus termostato nuo lauko kreivių, aš pradėjau taupyti maždaug 1-2 kubo per parą. Perėjus prie šito termostatato firmwares ir tinkamų automatikų, dar po 2-3 kūbus pradėjau taupyti per dieną.

nedievas, 2024-04-21:
Tai tokius Zigbee aš apšvietimui naudoju :) Gal yra kur info kaip pasijungti kaip Jūsų? Galvojau apie tuos kinietiškus kur daug relių ant pcb ir esphome jungiasi., bet nenorėjau bandyti per šildymo sezoną.

Laxilef, 2024-04-21:
> > Ok. After the weekend, please upload screenshots of these graphs again.
> 
> ![paveikslas](https://private-user-images.githubusercontent.com/6176288/324269465-a0babd7f-6026-4c13-80db-0e4900334f0b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM3NDI4OTMsIm5iZiI6MTcxMzc0MjU5MywicGF0aCI6Ii82MTc2Mjg4LzMyNDI2OTQ2NS1hMGJhYmQ3Zi02MDI2LTRjMTMtODBkYi0wZTQ5MDAzMzRmMGIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjFUMjMzNjMzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDg0MjcxMzU2YTU2ZTk3MzlkOWQ4OGYxOTQxMTQ0Zjg1ZGY5MDdlMjQ4MGMwZDM2ZDUwN2NjMjE4MGFiZTQ5MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.A2syrhdNIuUoqwzkdmySERWHRKT3EubJxQIgNOzN_H0)
> 
> Feel a lot chiller than before :)
> 
It looks like your boiler heats up to 80 degrees when set to ~40 degrees. It is not right. Try set the checkboxes as @Pajuris wrote.

Pajuris, 2024-04-22:
> Gal yra kur info kaip pasijungti kaip Jūsų?

Tai kad ten nereikia jokios info. Viskas labai paprasta. Ant kiekvieno kontūro montuoji Normaly closed galvą, ją pajungi prie pasirinkto jungiklio. Jungiklį prijungi prie HA. Kambariuose išmėtai pasirinktus temperatūros jutiklius. Juos taip pat prijungi prie HA. HA config faile, kuri virtualų termostatą, nurodai kad heateris yra tas galvą įijungiantis išjungiantis jungiklis, o temperatūros rodmenis imsi iš konkretaus kambario jutiklio. Viskas. Restartuoji HA, atsiranda termostatas, kurį gali valdyti kaip tinkamas.

Laxilef, 2024-04-22:
@nedievas could you then test the "Native heating control" option in 1.4.0-rc.24? This option must work with an outdoor temperature sensor connected to the boiler.
https://github.com/Laxilef/OTGateway/releases/tag/1.4.0-rc.24

nedievas, 2024-04-22:
> @nedievas could you then test the "Native heating control" option in [1.4.0-rc.24](https://github.com/Laxilef/OTGateway/releases/tag/1.4.0-rc.24)? This option must work with an outdoor temperature sensor connected to the boiler.

Before upgrade:
[image]

nedievas, 2024-04-22:
Upgraded. Connected external temperature sensor to boiler, disabled automation for outside temperature, equitherm and PID off. I am observing, that heating target is set to max 30 degrees??? Is it OK?

attaching the telnet log:
```Connected to 192.168.20.94.
Escape character is '^]'.
[00:11:52][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:11:52][OT][TRACE] ID:   33   Request:   210000   Response: 4021001f   Attempt:  1   Status: SUCCESS
[00:11:52][OT][HEATING][INFO] Set current indoor temp: 21.47 (converted: 21.47)
[00:11:52][MQTT][TRACE] Successfully publish 566 of 566 bytes to topic: opentherm/state
[00:11:52][OT][TRACE] ID:   24   Request: 10181578   Response: f0180000   Attempt:  1   Status: INVALID
[00:11:52][OT][HEATING][WARN] Failed set current indoor temp
[00:11:52][OT][HEATING][INFO] Set target indoor temp: 30.00 (converted: 30.00)
[00:11:52][OT][TRACE] ID:   16   Request: 10101e00   Response: 70100000   Attempt:  1   Status: INVALID
[00:11:52][OT][HEATING][WARN] Failed set target indoor temp
[00:11:53][OT][TRACE] ID:    0   Request:     3300   Response: 40003302   Attempt:  1   Status: SUCCESS
[00:11:54][OT][TRACE] ID:   26   Request: 801a0000   Response: 401a3b19   Attempt:  1   Status: SUCCESS
[00:11:54][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:11:54][MAIN][VERB] Free heap size: 24256 of 81920 bytes (min: 19216, diff: 0), max free block: 23864 (min: 15488, diff: 0, frag: 1%)
[00:11:54][OT][TRACE] ID:   25   Request: 80190000   Response: c0192780   Attempt:  1   Status: SUCCESS
[00:11:55][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:11:55][OT][TRACE] ID:   33   Request:   210000   Response: 4021001f   Attempt:  1   Status: SUCCESS
[00:11:55][OT][HEATING][INFO] Set current indoor temp: 21.47 (converted: 21.47)
[00:11:55][OT][TRACE] ID:   24   Request: 10181578   Response: f0180000   Attempt:  1   Status: INVALID
[00:11:55][OT][HEATING][WARN] Failed set current indoor temp
[00:11:55][OT][HEATING][INFO] Set target indoor temp: 30.00 (converted: 30.00)
[00:11:56][OT][TRACE] ID:   16   Request: 10101e00   Response: 70100000   Attempt:  1   Status: INVALID
[00:11:56][OT][HEATING][WARN] Failed set target indoor temp
[00:11:57][OT][TRACE] ID:    0   Request:     3300   Response: 40003302   Attempt:  1   Status: SUCCESS
[00:11:57][OT][TRACE] ID:   26   Request: 801a0000   Response: 401a3b19   Attempt:  1   Status: SUCCESS
[00:11:57][MQTT][MSG][TRACE] Topic: opentherm/state/set
>  {"temperatures": {"indoor":21.47}}

[00:11:57][MQTT][TRACE] Successfully publish 0 of 0 bytes to topic: opentherm/state/set
[00:11:57][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:11:57][OT][TRACE] ID:   25   Request: 80190000   Response: c0192480   Attempt:  1   Status: SUCCESS
[00:11:58][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:11:58][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:11:58][MQTT][TRACE] Successfully publish 566 of 566 bytes to topic: opentherm/state
[00:11:58][OT][TRACE] ID:   33   Request:   210000   Response: 4021001f   Attempt:  1   Status: SUCCESS
[00:11:58][OT][HEATING][INFO] Set current indoor temp: 21.47 (converted: 21.47)
[00:11:58][OT][TRACE] ID:   24   Request: 10181578   Response: f0180000   Attempt:  1   Status: INVALID
[00:11:58][OT][HEATING][WARN] Failed set current indoor temp
[00:11:58][OT][HEATING][INFO] Set target indoor temp: 30.00 (converted: 30.00)
[00:11:59][OT][TRACE] ID:   16   Request: 10101e00   Response: 70100000   Attempt:  1   Status: INVALID
[00:11:59][OT][HEATING][WARN] Failed set target indoor temp
[00:12:00][OT][TRACE] ID:    0   Request:     3300   Response: 40003302   Attempt:  1   Status: SUCCESS
[00:12:00][OT][TRACE] ID:   26   Request: 801a0000   Response: 401a3b19   Attempt:  1   Status: SUCCESS
[00:12:00][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:12:01][OT][TRACE] ID:   25   Request: 80190000   Response: c01921b3   Attempt:  1   Status: SUCCESS
[00:12:01][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:12:01][OT][TRACE] ID:   33   Request:   210000   Response: c021001e   Attempt:  1   Status: SUCCESS
[00:12:01][OT][HEATING][INFO] Set current indoor temp: 21.47 (converted: 21.47)
[00:12:02][OT][TRACE] ID:   24   Request: 10181578   Response: f0180000   Attempt:  1   Status: INVALID
[00:12:02][OT][HEATING][WARN] Failed set current indoor temp
[00:12:02][OT][HEATING][INFO] Set target indoor temp: 30.00 (converted: 30.00)
[00:12:02][OT][TRACE] ID:   16   Request: 10101e00   Response: 70100000   Attempt:  1   Status: INVALID
[00:12:02][OT][HEATING][WARN] Failed set target indoor temp
[00:12:03][OT][TRACE] ID:    0   Request:     3300   Response: 40003302   Attempt:  1   Status: SUCCESS
[00:12:03][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:12:04][MQTT][TRACE] Successfully publish 566 of 566 bytes to topic: opentherm/state
[00:12:04][OT][TRACE] ID:   26   Request: 801a0000   Response: 401a3b19   Attempt:  1   Status: SUCCESS
[00:12:04][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:12:04][OT][TRACE] ID:   25   Request: 80190000   Response: c0191f33   Attempt:  1   Status: SUCCESS
[... cut ...]
[00:12:24][OT][HEATING][INFO] Set current indoor temp: 21.47 (converted: 21.47)
[00:12:25][OT][TRACE] ID:   24   Request: 10181578   Response: f0180000   Attempt:  1   Status: INVALID
[00:12:25][OT][HEATING][WARN] Failed set current indoor temp
[00:12:25][OT][HEATING][INFO] Set target indoor temp: 30.00 (converted: 30.00)
[00:12:25][OT][TRACE] ID:   16   Request: 10101e00   Response: 70100000   Attempt:  1   Status: INVALID
[00:12:25][OT][HEATING][WARN] Failed set target indoor temp
```
[image]
[image]
[image]

nedievas, 2024-04-22:
Will leave fixed every room temperatures to 23 degrees for today. It's getting cold and I could not get target 23 temperature with equitherm and PID. Some rooms are as low as 20.4 and did not reach my desired temperature.

Laxilef, 2024-04-23:
> Upgraded. Connected external temperature sensor to boiler, disabled automation for outside temperature, equitherm and PID set to off, in the boiler set "curve temperature" to 21, DHW to 55. I am observing, that heating target is set to max 30 degrees??? Is it OK?

Thx for tests. In the firmware, room temperature is limited to 5..30 C. But looking at the logs, your boiler does not accept ID16 (Room Setpoin) and ID24 (Room temperature):
```
[00:12:01][OT][HEATING][INFO] Set current indoor temp: 21.47 (converted: 21.47)
[00:12:02][OT][TRACE] ID:   24   Request: 10181578   Response: f0180000   Attempt:  1   Status: INVALID
[00:12:02][OT][HEATING][WARN] Failed set current indoor temp
[00:12:02][OT][HEATING][INFO] Set target indoor temp: 30.00 (converted: 30.00)
[00:12:02][OT][TRACE] ID:   16   Request: 10101e00   Response: 70100000   Attempt:  1   Status: INVALID
[00:12:02][OT][HEATING][WARN] Failed set target indoor temp
```

The protocol documentation describes these IDs and they should probably operate in this mode:
[image]

I added a commit with the set target indoor temp to CH2 (ID23), can you build the firmware via platformio and test it?
With Opentherm options:
[x] DHW present
[x] Summer/winter mode
[ ] Heating CH2 always enabled
[x]Duplicate heating CH1 to CH2
[ ] Sync modulation with heating
[x] Native heating control

nedievas, 2024-04-23:
> > Will leave every room temperatures fixed to 23 degrees for today. It's getting cold and I could not get target 23 temperature with equitherm and PID. Some rooms are as low as 20.4 and did not reach my desired temperature.
> 
> I recommend starting with choosing Equitherm coefficients. Equitherm + PID is needed in rare cases.

Excuse me for wrong expression. I tried Equitherm, then tried PID as @Pajuris recommended. Not both at the same time.

Laxilef, 2024-04-24:
> More INVALID statuses

Thx. Strange, I don't have any more ideas yet. There may be a problem with the Master MemberID code 🤷‍♂️

> Excuse me for wrong expression. I tried Equitherm, then tried PID as @Pajuris recommended. Not both at the same time.

Ok. Please describe in more detail what the problem is with Equitherm? Did you change the coefficients as described in the [Wiki](https://github.com/Laxilef/OTGateway/wiki#fit-coefficients)?

nedievas, 2024-04-24:
> Thx. Strange, I don't have any more ideas yet. There may be a problem with the Master MemberID code 🤷‍♂️
>
>
> Ok. Please describe in more detail what the problem is with Equitherm? Did you change the [...]

[image]
Red arrow points custom commit. 

I tried, but did not understand how to use this excel calc. The house does not reach desired temperature in a few hours, it's slow process... I just observed, that the temperature is not reaching the desired after I connected OT gateway and is getting down:
[image]

Suggestion: Is it possible to make commit that OT just gathers info about boiler functioning and not regulating it? Like Observation mode?

Laxilef, 2024-04-24:
> Suggestion: Is it possible to make commit that OT just gathers info about boiler functioning and not regulating it? Like Observation mode?

I don't think this is possible. On most boilers, when an opentherm device is connected, the boiler goes into slave mode.

> I tried, but did not understand how to use this excel calc. The house does not reach desired temperature in a few hours, it's slow process... I just observed, that the temperature is not reaching the desired after I connected OT gateway and is getting down:

You don't have to use an Excel table. The wiki page has text instructions on the coefficients. But I'm worried about your last temperature graph, it shouldn't change that much.

[image]
The first part of the graph looks +- normal, the second does not. What changed in the settings before the second part?

nedievas, 2024-04-24:
> ![img](https://private-user-images.githubusercontent.com/34578544/325331417-6341ef14-8c60-4d64-9dde-5b7403c88524.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5ODE4NDgsIm5iZiI6MTcxMzk4MTU0OCwicGF0aCI6Ii8zNDU3ODU0NC8zMjUzMzE0MTctNjM0MWVmMTQtOGM2MC00ZDY0LTlkZGUtNWI3NDAzYzg4NTI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI0VDE3NTkwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYxNjdlYmQ2NmE0NWUwNDcxZGY0YjNlMTVkYjUxNmYzYjRlOGI2ZGJkYjEzZWI4OTBlMTczYmZjZWE5NGFiYTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Wz45WjDHjJ8OllpLyq5q8uO7a1N1T82ZBuKpG6oVooM) The first part of the graph looks +- normal, the second does not. What changed in the settings before the second part?

I went from Equitherm to PID.

Laxilef, 2024-04-24:
But in your previous graph there was also a "saw" instead of a smooth line. Is this when using PID or equitherm?
> ![Ekrano kopija 2024-04-24 152823](https://private-user-images.githubusercontent.com/6176288/325218655-ac73f4d8-5206-47c1-8b5f-bfcfaaf57395.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5ODE3MjksIm5iZiI6MTcxMzk4MTQyOSwicGF0aCI6Ii82MTc2Mjg4LzMyNTIxODY1NS1hYzczZjRkOC01MjA2LTQ3YzEtOGI1Zi1iZmNmYWFmNTczOTUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjRUMTc1NzA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWU4ZDM1OTEyNTQ1NmJhMTAxZTJmNDM2N2RhNjY2ODAwZTRmYTMyOTY2YzMzMzNhYWI5ZGNmMTNiOGFkNTg1MyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.7h2pBG3sUMHeCQJEiKgaqVKEMD91vjRrBgV0mBKl-SA) 

Here is an example of a graph from my boiler:
[image]

nedievas, 2024-04-24:
[image]

This is today. The [last commit version]( https://github.com/Laxilef/OTGateway/issues/58#issuecomment-2072179724). Equitherm and PID off.

[x] DHW present
[x] Summer/winter mode
[ ] Heating CH2 always enabled
[x]Duplicate heating CH1 to CH2
[ ] Sync modulation with heating
[x] Native heating control

Outdoor sensor connected to boiler.

Laxilef, 2024-04-24:
Native heating control does not work with your boiler, because in the logs:
```
[00:03:36][OT][HEATING][INFO] Set current indoor temp: 21.50 (converted: 21.50)
[00:03:37][OT][TRACE] ID:   24   Request: 90181580   Response: f0180000   Attempt:  1   Status: INVALID
[00:03:37][OT][HEATING][WARN] Failed set current indoor temp
[00:03:37][OT][HEATING][INFO] Set target indoor temp: 30.00 (converted: 30.00)
[00:03:37][OT][TRACE] ID:   16   Request: 10101e00   Response: 70100000   Attempt:  1   Status: INVALID
[00:03:37][OT][HEATING][WARN] Failed set target indoor temp
[00:03:37][OT][TRACE] ID:   23   Request: 90171e00   Response: f0170000   Attempt:  1   Status: INVALID
[00:03:37][OT][HEATING][WARN] Failed set target indoor temp to CH2
```

The boiler does not accept the current room temperature and the room temperature setpoint. This may be due to the boiler settings (parameters in the service menu) or such an implementation of the opentherm protocol in the boiler.

The only solution I see is to use equitherm in the firmware. Opentherm options:
[x] DHW present
[x] Summer/winter mode **(don’t know if it’s necessary or not for your boiler, try both)**
[ ] Heating CH2 always enabled
[x] Duplicate heating CH1 to CH2
[x] Sync modulation with heating
[ ] Native heating control

nedievas, 2024-04-24:
Will test right now.

nedievas, 2024-04-24:
The graph is getting better (will update in the morning). But log is still not good:
```
Escape character is '^]'.
[00:40:14][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:40:14][OT][TRACE] ID:   33   Request:   210000   Response: c0210024   Attempt:  1   Status: SUCCESS
[00:40:15][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:40:15][MQTT][TRACE] Successfully publish 564 of 564 bytes to topic: opentherm/state
[00:40:15][OT][TRACE] ID:    0   Request:     3300   Response: 40003302   Attempt:  1   Status: SUCCESS
[00:40:15][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3800   Attempt:  1   Status: SUCCESS
[00:40:16][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:40:16][OT][TRACE] ID:   25   Request: 80190000   Response: 401932e6   Attempt:  1   Status: SUCCESS
[00:40:16][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:40:17][OT][TRACE] ID:   33   Request:   210000   Response: c0210022   Attempt:  1   Status: SUCCESS
[00:40:18][OT][TRACE] ID:    0   Request:     3300   Response: 40003302   Attempt:  1   Status: SUCCESS
[00:40:18][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3800   Attempt:  1   Status: SUCCESS
[00:40:18][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:40:19][OT][TRACE] ID:   25   Request: 80190000   Response: 40193280   Attempt:  1   Status: SUCCESS
[00:40:19][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:40:19][OT][TRACE] ID:   33   Request:   210000   Response: c0210021   Attempt:  1   Status: SUCCESS
[00:40:20][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:40:20][OT][TRACE] ID:    0   Request:     3300   Response: 40003302   Attempt:  1   Status: SUCCESS
[00:40:20][MQTT][TRACE] Successfully publish 564 of 564 bytes to topic: opentherm/state
[00:40:21][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3800   Attempt:  1   Status: SUCCESS
[00:40:21][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:40:21][OT][TRACE] ID:   25   Request: 80190000   Response: 401931b3   Attempt:  1   Status: SUCCESS
[00:40:21][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:40:22][OT][TRACE] ID:   33   Request:   210000   Response: c0210021   Attempt:  1   Status: SUCCESS
[00:40:23][OT][TRACE] ID:    0   Request:     3300   Response: 40003302   Attempt:  1   Status: SUCCESS
[00:40:23][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3800   Attempt:  1   Status: SUCCESS
[00:40:23][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:40:24][OT][TRACE] ID:   25   Request: 80190000   Response: 40193099   Attempt:  1   Status: SUCCESS
[00:40:24][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:40:24][OT][TRACE] ID:   33   Request:   210000   Response: c0210021   Attempt:  1   Status: SUCCESS
[00:40:26][OT][TRACE] ID:    0   Request:     3300   Response: 40003302   Attempt:  1   Status: SUCCESS
[00:40:26][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:40:26][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3800   Attempt:  1   Status: SUCCESS
[00:40:26][MQTT][TRACE] Successfully publish 564 of 564 bytes to topic: opentherm/state
[00:40:26][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:40:26][OT][TRACE] ID:   25   Request: 80190000   Response: c0192ee6   Attempt:  1   Status: SUCCESS
[00:40:27][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:40:27][OT][TRACE] ID:   33   Request:   210000   Response: c0210021   Attempt:  1   Status: SUCCESS
[00:40:27][MQTT][TRACE] Successfully publish 480 of 480 bytes to topic: opentherm/settings
[00:40:28][OT][TRACE] ID:    0   Request:     3300   Response: 40003302   Attempt:  1   Status: SUCCESS
[00:40:28][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3800   Attempt:  1   Status: SUCCESS
[00:40:29][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:40:29][OT][TRACE] ID:   25   Request: 80190000   Response: 40192d33   Attempt:  1   Status: SUCCESS
[00:40:29][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:40:30][OT][TRACE] ID:   33   Request:   210000   Response: c0210021   Attempt:  1   Status: SUCCESS
[00:40:31][OT][TRACE] ID:    0   Request:     3300   Response: 40003302   Attempt:  1   Status: SUCCESS
[00:40:31][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3800   Attempt:  1   Status: SUCCESS
[00:40:31][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:40:32][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:40:32][OT][TRACE] ID:   25   Request: 80190000   Response: c0192b19   Attempt:  1   Status: SUCCESS
[00:40:32][MQTT][TRACE] Successfully publish 564 of 564 bytes to topic: opentherm/state
[00:40:32][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:40:32][OT][TRACE] ID:   33   Request:   210000   Response: c0210021   Attempt:  1   Status: SUCCESS
[... cut ...]
[00:42:06][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3800   Attempt:  1   Status: SUCCESS
[00:42:06][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:42:06][OT][TRACE] ID:   25   Request: 80190000   Response: 40191833   Attempt:  1   Status: SUCCESS
[00:42:06][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:42:07][OT][TRACE] ID:   33   Request:   210000   Response: c021001b   Attempt:  1   Status: SUCCESS
```

Laxilef, 2024-04-24:
ID 8 - CH2 setpoint
ID 18 - pressure
ID 19 - dhw flow rate
ID 28 - return heating temp
ID 57 - max heating temp

This is not critical and does not affect work. Duplicate heating CH1 to CH2 option can be disabled.

nedievas, 2024-04-25:
> ID 8 - CH2 setpoint ID 18 - pressure ID 19 - dhw flow rate ID 28 - return heating temp ID 57 - max heating temp
> 
> This is not critical and does not affect work. Duplicate heating CH1 to CH2 option can be disabled.

Here is the graph from night til this morning: 
[image]

Still saw like graph...
Will disable `Duplicate heating CH1 to CH2` option.

I have invalid ID 19 in logs, but I have some output in this state:
[image]

Though I do not believe in this meter. I have like 28322,59 L of hot water usage for that period. It's unrealistic. 28 cubic meters? :) Impossible.

ID 28:
[image]
Invalid?

ID 18 is disabled in my HA as it's always 0.

Laxilef, 2024-04-25:
> Though I do not believe in this meter. I have like 28322,59 L of hot water usage for that period. It's unrealistic. 28 cubic meters? :) Impossible.

Well, this is really unusual :) Don't you have an indirect heating boiler?

> Here is the graph from night til this morning: ![paveikslas](https://private-user-images.githubusercontent.com/6176288/325534977-5946f967-3248-486f-8d5d-3773c8d08c09.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQwMzk2ODksIm5iZiI6MTcxNDAzOTM4OSwicGF0aCI6Ii82MTc2Mjg4LzMyNTUzNDk3Ny01OTQ2Zjk2Ny0zMjQ4LTQ4NmYtOGQ1ZC0zNzczYzhkMDhjMDkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjVUMTAwMzA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDc3YzVhMGQ5NDAyOWYyMTUzODg5MTdkNDkzYTJlZTI3MzRmYWY1ZGViNDQwYzUwOWI1YmMxZTM4MmMxZWMxNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.KN6I28FkCPzwK690RTtqzkGDDiK78f8WdsF6hIq6ZBE)
> 
> Still saw like graph...

What about the indoor temperature? Perhaps this is normal for your boiler...

> ID 28:
> ![paveikslas](https://private-user-images.githubusercontent.com/6176288/325537559-9251e1ca-f157-4ba2-8835-1afbbf20f8de.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQwMzk2ODksIm5iZiI6MTcxNDAzOTM4OSwicGF0aCI6Ii82MTc2Mjg4LzMyNTUzNzU1OS05MjUxZTFjYS1mMTU3LTRiYTItODgzNS0xYWZiYmYyMGY4ZGUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjVUMTAwMzA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTA0ODQ4YzA0MWZjMTMyNTI5OTIzMzI4NGNiNjMyNThkNmFlYWIwMDg2MzQ1YzhjOTQ0NmNmZmZiMzRkM2YyMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.wnDKIyg_DQxOGa7oJUfbP6WzQYKNeQaZJ_yXthTS3x4)
> Invalid?

A temperature drop to 0 on the graph is unusual, but the rest of the time it stays at 50 degrees. What could this temperature mean? Perhaps the temperature is displayed only when the pump is running?

nedievas, 2024-04-25:
> > Though I do not believe in this meter. I have like 28322,59 L of hot water usage for that period. It's unrealistic. 28 cubic meters? :) Impossible.
> 
> Well, this is really unusual :) Don't you have an indirect heating boiler?
> 

I suppose it is indirect:)

> > Here is the graph from night til this morning: ![paveikslas](https://private-user-images.githubusercontent.com/6176288/325534977-5946f967-3248-486f-8d5d-3773c8d08c09.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQwMzk2ODksIm5iZiI6MTcxNDAzOTM4OSwicGF0aCI6Ii82MTc2Mjg4LzMyNTUzNDk3Ny01OTQ2Zjk2Ny0zMjQ4LTQ4NmYtOGQ1ZC0zNzczYzhkMDhjMDkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjVUMTAwMzA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDc3YzVhMGQ5NDAyOWYyMTUzODg5MTdkNDkzYTJlZTI3MzRmYWY1ZGViNDQwYzUwOWI1YmMxZTM4MmMxZWMxNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.KN6I28FkCPzwK690RTtqzkGDDiK78f8WdsF6hIq6ZBE)
> > Still saw like graph...
> 
> What about the indoor temperature? Perhaps this is normal for your boiler... Why is the setpoint temperature 30 degrees all this time? Is equitherm disabled?

Yes. Because I have outdoor sensor connected to the boiler and by this the boiler is working in "curve mode". Equitherm is disabled.

> A temperature drop to 0 on the graph is unusual, but the rest of the time it stays at 50 degrees. What could this temperature mean? Perhaps the temperature is displayed only when the pump is running?

ID 28 is Invalid in logs. It is `return heating temp` and the picture was for it. Why is it invalid if we get some numbers, as also  in ID 19 - DHW flow rate?

nedievas, 2024-04-26:
Equiterm from the red arrow. Lowered target temperature from 30 to 26 from the blue arrow:
[image]
 
Telnet logs are the same:
```
Escape character is '^]'.
[23:54:50][OT][TRACE] ID:   25   Request: 80190000   Response: 40193419   Attempt:  1   Status: SUCCESS
[23:54:51][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[23:54:51][OT][TRACE] ID:   33   Request:   210000   Response: c021002e   Attempt:  1   Status: SUCCESS
[23:54:52][OT][TRACE] ID:    0   Request: 80002300   Response: c0002302   Attempt:  1   Status: SUCCESS
[23:54:52][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3733   Attempt:  1   Status: SUCCESS
[23:54:53][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[23:54:53][OT][TRACE] ID:   25   Request: 80190000   Response: 40193419   Attempt:  1   Status: SUCCESS
[23:54:53][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[23:54:54][OT][TRACE] ID:   33   Request:   210000   Response: c021002e   Attempt:  1   Status: SUCCESS
[23:54:54][OT][HEATING][INFO] Set temp: 48.54 (converted: 48.54)
[23:54:54][OT][TRACE] ID:   57   Request: 90393000   Response: f0390000   Attempt:  1   Status: INVALID
[23:54:54][OT][HEATING][WARN] Failed set max heating temp
[23:54:54][OT][TRACE] ID:    1   Request: 90013089   Response: 50013080   Attempt:  1   Status: SUCCESS
[23:54:55][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[23:54:55][MQTT][TRACE] Successfully publish 575 of 575 bytes to topic: opentherm/state
[23:54:55][OT][TRACE] ID:    0   Request: 80002300   Response: c0002302   Attempt:  1   Status: SUCCESS
[23:54:56][MQTT][MSG][TRACE] Topic: opentherm/state/set
>  {"temperatures": {"indoor":23.43}}

[23:54:56][MQTT][TRACE] Successfully publish 0 of 0 bytes to topic: opentherm/state/set
[23:54:56][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3733   Attempt:  1   Status: SUCCESS
[23:54:56][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[23:54:56][OT][TRACE] ID:   25   Request: 80190000   Response: 40193419   Attempt:  1   Status: SUCCESS
[23:54:56][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[23:54:57][OT][TRACE] ID:   33   Request:   210000   Response: c021002e   Attempt:  1   Status: SUCCESS
[23:54:57][MQTT][TRACE] Successfully publish 479 of 479 bytes to topic: opentherm/settings
[23:54:58][OT][TRACE] ID:    0   Request: 80002300   Response: c0002302   Attempt:  1   Status: SUCCESS
[23:54:58][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3733   Attempt:  1   Status: SUCCESS
[23:54:58][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[23:54:59][MAIN][VERB] Free heap size: 25520 of 81920 bytes (min: 17736, diff: 0), max free block: 20936 (min: 15904, diff: 0, frag: 17%)
[23:54:59][OT][TRACE] ID:   25   Request: 80190000   Response: 40193419   Attempt:  1   Status: SUCCESS
[23:54:59][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[23:54:59][OT][TRACE] ID:   33   Request:   210000   Response: c021002e   Attempt:  1   Status: SUCCESS
[23:55:00][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[23:55:00][OT][TRACE] ID:    0   Request: 80002300   Response: c0002302   Attempt:  1   Status: SUCCESS
[23:55:01][MQTT][TRACE] Successfully publish 575 of 575 bytes to topic: opentherm/state
[23:55:01][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3733   Attempt:  1   Status: SUCCESS
[23:55:01][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[23:55:01][OT][TRACE] ID:   25   Request: 80190000   Response: 40193419   Attempt:  1   Status: SUCCESS
[23:55:02][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[23:55:02][OT][TRACE] ID:   33   Request:   210000   Response: c021002e   Attempt:  1   Status: SUCCESS
[23:55:03][OT][TRACE] ID:    0   Request: 80002300   Response: c0002302   Attempt:  1   Status: SUCCESS
[23:55:03][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3733   Attempt:  1   Status: SUCCESS
[23:55:04][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[23:55:04][OT][TRACE] ID:   25   Request: 80190000   Response: c0193400   Attempt:  1   Status: SUCCESS
[23:55:04][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[23:55:05][OT][TRACE] ID:   33   Request:   210000   Response: c021002e   Attempt:  1   Status: SUCCESS
[23:55:06][OT][TRACE] ID:    0   Request: 80002300   Response: c0002302   Attempt:  1   Status: SUCCESS
[23:55:06][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3733   Attempt:  1   Status: SUCCESS
[23:55:06][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[23:55:06][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[23:55:06][MQTT][TRACE] Successfully publish 573 of 573 bytes to topic: opentherm/state
[23:55:06][OT][TRACE] ID:   25   Request: 80190000   Response: c0193400   Attempt:  1   Status: SUCCESS
[23:55:07][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[23:55:07][OT][TRACE] ID:   33   Request:   210000   Response: c021002e   Attempt:  1   Status: SUCCESS
[... cut ...]
[23:56:01][OT][TRACE] ID:   26   Request: 801a0000   Response: c01a3733   Attempt:  1   Status: SUCCESS
[23:56:01][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[23:56:02][OT][TRACE] ID:   25   Request: 80190000   Response: c01933b3   Attempt:  1   Status: SUCCESS
```

[x] DHW present
[x] Summer/winter mode
[ ] Heating CH2 always enabled
[ ]Duplicate heating CH1 to CH2
[x] Sync modulation with heating
[ ] Native heating control

Outdoor sensor connected to boiler.

Now the indoor temperature is rising and the comfort returned to my house :D
[image]

Laxilef, 2024-04-26:
> Equiterm from the red arrow. Lowered target temperature from 30 to 26 from the blue arrow:
> ![Ekrano kopija 2024-04-25 231122](https://private-user-images.githubusercontent.com/6176288/325902741-ad82532a-f30f-4f24-8a88-2c1794c49b15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQxNTI4NjUsIm5iZiI6MTcxNDE1MjU2NSwicGF0aCI6Ii82MTc2Mjg4LzMyNTkwMjc0MS1hZDgyNTMyYS1mMzBmLTRmMjQtOGE4OC0yYzE3OTRjNDliMTUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjZUMTcyOTI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OWVlN2ZhNzZmNzA0MzU5MWQ0N2FkOTQ5M2U2MDFjNzczOTVmNjlhMjE2YWI3MGI3ZmVlZjIwZjA1OWFjOGU0ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.CZxjIW_7BO8xkbZHTzYOtYI1ncaYvlSILT0txc9M8YU)

This looks like normal for equitherm :)

> Outdoor sensor connected to boiler.
> Now the indoor temperature is rising and the comfort returned to my house :D

equitherm = off?

nedievas, 2024-04-26:
> This looks like normal for equitherm :)
>
>
> equitherm = off?

No. Equitherm N=0.7; K=2; T= (was 6, now 4);

Laxilef, 2024-04-26:
> Outdoor sensor connected to boiler.

> No. Equitherm N=0.7; K=2; T= (was 6, now 4);

Hmm. How does this work with outdoor temperature sensor connected to the boiler?

Earlier you wrote:
> To disable this mode on the boiler I have to detach the outside temperature sensor from boiler. The boiler then automatically sets higher temperature for heating, as per manual.

nedievas, 2024-04-27:
> Hmm. How does this work with outdoor temperature sensor connected to the boiler?
>
> Earlier you wrote:

You asked to build the commit with PLATFORMIO and test it. Later you recommended to disable Native heating control.

nedievas, 2024-04-27:
Current work of the boiler:
[image]

nedievas, 2024-04-29:
ERROR management is not working.

Situation:
This evening during preparation of my kid for a bath I felt that hot water is just barely warm. Checked the HA:
[image]
[image]
[image]

Boiler displayed `F2` error code.

Tried to reset with active fault reset button:
[image]

No Luck! Had to go downstairs and push `R` on the boiler display to make that reset.

nedievas, 2024-04-29:
ERROR management is not working.

Situation:
This evening during preparation of my kid for a bath I felt that hot water is just barely warm. Checked the HA:
[image]
[image]
[image]

Boiler displayed `F2` error code.

Tried to reset with active fault reset button:
[image]

No Luck! Had to go downstairs and push `R` on the boiler display to make that reset.

Temperature log showed the spike in the temperature of  90,1°C that triggered the error and boiler protective shutdown:
[image]

Laxilef, 2024-05-05:
Hi @nedievas 
I assume that resetting errors using opentherm does not work with your boiler or need to use other, undocumented IDs for this.
I don’t know why there are so many problems with this boiler, although everything works fine for @Pajuris with a similar boiler. You might want to try to figure this out yourself. Let me know if there is any news.

nedievas, 2024-05-09:
Caught some FAULT log:
```
Connected to 192.168.20.94.
Escape character is '^]'.
[00:01:35][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:01:35][OT][TRACE] ID:   33   Request:   210000   Response: c0210027   Attempt:  1   Status: SUCCESS
[00:01:36][OT][TRACE] ID:    0   Request: 80002300   Response: 40002305   Attempt:  1   Status: SUCCESS
[00:01:37][OT][TRACE] ID:   26   Request: 801a0000   Response: 401a2799   Attempt:  1   Status: SUCCESS
[00:01:37][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:01:37][OT][TRACE] ID:   25   Request: 80190000   Response: c0193080   Attempt:  1   Status: SUCCESS
[00:01:38][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:01:38][OT][TRACE] ID:   33   Request:   210000   Response: c0210027   Attempt:  1   Status: SUCCESS
[00:01:39][OT][TRACE] ID:    0   Request: 80002300   Response: 40002305   Attempt:  1   Status: SUCCESS
[00:01:39][OT][TRACE] ID:   26   Request: 801a0000   Response: 401a2799   Attempt:  1   Status: SUCCESS
[00:01:40][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:01:40][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:01:40][MQTT][TRACE] Successfully publish 569 of 569 bytes to topic: opentherm/state
[00:01:40][OT][TRACE] ID:   25   Request: 80190000   Response: c0193080   Attempt:  1   Status: SUCCESS
[00:01:40][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:01:40][OT][TRACE] ID:   33   Request:   210000   Response: c0210027   Attempt:  1   Status: SUCCESS
[00:01:41][OT][TRACE] ID:    0   Request: 80002300   Response: 40002305   Attempt:  1   Status: SUCCESS
[00:01:42][OT][TRACE] ID:   26   Request: 801a0000   Response: 401a2799   Attempt:  1   Status: SUCCESS
[00:01:42][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:01:42][OT][TRACE] ID:   25   Request: 80190000   Response: c0193080   Attempt:  1   Status: SUCCESS
[00:01:43][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:01:43][OT][TRACE] ID:   33   Request:   210000   Response: c0210027   Attempt:  1   Status: SUCCESS
[00:01:44][OT][TRACE] ID:    0   Request: 80002300   Response: 40002305   Attempt:  1   Status: SUCCESS
[00:01:44][OT][TRACE] ID:   26   Request: 801a0000   Response: 401a2799   Attempt:  1   Status: SUCCESS
[00:01:45][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:01:45][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:01:45][OT][TRACE] ID:   25   Request: 80190000   Response: c0193080   Attempt:  1   Status: SUCCESS
[00:01:45][MQTT][TRACE] Successfully publish 569 of 569 bytes to topic: opentherm/state
[00:01:45][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:01:46][OT][TRACE] ID:   33   Request:   210000   Response: c0210027   Attempt:  1   Status: SUCCESS
[00:01:47][OT][TRACE] ID:    0   Request: 80002300   Response: 40002305   Attempt:  1   Status: SUCCESS
[00:01:47][OT][TRACE] ID:   26   Request: 801a0000   Response: 401a2799   Attempt:  1   Status: SUCCESS
[00:01:47][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:01:48][OT][TRACE] ID:   25   Request: 80190000   Response: c0193080   Attempt:  1   Status: SUCCESS
[00:01:48][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:01:48][OT][TRACE] ID:   33   Request:   210000   Response: c0210027   Attempt:  1   Status: SUCCESS
[00:01:49][OT][TRACE] ID:    0   Request: 80002300   Response: 40002305   Attempt:  1   Status: SUCCESS
[00:01:50][OT][TRACE] ID:   26   Request: 801a0000   Response: 401a2799   Attempt:  1   Status: SUCCESS
[00:01:50][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:01:50][OT][TRACE] ID:   25   Request: 80190000   Response: c0193080   Attempt:  1   Status: SUCCESS
[00:01:50][MAIN][VERB] Free heap size: 24344 of 81920 bytes (min: 19936, diff: 0), max free block: 17304 (min: 16008, diff: 0, frag: 28%)
[00:01:51][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:01:51][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
[00:01:51][OT][TRACE] ID:   33   Request:   210000   Response: c0210027   Attempt:  1   Status: SUCCESS
[00:01:51][MQTT][TRACE] Successfully publish 569 of 569 bytes to topic: opentherm/state
[00:01:52][OT][TRACE] ID:    0   Request: 80002300   Response: 40002305   Attempt:  1   Status: SUCCESS
[00:01:52][OT][TRACE] ID:   26   Request: 801a0000   Response: 401a2799   Attempt:  1   Status: SUCCESS
[00:01:53][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:01:53][OT][TRACE] ID:   25   Request: 80190000   Response: c0193080   Attempt:  1   Status: SUCCESS
[00:01:53][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:01:54][OT][TRACE] ID:   33   Request:   210000   Response: c0210027   Attempt:  1   Status: SUCCESS
[00:01:55][OT][TRACE] ID:    0   Request: 80002300   Response: 40002305   Attempt:  1   Status: SUCCESS
[... cut ...]
[00:03:49][OT][TRACE] ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID
[00:03:49][OT][TRACE] ID:   25   Request: 80190000   Response: 40193066   Attempt:  1   Status: SUCCESS
[00:03:49][OT][TRACE] ID:   28   Request: 801c0000   Response: 701c0000   Attempt:  1   Status: INVALID
[00:03:50][OT][TRACE] ID:   33   Request:   210000   Response: c0210027   Attempt:  1   Status: SUCCESS
[00:03:51][MQTT][TRACE] Successfully publish 6 of 6 bytes to topic: opentherm/status
```
ID 4 fails?

Laxilef, 2024-05-10:
Yes. ID 4 is used to reset the fault, but the boiler does not accept data.

nedievas, 2024-05-27:
I suppose it is as it is. I think `fault reset` is not supported for the majority of boilers [link](https://otgw.tclcode.com/matrix.cgi#boilers)

Laxilef, 2024-05-27:
Nope. In the table, column ID4 is empty for most boilers, that is ID4 was not tested. This does not mean that this feature does not work for these boilers.

closed 2024-06-16 by github-actions[bot]