Laxilef/OTGateway #59: Build is broken: lib/WiFiManagerParameters/WiFiManagerParameters.h:33:15: error: 'const char* CheckboxParameter::getValue() const' marked 'override', but does not override

jandebleser, 2024-04-16:
On version 1.3.3:
I'm getting this error:
`  PackageException: Got the unrecognized status code '404' when downloaded https://github.com/Laxilef/WiFiManager/archive/refs/heads/patch-1.zip`

Replacing the line in lib_deps with https://github.com/tzapu/WiFiManager.git#v2.0.17 gives this error:

`lib/WiFiManagerParameters/WiFiManagerParameters.h:33:15: error: 'const char* CheckboxParameter::getValue() const' marked 'override', but does not override
`
It seems your PR did not make it into master?
https://github.com/tzapu/WiFiManager/pull/1662

I added my own fork with the change you made: 
https://github.com/jandebleser/WiFiManager.git#virtualGetValue
Now it's building again.

jandebleser, 2024-04-16:
Nevermind, I was on a detached head somehow.

closed 2024-04-16 by jandebleser

Laxilef, 2024-04-16:
Please use the master branch. Version 1.3.3 is old and not supported.