Laxilef/OTGateway #61: [noob question] can't connect to wifi

papinist, 2024-05-01:
Hi,
I know this sound as a stupid question, but I'm stuck in initial wifi setup and I cannot go past this.

I'm using a D1 mini (tried a D1 mini32 too, but I am not able to put it in bootloader mode so I can't flash it..) and a Diyless thermostat shield, Flashed with esp flash download tool, powered it and the opentherm gateway AP appears.
Connect with my phone and the main page appears. I then go to network page, it finds my wifi network (Home_wifi); now, touching network names does nothing (don't know if it is supposed to do something..), so I write the ssid in the corresponding field and the password. Click Save and wait.. opentherm AP goes away (so I think it is connected to my home wifi) but checking my router, I can't find it.
After some time the opentherm AP shows up again, reconnect with my phone and in the main page it reports my wifi ssid with signal 0%.

I think I'm missing something but can't figure out what it is...

[image]

[image]

Laxilef, 2024-05-05:
Hi
Do you use a 5 GHz network? If it is named the same as the 2 GHz network, then this will not work. You need to name the networks differently, for example Home_wifi_2g and Home_wifi_5g. This is due to the outdated SDK for ESP8266, which will probably never support 5 GHz networks.

papinist, 2024-05-05:
Thank you for your reply!
Yes, this exactly my case, I have a fritz!box and it defaults both 2.4 and 5ghz networks with the same name.
But I already have about 70 wifi devices, mostly esp8266 with tasmota or older shelly devices, or other d1 minis, and never had issues connecting to this same wifi network.
Is this somewhat different?

Laxilef, 2024-05-05:
Hm. No, Tasmota also uses the NonOS SDK and the Arduino framework. You can try to build the firmware with the ``PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH`` flag instead of ``PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY`` (platformio.ini), but I have not tested this. Users have previously reported network problems on the ESP8266 when using 2G and 5G networks with the same name. What channel is used for the 2G network?

papinist, 2024-05-05:
Channel management in my fritzbox is set to auto, at this moment 2.4 ghz is using channel 1 and 5 ghz channel 36.

I'm not sure if I am able to build a firmware from scratch, my knowledge about that is very limited :-O

Laxilef, 2024-05-06:
Channel 1 - ok. Problems may occur on channels 12-14. Hmm, strange :) What is your board? Please write the exact name.

papinist, 2024-05-09:
It's a simple D1 mini bought on aliexpress. I have other boards of the same batch that I'm using in other projects (clocks and similar) that are working.
I tried many times the password, it's 10 digits all numbers so it's easy to remember.
No WPA3 here.. but as I said, I have all blinds (12) controlled by esp8266 boards (4-relay esp-12f boards), a dozen sonoff basici\mini for lights and for garden valves and many other 'smart' appliances mostly using esp8266 chip and never had issues.. so this is really strange.

[image]
[image]

Laxilef, 2024-05-10:
Ok, pls test this firmware with ``PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH``
[build_HIGHER_BANDWIDTH.zip](https://github.com/Laxilef/OTGateway/files/15269413/build_HIGHER_BANDWIDTH.zip)

If this does not help, could you try flashing any other firmware (tasmota for example) and try connecting to Wifi?

papinist, 2024-05-13:
Thanks!
Now I can't even flash it.. always get "com fail" from flash download tool... is there any special procedure to put the d1 mini in flash mode? I only have a 'reset' button but it seems to do nothing.

Laxilef, 2024-05-15:
No, 8266 are flashed normally without pressing any buttons. Upload a screenshot of the error.

SanFable, 2024-06-01:
Hello,

I had same problem with 1.4 RC versions, I tried different ESPs (32 and 8266) and finally discovered that v1.3.3 is working flawlessly, I'm on WPA3.

When I was connecting to wifi, following error popped out in serial logs:

ESP32 WIFI CONNECTION FAILED REASON 6

PS thank you for great project :)

Laxilef, 2024-06-02:
Hi @SanFable 
Hm, very very strange, because SDK for esp8266 [does not support wpa3](https://github.com/esphome/issues/issues/3971)
Can you enable debug on 1.4.0, save logs to a file and upload?

SanFable, 2024-06-02:
My bad, Thank you for that info, by luck I finally landed on WPA2 for IoT network.
I will try to gather logs tomorrow

Laxilef, 2024-06-02:
Thx. WPA3 on esp32 with SDK 3.0.0 should work fine, I'll try to compile and upload the build for testing this week in this issue.

SanFable, 2024-06-03:
Hello @Laxilef,

I have gathered the logs, I have used esp32 nodemcu, two scenarios, 
[Esp32WifiConnect_SpecifiedWifiSettings.txt](https://github.com/user-attachments/files/15533513/Esp32WifiConnect_SpecifiedWifiSettings.txt)
[Esp32WifiConnect_SpecifiedWifiAndStaticAddressSettings.txt](https://github.com/user-attachments/files/15533511/Esp32WifiConnect_SpecifiedWifiAndStaticAddressSettings.txt)

First one is first reboot after flash, Im connecting to AP, debug options are set already, going into wifi settings, writing SSID, password and channel.

In second logs I'm filling out the dhcp settings (but with marked DHCP anyway), I have tried without DHCP without any other result, just wanted to include two logs because I saw different error once:
Connection failed. Status: 3, reason: 7 vs Connection failed. Status: 2, reason: 6

Flashed factory file with filesystem
[image]

papinist, 2024-06-07:
> No, 8266 are flashed normally without pressing any buttons. Upload a screenshot of the error. As a workaround, you can update the firmware via OTA.

Sorry for the late reply.. I missed the "new message" notification.

This is the message I got:

[image]

SanFable, 2024-06-07:
@papinist check if you have installed drivers for built usb->serial adapter

@Laxilef Sorry for late reply, I have tried without success, after that I pressed reset button and same thing, 
[NETWORK][TRACE] Connection failed. Status: 2, reason: 6
logs below
[esp32newsdk.txt](https://github.com/user-attachments/files/15745013/esp32newsdk.txt)

papinist, 2024-06-07:
> @papinist check if you have installed drivers for built usb->serial adapter and you are selecting correct COM device

Yes I have already flashed this exact 8266 with this pc and this program, and the COM port is correct.

Laxilef, 2024-06-08:
@papinist try run the flasher as admin and SPI MODE = DIO
@SanFable no idea... try this build 
[nodemcu_32s_core_3+wifi_auto_policy.zip](https://github.com/user-attachments/files/15748344/nodemcu_32s_core_3%2Bwifi_auto_policy.zip)

papinist, 2024-06-09:
> @papinist try run the flasher as admin and SPI MODE = DIO

Already tried as admin, and with SPI MODE = DIO nothing changes

SanFable, 2024-06-09:
@Laxilef 
LOL, I got it!
even I have tried standard 1.4.0 from release section and its working
You won't believe...

The problem is password forwarding from the form.

On my UniFi-IoT network I have password which has 10 numbers, starting with 0, like: **0**123123123
After i pass the wifi details in logs I see that only 123123123 is sent, so i get error like wrong password.

I have created another wifi network with any other random password like noobnoob123 it got connected.

So there must be something what got changed after 1.3.3 version regarding form and first password character (in my case) 0 is getting lost

@papinist Maybe its same issue in your case?

papinist, 2024-06-09:
> @papinist Maybe its same issue in your case?

my password is starting with "0" too!

Laxilef, 2024-06-11:
Hi guys
Thanks for researching the problem, it will be fixed in 1.4.1

Laxilef, 2024-06-11:
@papinist don’t know why you get an error when flashing, try [nodemcu-flasher](https://github.com/nodemcu/nodemcu-flasher) for esp8266

Laxilef, 2024-06-13:
Pls test build ver 1.4.1

papinist, 2024-06-13:
> @papinist don’t know why you get an error when flashing, try [nodemcu-flasher](https://github.com/nodemcu/nodemcu-flasher) for esp8266

Thank you, I finally managed to flash 1.4.1.. but same as before, after entering wifi password, I got disconnected from AP and then otgateway appears again, connecting to it I still see a red dot right to "connected" and signal 0%

Laxilef, 2024-06-13:
> Thank you, I finally managed to flash 1.4.1.. but same as before, after entering wifi password, I got disconnected from AP and then otgateway AP appears again, connecting to it I still see a red dot right to "connected" and signal 0%

I think the problem may be with your access point settings. I just tested version 1.4.1 on Mikrotik with WPA2/WPA3 and only WPA3 with password 00012345678.
Esp32 connects successfully. I also see in the logs that the password is saved correctly. I remind you that Esp8266 will not work with WPA3.

SanFable, 2024-06-14:
@Laxilef 
Confirming, 1.4.1 successfully connected to my UniFi-IoT network.

Thank you very much for fix :)

papinist, 2024-06-14:
> I think the problem may be with your access point settings. I just tested version 1.4.1 on Mikrotik with WPA2+WPA3 and only WPA3 with password 00012345678. Esp32 connects successfully. I also see in the logs that the password is saved correctly. I remind you that Esp8266 will not work with WPA3 and will not work well with WPA2+WPA3.

I'm using a fritz!box 7530 with 3 repeater configured in a mesh network. I'm using only wpa2 and at this moment I have 76 devices connected, of which 27 are tasmotized devices, so sonoff or other board with esp8266 and a couple of esp32.
I really don't know what could be wrong only with this.

Laxilef, 2024-06-15:
Can you take a screenshot of your all wifi settings on fritz!box? Without passwords, of course.
In version 1.4.2 I added "disconnect raw reason" to the logs, please look at the number and write here.

Laxilef, 2024-06-29:
No details, removing the bug label and close issue.

commit 65cabf25 (2024-06-11): fix: casting types in setupForm fixed #61
commit bb9b6a5f (2024-06-11): fix: casting types in setupForm fixed #61
closed 2024-06-29 by Laxilef