Laxilef/OTGateway #68: Cannot compile any 8266 configuration.  ESP32 ok.

dee-mos, 2024-06-12:
It's a master branch archive  (https://github.com/Laxilef/OTGateway/archive/refs/heads/master.zip)

```
Executing task: C:\Users\user\.platformio\penv\Scripts\platformio.exe run --environment d1_mini_pro 

Warning! Ignore unknown configuration option `version` in section [env]
Processing d1_mini_pro (platform: espressif8266; board: d1_mini_pro; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini_pro.html
PLATFORM: Espressif 8266 (2.7.4+20221113) > WeMos D1 mini Pro
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 2.7.4+9
 - tool-esptoolpy @ 1.40301.221017 (4.3.1)
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 48 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 7.0.4
|-- OpenTherm Library @ 1.1.5+sha.dab2c13
|-- ArduinoMqttClient @ 0.1.8
|-- ESP Telnet @ 2.2.1
|-- FileData @ 1.0.2
|-- GyverPID @ 3.3.2
|-- GyverBlinker @ 1.1.0
|-- OneWire @ 2.3.8+sha.72249e2
|-- DallasTemperature @ 3.11.0
|-- TinyLogger @ 1.1.0
|-- ESP8266Scheduler @ 1.2.0
|-- LittleFS(esp8266) @ 0.1.0
|-- NetworkUtils
|-- MqttWriter
|-- CustomOpenTherm
|-- BufferedWebServer
|-- DNSServer @ 1.1.1
|-- WebServerHandlers
|-- ESP8266WebServer @ 1.0
|-- Equitherm
|-- HomeAssistantHelper
Building in release mode
Compiling .pio\build\d1_mini_pro\src\main.cpp.o
Compiling .pio\build\d1_mini_pro\libe9f\ESP Telnet\ESPTelnetBase.cpp.o
Compiling .pio\build\d1_mini_pro\lib797\ESP8266Scheduler\Scheduler.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\Print.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\Schedule.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\StackThunk.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\TypeConversion.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\Updater.cpp.o
Compiling .pio\build\d1_mini_pro\FrameworkArduino\WMath.cpp.o
In file included from .pio\libdeps\d1_mini_pro\ESP8266Scheduler\src\Scheduler.cpp:2:0:
.pio\libdeps\d1_mini_pro\ESP8266Scheduler\src\Task.h: In member function 'virtual void Task::yield()':
.pio\libdeps\d1_mini_pro\ESP8266Scheduler\src\Task.h:21:26: error: 'cont_suspend' was not declared in this scope
     cont_suspend(&context);
                          ^
Compiling .pio\build\d1_mini_pro\FrameworkArduino\WString.cpp.o
*** [.pio\build\d1_mini_pro\lib797\ESP8266Scheduler\Scheduler.cpp.o] Error 1
.pio\libdeps\d1_mini_pro\ESP Telnet\src\ESPTelnetBase.cpp: In member function 'void ESPTelnetBase::processClientConnection()':
.pio\libdeps\d1_mini_pro\ESP Telnet\src\ESPTelnetBase.cpp:39:32: error: 'using TCPServer = class WiFiServer' has no member named 'accept'
   TCPClient newClient = server.accept();
                                ^
*** [.pio\build\d1_mini_pro\libe9f\ESP Telnet\ESPTelnetBase.cpp.o] Error 1
In file included from src\main.cpp:9:0:
lib\NetworkUtils/NetworkMgr.h: In member function 'bool NetworkUtils::NetworkMgr::resetWifi()':
lib\NetworkUtils/NetworkMgr.h:178:34: error: no matching function for call to 'ESP8266WiFiClass::disconnect(bool, bool)'
       WiFi.disconnect(false, true);
[... 123 lines cut ...]
         const bss_info* info = WiFi.getScanInfoByIndex(i);
                                     ^
src\PortalTask.h: In lambda function:
src\PortalTask.h:527:48: error: 'esp_is_8285' was not declared in this scope
       doc["system"]["chipModel"] = esp_is_8285() ? "ESP8285" : "ESP8266";
                                                ^
In file included from src\main.cpp:8:0:
.pio\libdeps\d1_mini_pro\TinyLogger\src/TinyLogger.h: In instantiation of 'void TinyLogger::printf(const __FlashStringHelper*, Args ...) [with Args = {const char*, String}]':
.pio\libdeps\d1_mini_pro\TinyLogger\src/TinyLogger.h:299:5:   required from 'void TinyLogger::printFormatted(TinyLogger::Level, ST, bool, MT, Args ...) [with ST = const __FlashStringHelper*; MT = const __FlashStringHelper*; Args = {const char*, String}]'
.pio\libdeps\d1_mini_pro\TinyLogger\src/TinyLogger.h:343:5:   required from 'void TinyLogger::serrorln(ST, MT, Args ...) [with ST = const __FlashStringHelper*; MT = const __FlashStringHelper*; Args = {const char*, String}]'
lib\WebServerHandlers/UpgradeHandler.h:198:112:   required from here
.pio\libdeps\d1_mini_pro\TinyLogger\src/TinyLogger.h:184:9: error: cannot pass objects of non-trivially-copyable type 'class String' through '...'
         stream->printf_P(reinterpret_cast<PGM_P>(msg), args...);
         ^
In file included from src\main.cpp:9:0:
lib\NetworkUtils/NetworkMgr.h: In member function 'bool NetworkUtils::NetworkMgr::setWifiHostname(const char*)':
lib\NetworkUtils/NetworkMgr.h:400:5: warning: control reaches end of non-void function [-Wreturn-type]
     }
     ^
*** [.pio\build\d1_mini_pro\src\main.cpp.o] Error 1
=========================================================================== [FAILED] Took 6.33 seconds ===========================================================================
Environment    Status    Duration
-------------  --------  ------------
d1_mini_pro    FAILED    00:00:06.325
```

Laxilef, 2024-06-13:
> framework-arduinoespressif8266 @ 2.7.4+9

You need to update the framework. Look at the platformio panel under Platforms > Updates.

dee-mos, 2024-06-13:
У меня была такая мысль, но именно в Platform -> Updates ничего не предлагалось. И какие-то апдейты накатывались во время запуска PlatformIO и первой компиляции...
А проблема решилась только ручным удалением старой платформы тут: Paltforms -> Installed- > Espressif 8266  (2.7.4)
с последующей установкой уже самой свежей Espressif 8266 (4.2.1)

Laxilef, 2024-06-13:
Уже внёс изменения в platformio.ini, должно нормально теперь подтягивать версию фреймворка 4.2.1 при сборке прошивки.

closed 2024-06-19 by dee-mos