Laxilef/OTGateway #69: BOM.xlsx, не хватает всех Manufacturer Part

RasimKalimullin, 2024-06-14:
Приветствую! хотел заказать pcb с монтажом, но по led(ряды 7-8), резисторам(10-15) и клеммам(23) не хватает Manufacturer Part. Я по ктаалогам попробую подобрать, но боюсь ошибиться, есть возможность добавить? Спасибо

RasimKalimullin, 2024-06-14:
[BOM_SPD.xlsx](https://github.com/user-attachments/files/15841647/BOM_SPD.xlsx)
Было бы хорошо, если проверили компоненты, спасибо!

Laxilef, 2024-06-14:
Резисторы и светодиоды подбираются по наличию в нужном размере (0805) и соответствующего номинала.
Проще брать из Basic списка - они дешевле. Но партномера там могут меняться дважды в день, поэтому в bom их нет и не будет.

Клеммы и прочие выводные детали лучше распаять самостоятельно.

closed 2024-07-04 by github-actions[bot]