Laxilef/OTGateway #76: OpenTherm Settings rollback to default

Quiss, 2024-06-26:
Hello @Laxilef, Thank you for the work, everything is just great.

I encountered a problem where after entering the settings - they are saved on the device, but reset in the settings panel to default values while other parameters remain unchanged. For example, in GPIO I use 7 / out 8, but after saving - I see 13 and 15 again, but the adjustment starts working!

This also applies to switches. It would be convenient to "modify" instead of entering again. Thank you!

[image]

Laxilef, 2024-06-26:
Hi @Quiss, thanks!
GPIO 7 and 8 is not valid for nodemcu 8266, see pinout:
[image]

Most likely you meant D7 and D8, which are equal to GPIO 13 and 15?

closed 2024-06-26 by Laxilef

Quiss, 2024-07-01:
@Laxilef why? (I don't understand microcontrollers)

When specifying 7 8 everything works, but as I said above - it resets to the default value

 [SmartTherm](https://www.ozon.ru/product/kontroller-kotla-otopleniya-smarttherm-808382394/) usage 7 and 8
https://drive.google.com/file/d/1Wv_77LTP45Vh3wCga1CM9oiGG6yuRg94/view
[image]

Laxilef, 2024-07-01:
Hi @Quiss,

See pinout
D7 = GPIO13
D8 = GPIO15

Quiss, 2024-07-01:
@Laxilef I'll have to learn this, Thanks!)