Laxilef/OTGateway #78: Cannot flash Lolin D1 mini v4.0.0

gunarser, 2024-08-20:
I've been using OT GW for quite a while now on my ESP32-S3-DevKitC-1 module. Everything seems to work fine with it.
Yesterday I wanted to finish the testing phase and move the code to a recently acquired Lolin D1 mini v4.0.0 (https://www.wemos.cc/en/latest/d1/d1_mini.html) that has USB C connector.
I downloaded firmware_d1_mini_1.4.3.bin and filesystem_d1_mini_1.4.3.bin files and tried to flash them using Flash Download Tools V3.9.7 but the upload failed.
I tried to change baud rate and different other settings but it continued to fail. If I flashed only the firmware then everything finished fine but I could not connect to the OT GW web interface. The issue seems to be only with filesystem flashing.
I tried to flash another D1 mini module but it failed as well.
What could be the reason for fails and how could I resolve it?
Thanks!

Laxilef, 2024-08-20:
Hi @gunarser,
Maybe you are using the wrong address for the filesystem? What error occurs when writing?

Laxilef, 2024-08-20:
Is there anything connected to the ESP pins at the time of flashing? drivers for CH340 installed?

gunarser, 2024-08-20:
- No, there is nothing connected to the pins. These are brand new modules, nothing is soldered on them. I just tried to flash OT Gateway images.
- Yes, latest CH340 drivers are installed.
- With 115200 baud rate nothing improves.
Tested on both D1 minis. On both it fails with the filesystem image.

Laxilef, 2024-08-20:
Strange, try via Chrome/Edge: https://espressif.github.io/esptool-js/

gunarser, 2024-08-20:
With https://espressif.github.io/esptool-js the same thing, i.e. filesystem flash fails. But it succeeded with https://esp.huhn.me web tool.
Thank you for the suggestion! Now my D1 mini is running the OT GW image and ready for production :)
Perhaps Espressif tools just don't like Windows 11...

Laxilef, 2024-08-20:
> Perhaps Espressif tools just don't like Windows 11...

I also use Win 11, the firmware is written normally using ESP download tool 🤷🏻‍♂️

closed 2024-08-21 by gunarser