Laxilef/OTGateway #82: HotSpot doesn't appear

maframan, 2024-09-26:
I tried to flash with ESP Flash Download Tool three different esp board (esp8266 d1 mini, nodemcu esp32 s, nodemcu esp32 s3) but in all cases the hotspot doesn't appear. I followed the out guide and for each board  used the correct filesystem and image file.
For d1 mini: filesystem_d1_mini_1.4.4.bin and  firmware_d1_mini_1.4.4.bin
For nodemcu esp32-s and esp32s3: filesystem_nodemcu_32_1.4.4.bin and firmware_nodemcu_32_1.4.4.factory.bin

Below the ESP Flash tool screenshot:

[image]

where am I going wrong?

edems88, 2024-09-27:
I have the same thing.

Laxilef, 2024-09-27:
Hi,

You are using the wrong addresses. The firmware file should be written to address 0x0, and the file system to the address as written in the manual:

ESP8266: 0x300000
ESP32: 0x370000

Not the other way around.

maframan, 2024-09-27:
Thanks, You are right

closed 2024-09-27 by Laxilef