Laxilef/OTGateway #83: Can not modify boiler values

emgarrote, 2024-10-07:
First of all, I want to thank you for this awesome project.

I have successfully connected my boiler (Ferroli Bluehelix Hitech RRT 28C) to the home assistant by means of the DIYless master opentherm. I can read all the values from the boiler (dwh and heating temperatures, dwh flow rate, pressure, etc).

However, when I try to enable/disable heating or dwh it doesn't work. I also can't modify heating or dwh temperatures. I have tried to change the master member ID, trying all the possible values from 0 to 255, also modifying the summer/winter mode checkbox, with no success.

What can I do? Is it possible to control the heating in on-off mode (as if it were a relay) as long as we do not get the modulating control to work?

Laxilef, 2024-10-08:
Hi,

> trying all the possible values from 0 to 255, also modifying the summer/winter mode checkbox, with no success.

Did you reboot the ESP after changing the member master id?

emgarrote, 2024-10-08:
Hi,

> Did you reboot the ESP after changing the member master id?

No, I didn't. After your message I have tried the numbers (0, 4, 211) restarting after every change with no luck.

> I can publish the code for bruteforce member master id, but you will need to build the firmware and flash your ESP, and then monitor the status through the console. Do you have such an opportunity? Can you handle this?

No problem, I can handle that. 
Is there a possibility that this could block the boiler?

Thanks for your help!

Laxilef, 2024-10-10:
> Is there a possibility that this could block the boiler?

No, I haven't encountered anything like that.
Can you upload logs with temperature set? Thx.

emgarrote, 2024-10-10:
I connected the computer to see the logs and I realised that it was actually working with the masterID 0. I thought it wasn't because I was expecting that the values changed inmediately, and it requires like a minute to update. I also realised that if the heating mode is not activated, the heating temperature does not update.

In addition, I thought that the DHW and Heating checkboxes were intended to commute the DWH and heating modes of the boiler, but they don't. To change these modes I need to do it manually in the boiler. Then, checking on the heating in the gateway, once the winter mode is on, it starts heating.

In conclusion, it actually works fine with MasterID code 0 and summer/winter mode disabled. I simply made a mistake expecting it to work differently. I hope this thread will help others with the same question.

Thanks for your help and for this awesome project!

closed 2024-10-10 by emgarrote

Laxilef, 2024-10-10:
Thanks for the info! Could you upload a screenshot of your opentherm settings to [your message](https://github.com/Laxilef/OTGateway/issues/3#issuecomment-2395169957)?

> In addition, I thought that the DHW and Heating checkboxes were intended to commute the DWH and heating modes of the boiler, but they don't. To change these modes I need to do it manually in the boiler. Then, checking on the heating in the gateway, once the winter mode is on, it starts heating.

Most boilers allow turn on/off heating and DHW via opentherm. Checkboxes on the dashboard page should do this.