Laxilef/OTGateway #85: Does anyone know about a pre-build solution like this?

ovizii, 2024-10-18:
Sorry to open an issue for this simple question, but it seems discussions are not active on this GitHub project. 
I'd like to give this a try with my Bosch GC2200 W boiler, but I have zero skills for soldiering and electronics in general. 

Is anyone aware of a place to buy this all pre-soldiered in a nice case ready to use? I am able to connect to it and flash new firmware versions, but anything concerning electronics and wires makes me hesitant.

Btw. I found some semifinished products with their own firmware solution, so the next question is, could your firmware be compatible with that hardware?

https://diyless.com/product/esp8266-opentherm-gateway
https://3dstar.ro/esp-opentherm-v12-ae

Laxilef, 2024-10-18:
Hi,
Of course you can use ready-made solutions. I recommend you to look at this:
https://diyless.com/product/opentherm-thermostat

Laxilef, 2024-10-18:
Are you sure your boiler supports opentherm? Bosch usually uses EMS bus.

closed 2024-10-21 by ovizii