Laxilef/OTGateway #86: Communication Error Opentherm Gateway to OEM Thermostat

maxmilan, 2024-10-20:
Hi there,

first of all (you heared it too many times but hey thats it) congrats and very big thank you for this far sophisticated software for Opentherm! I suppose you help a lot of people outside there to safe energy and safe the planet.

I have a Broetje Boiler BOK (Oil) central heater which communicates to a Opentherm Thermostat "OT RGI" which is basically "Siemens QAA73.210". I bought a Opentherm Gateway from diyless to reach the boiler, I complete it with an ESP S2 and connect it via WLAN to Homeassistant. So far i can reach it and can read out most values. (at least what the boiler can - it is a bit of a low cost boiler... I didnt bought it, i have to take it...)

My problem is now that the Room Thermostat (OT RGI) shows Error 88E what is regarding to manual is a communication failure. I belive it has to do with the Master Member ID. I tried out around different IDs (0-25, 50-60, 110,-115, even over 300...) but nothing changes. DUring the procedure the boiler runs amok due and "boil" everything and everybody. You mention to "bruteforce" it, but also that there is a flash file available to bruteforce it via the ESP. I think the ID has to be inside 0-255? am I right?

or could be there another failure that the gateway didnt passthrough the communication to the thermostat?
I would appreciate you help very much, since i really like to use you code to check the personal life of my boiler.

best regards
Max

Laxilef, 2024-10-20:
Hi,

Thanks! Did I understand correctly that you want to use opentherm thermostat and esp together? If yes, then it is not supported by firmware.
The problem is that the OpenTherm protocol only requires one master thermostat on bus. That is, you can use either the ESP thermostat or the "OT RGI" thermostat, but not both.
The only possible option is to connect the ESP in the gap, that is, the ESP must emulate the slave and transmit commands to the boiler.
Same issue: https://github.com/Laxilef/OTGateway/issues/28

maxmilan, 2024-10-21:
Okay I just saw you answered as well on GitHub to my question.
In a nutshell:
I want between my heater and thermostat the ot gateway like the I understood from pictures of diyless.com. I want continuously using my thermostat and also, if necessary alter the value, FX temperatures for roo or domestic hot water via HA. Isn't it that what a gateway do? Sitting between master and slave, monitore and alter the values?
BR Max

maxmilan, 2024-10-21:
Hi Laxilef,

soo thank you for your explanation. I think you are right and it seems beside all what I read before in advance I missed that what I was aiming for is not possible so far. Pity but thats life - next time i read the documentation even deeper and from the beginning. Thank you I learned a lot about heaters from your project and how thing work out.

Sorry about the confusion I bring up but thank you for answering fast and precisely to all of you!
Big up again for the project and I will stare now into the fire and invent a new solution for my problem (like we and ancestors homo sapiens did last 20k years...)

all the best and maybe with my next heater and or habitat I will use this repo here. 
Beside if I find a solution to my problem I give you a hint here, but so far you can close this issue here.
thx BR Max

Laxilef, 2024-10-21:
> Sitting between master and slave, monitore and alter the values?

If we change any value in the home assistant and transmit it to the boiler, then your thermostat (Siemens) will overwrite it with its own. For example, if you set the heating temperature to 60 degrees via the home assistant, we cannot tell the master "now the heating temperature is 60 degrees", because he is the master. In other words, when installing the ESP in the gap between the boiler and the thermostat (Siemens), the ESP will be the master for the boiler, and the master for the ESP will be your thermostat (Siemens).

This project and home assistant can do the same thing as your thermostat and can be controlled with an HMI panel or android tablet on the wall. If you are interested, I can take a photo of how it was done in my case.

maxmilan, 2024-10-22:
okay then I understood the Opentherm from the beginning wrong, or at least what wikipedia tells about Opentherm 3.0
I thought I can work on it Multi Point-to-point and can accept more than one master. I figured also out how this is meant: There is an additional shield (board) for my heater with what I can extent the system in terms of more heating circuits, sensors, pumps, valves etc. and there are also additional connectors for OT (at least plus 3 I suppose). So it seems if Id like to use your solution i have to add this board or remove the siemens thermostat, which I want to use as a backup in case that ESP or Homeassistant fails in the middle of the winter....

I would be very interested in the project you told with the android tablet or the HMI panel (? I dont know what it is; a human interface?) at least as long as it did not make you any trouble, so i could get some ideas from it how it could look like also instead of using the OT RGI Siemens...

Laxilef, 2024-10-23:
> I would be very interested in the project you told with the android tablet or the HMI panel (? I dont know what it is; a human interface?) at least as long as it did not make you any trouble, so i could get some ideas from it how it could look like also instead of using the OT RGI Siemens...

I use a wall-mounted [tablet yc-sm10p](https://portworld-solu.com/portfolio-item/yc-sm10p-10-1-smart-home-poe-touch-control-panel/), you can buy it on aliexpress. I installed the [Fully Kiosk Browser app](https://www.fully-kiosk.com/) on tablet, which always displays the home assistant dashboard in full-screen mode.
The tablet dashboard displays thermostats for temperature control. Everything else is done by automation in the home assistant without my participation.

Screenshot of the dashboard for tablet:
[image]

I'll add a photo on the weekend.

JoyceKimberly, 2024-10-27:
> okay then I understood the Opentherm from the beginning wrong, or at least what wikipedia tells about Opentherm 3.0 I thought I can work on it Multi Point-to-point and can accept more than one [...]

I also understood this to be the way a "gateway" is supposed to work. The basic code examples from both Diyless and Ihormelnyk also show this to be possible.

To be fair, as great as this project is, it is a thermostat replacement, not a gateway, and really shouldn't be called a gateway to prevent confusion.

Other than that this really is a great project 👍

Laxilef, 2024-10-28:
Hi @JoyceKimberly
I wrote above why this will not work correctly. The slave (ESP) cannot override the values ​​on the master (existing thermostat).
Ihor Melnyk examples [use transparent/proxy control](https://github.com/ihormelnyk/opentherm_library/blob/master/examples/OpenThermGatewayMonitor_Demo/OpenThermGatewayMonitor_Demo.ino) of the master (thermostat) by the boiler (slave) via ESP.

dafik, 2024-11-08:
there is another project that acts as a proxy between the thermostat and the boiler and can overwrite the thermostat messages.

https://otgw.tclcode.com/ the assembled version is available on nodo.nl