Laxilef/OTGateway #87: Ideal Logic Heat2 H18: Can enable/disable heating but unable to set heating temperature

grapefruitmanor, 2024-10-24:
Hi,

### Problem
I have an Ideal Logic Heat2 H18 heat-only boiler and, with the default settings, I am able to enable and disable the heating but am unable to change the target heating temperature; the target temperature shown on the boiler LCD display never changes and it fires only up to the displayed temperature rather than the one requested via the UI.
I see that very few sensor parameters are populated, with most remaining at zero (modulation, fault and diagnostic codes, heating temp and return temp, flame and fault are the only parameters which I have seen to change during operation).

I'm running on a Wemos C3 mini using Ihor Melnyk's OpenTherm adapter.

### Investigation
OpenTherm diagnostic reports the following:
```
Vendor:          ATAG/Brötje/ELCO/GEMINOX
Member ID:       4
Flags:           64
Type:            172
Version:         1
OT version:      4.097656
Heating limits:  30...80 °C
DHW limits:      30...60
```

I've tried several different Master MemberID codes (from 0 to 15: restart after each change, enable heating, change target temperature in the UI, wait for 1 minute then check temperature reported on boiler LCD display) with no success in changing the target temperature. I've also tried every combination of Summer/winter mode, Heating CH2 always enabled, and duplicate heating CH1 to CH2, again restarting after each change. Currently, only 'heating CH2 always enabled' is checked) and Master MemberID code is set to `0`.

With heating enabled, the following regular traces are seen via Telnet (I have DHW disabled because it's a heat-only boiler):
```
[00:15:04][OT][TRACE] ID:    0   Request:     1100   Response: c0001100   Attempt:  1   Status: SUCCESS
[00:15:05][OT][TRACE] ID:   25   Request: 80190000   Response: c019260c   Attempt:  1   Status: SUCCESS
[00:15:05][OT][TRACE] ID:   28   Request: 801c0000   Response: c01c1710   Attempt:  1   Status: SUCCESS
[00:15:05][OT][TRACE] ID:   33   Request:   210000   Response: 60210000   Attempt:  1   Status: INVALID
```

When sending a new target temperature request via the UI, the following Telnet traces are seen:
```
[00:14:53][PORTAL][WEBSERVER][TRACE] Request /api/settings 88 bytes: {"heating":{"enable":true,"turbo":false,"target":40},"dhw":{"enable":false,"target":60}}
...
[00:15:00][REGULATOR][INFO] New target: 40.00
...
[00:15:01][OT][HEATING][INFO] Set temp: 40.00 (converted: 40.00)
[00:15:02][OT][TRACE] ID:   57   Request: 90392800   Response: 60392800   Attempt:  1   Status: INVALID
[00:15:02][OT][HEATING][WARN] Failed set max heating temp
[00:15:02][OT][TRACE] ID:    1   Request: 10012800   Response: d0012800   Attempt:  1   Status: SUCCESS
```

### Questions
- Since I can enable/disable heating, does that mean that the master member ID is likely correct?
- Is there a way to definitively determine if the boiler has accepted the configured master member ID?
- Does a status of `SUCCESS` for ID `1` indicate that the boiler in theory accepted the new target temperature?

At the moment I'm not sure how to diagnose the issue further so would appreciate any help you can provide.

Thanks in advance for your help - I'm very excited to get this working and appreciate all the hard work done by everyone on this project!

Laxilef, 2024-10-24:
Hi,

> Since I can enable/disable heating, does that mean that the master member ID is likely correct?

Most likely yes.

> Is there a way to definitively determine if the boiler has accepted the configured master member ID?

Typically, boilers respond with INVALID status to ID 1 request, or the setpoint temperature does not actually change and heating cannot be turned on/off. That is, you can read parameters, but not write them.

> Does a status of SUCCESS for ID 1 indicate that the boiler in theory accepted the new target temperature?

dlduk, 2024-10-24:
Thanks for your quick response.

Good to know that the first hurdle of master member ID is most likely OK since I can turn on and off heating.

>0x2800 = 10240 (dec)
>10240 / 256 = 40
Of course, my mistake - thanks for clarifying.

Both equitherm and PID are disabled  - I left most settings at their default, other than adding MQTT, outdoor and indoor sensor support, and disabling 'DHW present'. I had previously tried toggling that and didn't see a difference.

I'm confused as to why the boiler would accept the new target temperature but not update the LCD display and not attempt to use the new target temperature.

grapefruitmanor, 2024-10-24:
Thanks for your quick response.

Good to know that the first hurdle of master member ID is most likely OK since I can turn on and off heating and ID 1 response was SUCCESS.

>0x2800 = 10240 (dec)
10240 / 256 = 40

Of course, my mistake - thanks for clarifying.

Both equitherm and PID are disabled - I left most settings at their default, other than adding MQTT, outdoor and indoor sensor support, and disabling 'DHW present'. I had previously tried toggling that and didn't see a difference.

I'm confused as to why the boiler would accept the new target temperature but not update the LCD display and not attempt to use the new target temperature.

Laxilef, 2024-10-24:
> Both equitherm and PID are disabled - I left most settings at their default, other than adding MQTT, outdoor and indoor sensor support, and disabling 'DHW present'. I had previously tried toggling that and didn't see a difference.

I meant the settings in the boiler service menu. Look at this:
[image]
[Logic_Heat2_Installation___Servicing_-_07.2023_UIN_228298_A06.pdf](https://github.com/user-attachments/files/17508973/Logic_Heat2_Installation___Servicing_-_07.2023_UIN_228298_A06.pdf)

If you have the curve mode enabled on the boiler, the boiler operates in a different mode and it expects the room temperature setting, not the coolant temperature setting.

> but not update the LCD display

In fact, this is normal; when connecting opentherm devices, the setpoint temperature is usually not displayed and does not change on the boiler display.

grapefruitmanor, 2024-10-24:
Weather compensation is automatically enabled/disabled on the detection of an external sensor on the weather compensation boiler terminals (see [the weather compensation kit installation manual](https://idealboilers.ie/uploads/documents/[Weather_Compensation_Kit_Installation_Guide.pdf](https://idealboilers.ie/uploads/documents/Weather_Compensation_Kit_Installation_Guide.pdf))) so I can't change it via the service menu.

>In fact, this is normal; when connecting opentherm devices, the setpoint temperature is usually not displayed and does not change on the boiler display.

That's surprising to me - now I know not to use that as a guide.

My expectation was that if I set to the target heating temperature to, e.g. 40 °C with a current heating temp of 28 °C, then the boiler would fire up to 40 (with some overshoot) then hold this new temperature, meaning in the gateway UI I should see 'Heating temp' showing this behaviour, and I should see/hear the boiler firing to meet the new target. Is that correct?

I'll do some more observations on 'heating temp' while setting different target temperatures and report back with the log data.

Laxilef, 2024-10-24:
> Weather compensation is automatically enabled/disabled on the detection of an external sensor on the weather compensation boiler terminals (see [the weather compensation kit installation manual](https://idealboilers.ie/uploads/documents/%5BWeather_Compensation_Kit_Installation_Guide.pdf%5D(https://idealboilers.ie/uploads/documents/Weather_Compensation_Kit_Installation_Guide.pdf))) so I can't change it via the service menu.

Okay, do you have an external sensor connected?

> My expectation was that if I set to the target heating temperature to, e.g. 40 °C with a current heating temp of 28 °C, then the boiler would fire up to 40 (with some overshoot) then hold this new [...]

Yes, the boiler should start heating and maintain the setpoint temperature, but the boiler display may not change the setpoint temperature.

grapefruitmanor, 2024-10-24:
Sorry, I should've mentioned (that's the problem with your head deep inside a task): no, I don't have an external sensor connected.

Laxilef, 2024-10-24:
Then this is really strange :)

grapefruitmanor, 2024-10-24:
Oo, no, I had the knob set to 30 °C, thinking OpenTherm would be able to override whatever it's set to. Is it common for boilers to behave like that? Did you discover this by chance?
It strikes me as very odd. Regardless, increasing the display temperature to 80 will be the first thing I do when I next get chance to test this. Fingers crossed this is what has been blocking it!

Laxilef, 2024-10-24:
I saw similar behavior on the Bosch Gaz 6000. The temperature set on the boiler panel was the maximum temperature that could be set via Opentherm.
But there it was displayed correctly in the "OpenTherm diagnostic" section.

critictidier, 2024-10-25:
> Did you discover this by chance?

I did indeed 🫠 good luck 🤞

grapefruitmanor, 2024-10-25:
I just tested turning the boiler's own target temperature up to 80 °C via the physical knob, then set a few target temperatures via OpenTherm and I'm pleased that it's now working as expected, so a big thank you to both of you for your help in getting it working! 😄
The OpenTherm diagnostic never updates with the value set by the knob.

A lot of useful information has come out of this investigation which I couldn't find documented anywhere in the Wiki:
- The target temperature set by OpenTherm is often not displayed/updated on the boiler display
- Some boilers require the target temperature on the panel to be set to at least the maximum expected to be set by OpenTherm, acting as a hard limit on the OpenTherm requests
- Some boilers report the target temperature on the panel as the max heating limit in OpenTherm diagnostic
- The heating must be enabled before the controller will issue a target temperature request to the boiler
- Checking for a `SUCCESS` response to ID `1` in the logs indicates that the master member ID is probably correct
- `(ID_1_request & 0xFFFF) / 256d` = requested target temperature in °C

What is your policy regarding Wiki edits? I'd be happy to add the information above to the FAQs page if that's OK. I think it may help others who face this confusing situation and prevent some simple issues like this from being raised. I'd also like to update the boiler compatibility list for my boiler (if added, that would mean 3 Ideal boilers in 'Other' so maybe at that point it would be worth having an 'Ideal' section).

Also, I have an external pump and would like to be able to drive it via a Shelly relay and HA's Shelly integration, rather than having some GPIO>Bluetooth transmitter to HA, but that means I need the external pump GPIO state published over MQTT and I couldn't find it in any of the entity lists (I have it enabled on the gateway). Am I right in assuming that the state currently isn't published? I do wonder if the 5+ sec delay via MQTT>HA>relay activation might be too slow for the boiler and throw a fault due to heating on but no flow (I saw this happen sometimes during my testing if I had the pump disabled and heating on).

Laxilef, 2024-10-25:
> What is your policy regarding Wiki edits?

I'll be glad if you do it :)

> Also, I have an external pump and would like to be able to drive it via a Shelly relay and HA's Shelly integration, rather than having some GPIO>Bluetooth transmitter to HA, but that means I need [...]

You can turn on an additional pump by the event of turning on the ``binary_sensor.opentherm_heating`` sensor in the home assistant.

critictidier, 2024-10-26:
> I'm pleased to say that it's now working as expected, so a big thank you to both of you for your help in getting it working!

Congratulations 🥳 we have very similar boiler model, you might also find my “wiki” useful https://github.com/Alexwijn/SAT/discussions/40

grapefruitmanor, 2024-10-26:
> Congratulations 🥳 we have very similar boiler model, you might also find my “wiki” useful [Alexwijn/SAT#40](https://github.com/Alexwijn/SAT/discussions/40)

In fact, it seems we also have a very similar house configuration (S plan, two CH zones, one hot water cylinder).
SAT is very interesting to me, though it looks like it's only compatible with the OTGW firmware, at least trying to control via MQTT. How are you running it?

critictidier, 2024-10-26:
Yes, it’s that firmware. There is more ongoing development if you read around ESPhome, SAT and adapters like yours. All these projects including this one are wonderful. They all take our heating systems to the next level! 🥰

Laxilef, 2024-10-27:
Hi @grapefruitmanor 

I have made the wiki available for editing. If you're still ready - welcome.

grapefruitmanor, 2024-10-27:
> Hi @grapefruitmanor
> 
> I have made the wiki available for editing. If you're still ready - welcome.

Great, thanks. I've been busy today and yesterday so I planned to make the edits tomorrow when I should have some time.

grapefruitmanor, 2024-10-29:
The Wiki edits are all done on the Compatibility and FAQ pages so at least I've been able to give something small back 😄 

One last question (which may end up with another FAQ contribution...): it looks like I can enable both PID and equitherm simultaneously but are they opposing concepts or can they work in parallel?
I would've thought that normally they oppose each other because load compensation is inherently reactive whereas weather compensation is proactive, i.e. when the outdoor temperature starts to drop but the internal temperature is unchanged, weather compensation would request a higher setpoint whereas PID would not.

Am I right in thinking that the K and T equitherm coefficients apply a level of PID-like behaviour by modifying the curve to account for steady-state error (in the case of K, behaving like an integral term) and response time (in the case of T, behaving like a derivative term)?

My house is only 2 years old and supposedly very energy-efficient, yet I found that with heating target set to 20 °C, N=1.0 is required to keep the temperature at 20 ± 0.5 °C (though I do keep window trickle vents open which will presumably account for some heat loss), whereas I was expecting closer to 0.7 or less as your description suggests. All radiator TRVs are set to ~22 °C and zone thermostats to 23 °C, ensuring that the pump and radiators are all running constantly without restriction. It's no big problem but it is curious to me.

Laxilef, 2024-10-29:
Thank you for your contribution! :)

Equtherm and PID actually work differently:
- Equtherm calculates the temperature using an algorithm based on outdoor temperature and makes a correction based on the indoor temperature (T coefficient). The algorithm is not very complicated, take a [look at the code](https://github.com/Laxilef/OTGateway/blob/master/lib/Equitherm/Equitherm.h). Briefly:
N - curve, K - curve slope, T - thermostat error correction.
- The PID adjusts the temperature so that the indoor temperature is equal to the setpoint.

When using both modes, PID replaces the correction for the thermostat error (coefficient T). The influence of PID on the calculated temperature from Equtherm can be limited by the PID max temperature parameter.

**Example:**
The maximum PID temperature is set to 10 degrees.
Equtherm has calculated a temperature of 40 degrees, but the indoor temperature is below the set point (for example, a powerful hood is running or the windows are open).
The PID will change +- 10 degrees for 40 degrees. That is, the final temperature can be from 30 to 50 degrees.

> My house is only 2 years old and supposedly very energy-efficient, yet I found that with heating target set to 20 °C, N=1.0 is required to keep the temperature at 20 ± 0.5 °C (though I do keep [...]

The coefficients for Equtherm are selected empirically, since there are many factors: different insulation technologies, many large windows, powerful ventilation (exhaust), etc.

> Oh, and I couldn't find a description for how turbo mode for heating works.

Turbo mode is only relevant when using only Equtherm, it simply temporarily [increases](https://github.com/Laxilef/OTGateway/blob/11b1277d79581a4420501c0be3c0533d5525bb4a/src/RegulatorTask.h#L219C1-L223C8) the T coefficient to 10.

robbaker92, 2024-11-11:
> Is the physical temperature knob turned to at least the temperature you require? I have an ideal logic h24 and the physical knob always takes priority over what opentherm requests.

This is true of Ideal c30 model also.

closed 2024-12-02 by github-actions[bot]