Laxilef/OTGateway #88: Частые "Software reset due to exception/panic"

upais, 2024-10-29:
Добрый день.
Начал замечать, что контроллер падает\ребутается каждые несколько часов. Раньше тоже такое было, но редко, может 1-2 в месяц.
Причина: Software reset due to exception/panic

Отладку включил, консолью подключился, но отловить довольно проблемно. Как-то еще можно диагностику провести?

[image]
[image]

Laxilef, 2024-10-29:
Здравствуйте, попробуйте бету.
[build.zip](https://github.com/user-attachments/files/17562652/build.zip)

upais, 2024-10-30:
Неа, тоже самое, причина только: Reset due to other watchdogs
До прошивки беты решил еще раз прошить 1.4.5, т.к. падать стал каждые 3-7мин. На Бете вроде реже, ночью даже 4 часа проработал. Настройки не менялись, после перепрошивки восстанавливал бэкап двухнедельной давности.
[debug (1).json](https://github.com/user-attachments/files/17566846/debug.1.json)
[image]

Laxilef, 2024-10-30:
Спасибо за логи. А какой датчик ble используете?

upais, 2024-10-30:
LYWSD03MMC на прошивке pvvx.
С ним тоже ничего не менялось.

Laxilef, 2024-10-30:
Ок, попозже выложу новый билд для тестов.

upais, 2024-10-30:
Есть подозрения, что падает из-за взаимодействия с BLE датчиком?
Я недавно переключил на MQTT, пока не показатель, но вот 2 часа не падает. 

Кстати, а возможно, чтобы при старте контроллер опрашивал датчик и получал актуальную температуру? Либо хотя бы запоминал последнюю, если датчик невозможно принудительно опросить. А то основная проблема падений, это то что после старта контроллер считает, что на внутреннем датчике 0гр, а датчик может долго не передавать новые данные (у меня было более часа), то котел греет и греет.

upais, 2024-10-30:
Да вроде должна быть нормальная связь, я его и перемещал ближе, там метров 4-5 через стену.
Попробую ему мощность увеличить, интереса ради.
А так, по mqtt уже более 5 часов.

Laxilef, 2024-10-30:
Попробуйте с этим билдом
[build.zip](https://github.com/user-attachments/files/17580063/build.zip)

На качество связи с датчиком, однако, это не повлияет, только должно устранить краши.

upais, 2024-10-31:
Предположительно, решилась проблема. Порядка 1,5 часов работало без падений, а в логах сообщения, что не удалось подключиться к BLE датчику. Я решил изменить ему мощность, она дефолтовая стояла и тоже не смог подключится, далее вынул батарейку и больше он с ней даже не включился. Похоже из-за низкого заряда датчик подглючивало. Хотя когда это началось датчик передавал данные примерно так же оперативно как и раньше.

Laxilef, 2024-10-31:
> Порядка 1,5 часов работало без падений

Я так и не понял, вы последний билд прошили или нет?

upais, 2024-10-31:
Конечно, это с ним 1,5 часа проработало. По этому и говорю, что скорей всего, проблема решена. Хоть 1,5 часа не много, но на предыдущих падало каждые 5-15мин.

Laxilef, 2024-10-31:
Хорошо. Понаблюдайте еще, если все хорошо, то можете закрыть issue.

upais, 2024-11-01:
Большое спасибо за оперативную помощь!

closed 2024-11-01 by upais