Laxilef/OTGateway #90: Viessmann. Radiators upstairs, underfloor heating downstairs

Daveblanche, 2024-11-12:
Hi there
The gateway works well, but I am struggling to figure out how to allow the radiators upstairs to call for heat from the boiler. The radiators are controlled by tado and the extension box is worded in as a standard on/off relay based on the call for heat from the thermostats upstairs.
It’s not on opentherm.
The underfloor heating (I presume is channel2?) works fine but seems to run the show.
I don’t *think* this was the case before OpenthermGateway. I think the rads were able to call for heat via the tado relays. 
what do the channel 1 and 2 controls actually do?

Daveblanche, 2024-11-12:
To be fair, it is probably a non issue once I have the heatcurve tuned to the Heatloss….the boiler/heating will constantly have warm water for the radiators…which run in a temperature limited state. They are upstairs and run at 18c instead of 20c downstairs in the ufh.

Laxilef, 2024-11-12:
Hi,

To be honest, I didn't understand anything. What is the question?

Daveblanche, 2024-11-12:
The radiators upstairs don’t seem able to call for heat. They are controlled by tado via simple relay switching (old school). Tado controls dhw  and this is fine. 
the downstairs is controlled by the secondary circuit, underfloor heating. It works fine. 
what do the channel options in opentherm settings in this project do? Can I arrange things via settings that means that the radiators can call for heat/flame/modulation from the boiler?

Laxilef, 2024-11-13:
Can you draw a diagram of your heating system and how it is connected to the boiler?
And what is the model of your boiler?

The firmware uses the second channel, but it duplicates the parameters from the first channel.

Laxilef, 2024-11-13:
I meant the hydraulic diagram and the diagram of connecting Tado to the boiler.

Daveblanche, 2024-11-14:
I have an admission to make. I’m normally quite intelligent but not this time.
My system is set up as priority dhw.
I had the gateway dhw on all the time, via the checkbox, so my radiators wouldn’t heat up. 
I missed that point.
Obviously they got colder more quickly than the underfloor heating so I focussed on the radiators in my first post. 
It took me a while to realise that „duplicate ch2 to dhw“ actually means set (or mirror) dhw onto ch2 in the gateway. 
Funnily enough, I ran out of hot water yesterday. Once set so that ch2 is recognised by the system as dhw, it all started working…. Immediately. 
This is my fault and nothing to do with the gateway you developed: which is awesome and now working beautifully.
Thank you so much for the offer of support, but I was just being slow. 
D

Laxilef, 2024-11-14:
If the problem is solved and you don't need my help, you can close it :)

Daveblanche, 2024-11-14:
Veissmann 100w-B1hc  30Kwh limited to 14Kwh max Output, system boiler with priority dhw.
300l water tank.
Previously installed weather compensation sensor and demand box for dhw. Weather sensor works with OpenthermGateway being reported via opentherm.
Heating upstairs: radiators x10 with Tado thermostats on wall and radiators.
Heating downstairs: 120sqm of underfloor heating running through a Heatmiser system and standard  hot opentherm) manifold and controller.

mistakes I made:
Going to quickly with changes to equitherm basic co-efficient. It does take patience.
I left dhw on constantly in the checkbox….which meant my heating didn’t work (priority dhw- foolish mistake)
I didn’t duplicate channel 2 to dhw via the checkbox. I misunderstood what duplicate meant for someone who might not use English as a first language. My issue, not anyone else’s. 

Settings:
Equitherm basic co-eff is 3.5 but I might have to drop this a little tomorrow.
Haven’t yet tuned the other two but time constant will probably be 2 and delta (k) will be 2 or 3.
Heating: hysteresis is 0.1. min is 20c max is 65c. Max modulation is 38.
I altered the customize.yaml to give 0.1c changes in temp because my wife is VERY fussy on temperature.

Opentherm settings: (checked boxes)
Dhw present
Duplicate dhw to ch2
Sync modulation with heating
I unchecked get max and min from heater as my Veissmann froze on those a couple of times. But the gateway is far more advanced than the boiler, so I see no issue (yet)

Thank you for the offer of help.
All closed!

closed 2024-11-14 by Daveblanche