Laxilef/OTGateway #91: Where can I find the MQTT set topics/commands

Imaginous, 2024-11-12:
I want to use the OTGateway with my own domotica software.

I searched, but I couldn't find a list of MQTT commands I can send to set the wanted heating temperature, the wanted DHW temperature, the current room temperature and the current outside temperature.

Probably I didn't look hard enough, but if anyone can point me in the right direction ;)

Laxilef, 2024-11-12:
Hi,

``prefix/setttings``
``prefix/setttings/set`` - any values ​​as in ``prefix/settings``

``prefix/state``
``prefix/state/set`` - any values:
```json
{
  "actions": {
    "resetDiagnostic": true,
    "resetFault": true,
    "restart": true
  },
  "temperatures": {
    "indoor": 1.0,
    "outdoor": 1.0
  }
}
```

Imaginous, 2024-11-12:
Thank you for the quick response.

closed 2024-11-12 by Imaginous

menyuguren, 2025-07-03:
"I used opentherm/settings/set to modify:
{
  "master": {
    "heating": {
      "targetTemp": 55,
      "setpointTemp": 55
    }
  }
}
But it didn’t work, and I received the error message: SyntaxError: Unexpected end of JSON input."