Laxilef/OTGateway #92: Boiler Immergas Victrix Zeus 25

VladaWaas, 2024-11-13:
Actualy status:
[image]
This is compatible points with OpenTherm:
[image]
Could you add points: 19,77 and 85 to your firmware?
Thank you in adwance.
Vladimir

Laxilef, 2024-11-13:
Hi,

ID 19 - DHW flow rate, it is already supported.
What do ID 77 & 85?

VladaWaas, 2024-11-13:
Hello,
thx for fast respose. ID flow rate is permanently zero on interface. Can I any mistake in my setup?
ID 77 represented intake relative ventilation level 0-100%
ID 85 represented actual inlet fan speed in rpm

Laxilef, 2024-11-13:
I will add polling ID 85 to the new version.

https://github.com/Laxilef/OTGateway/wiki#fit-coefficients

Laxilef, 2024-11-13:
> ID19- flow rate is permanently zero on interface. Can I have any mistake in my setup?

Well, not all boilers report this parameter.

VladaWaas, 2024-11-13:
> > ID19- flow rate is permanently zero on interface. Can I have any mistake in my setup?
> 
> Well, not all boilers report this parameter.

Immergas is not in the OpenTherm association, but its extension modules are based on OpenTherm protocols. According to a non-public information from their developer, ID 19 should be active. But it is possible that the query string may differ in some small way from the OT spec.

VladaWaas, 2024-11-13:
> I will add polling ID 85 to the new version.

Thank you!
> 
> > I have one more question about setting the equithermal constants.
> 
> https://github.com/Laxilef/OTGateway/wiki#fit-coefficients

I reading it last week. I did exactly according to your instructions immediately after installation. But I'm a pretty impatient person. Any intervention in the system has a long inertia. I will have to improve! :))))

Laxilef, 2024-11-13:
> Immergas is not in the OpenTherm association, but its extension modules are based on OpenTherm protocols. According to a non-public information from their developer, ID 19 should be active. But it is possible that the query string may differ in some small way from the OT spec.

Hmm, in the firmware we interpret response as float. Can you open the hot water tap and start logging?

VladaWaas, 2024-11-13:
> Hmm, in the firmware we interpret response as float. Can you open the hot water tap and start logging?

yes I do. Can I acces via telnet in debug mode?

VladaWaas, 2024-11-13:
[putty.log](https://github.com/user-attachments/files/17738751/putty.log)

Laxilef, 2024-11-14:
``ID:   19   Request: 80130000   Response: 70130000   Attempt:  1   Status: INVALID``
Your boiler responds with INVALID response to DHW flow rate request

VladaWaas, 2024-11-15:
yes i noticed it as soon as i opened telnet. But I don't want to give up yet. I will ask the developer technician from the manufacturer if it is the same request at Imerrgas as specified by opentherm. Can you please send me the script of your command to fetch the values?

VladaWaas, 2024-11-15:
so I don't have positive information :(
 ID19 is active only in models with direct flow heating without a hot water storage tank

closed 2024-12-05 by github-actions[bot]